scikit-rf/scikit-rf #833: 'Network' object has no attribute 'plot_s_vswr'

hansen-mc, 2023-02-06:
Hi everyone, 
I simply tried to plot a touchstone, which does not sem to work fine. 
I am using conda with scikit-rf 0.24.1 and spyder. 

```
from skrf.data import ring_slot
import matplotlib.pyplot as plt

a = ring_slot['80-90ghz']
a.plot_s_vswr()

plt.show()
```

this leads to 
```AttributeError: 'Network' object has no attribute 'plot_s_vswr'```

also using different plot variants doesnt work or with or without plt.show()

Is it me or a real issue?
Thanks!

jhillairet, 2023-02-06:
Your example works fine and display the figure on Spyder and Jupyter on my machine.

FranzForstmayr, 2023-03-10:
There's probably something wrong with your matplotlib installation, maybe it's too old.
Could you please update to a recent version? 

skrf silently fails, when there's an issue when importing matplotlib to allow the usage without having it installed.

jhillairet, 2023-03-12:
@hansen-mc did you try (re)installing matplotlib ? and I can close this issue?

closed 2023-03-15 by FranzForstmayr