scikit-rf/scikit-rf #839: Failed to load simple CITI file without VAR additionally to frequency

Nemesis82, 2023-02-09:
Loading a simple touchstone file without any `VAR` except for `freq` is currently not possible. In the routine to generate a `NetworkSet` the `params` property is called. The property returns all sweeps except for the frequency, i.e., if there is no additional sweep, an empty list is returned. An example CITI file is provided as attachement [example.txt](https://github.com/scikit-rf/scikit-rf/files/10697672/example.txt).

A workaround is to provide a dummy sweep in case only the frequency sweep is present.

```
    def _parse_citi(self, lines):
...
# Workaround solution
        if len(self._params.keys()) == 1:
            self._params["dummy"] = {}
            self._params["dummy"]['values'] = np.array([0])
            self._params["dummy"]['occurences'] = 1
# End or routine
```

jhillairet, 2023-02-09:
can you provide a pull request (for #840 also?)

Nemesis82, 2023-02-09:
I yes can do this. However, I need to run the test scripts for the CITI files with the modified code, first. Will do tomorrow in the office.

jhillairet, 2023-03-20:
@Nemesis82 , is there any chance you could test the PR #854 ?

PR #854 (2023-03-21): Fix CITI parser
fix #839 and #840 
- fix CITI parser to support simple CITI file
- test parsed values
- parse port impedance data
closed 2023-03-21 by jhillairet