scikit-rf/scikit-rf #877: Incorrect documentation or implementation of Frequency.unit

eendebakpt, 2023-04-04:
The implementation of `Frequency.unit` and the docstring do not match. An example:
```
import skrf

f = skrf.F(1,5,5, unit='GHz')
print(f.unit)
#f.unit?
```
returns `GHz` (uppercase).

But the docstring claims:
```
        """
        Unit of this frequency band.

        Possible strings for this attribute are:
        'hz', 'khz', 'mhz', 'ghz', 'thz'

        Setting this attribute is not case sensitive.

        Returns
        -------
        unit : string
            lower-case string representing the frequency units
        """
```

jhillairet, 2023-04-04:
Right. In fact it takes as input whatever upper or lower cases ('ghz', 'GHz', 'GHZ', etc). What would you propose to be more clear in the doc?

eendebakpt, 2023-04-04:
> Right. In fact it takes as input whatever upper or lower cases ('ghz', 'GHz', 'GHZ', etc). What would you propose to be more clear in the doc?

I do not now the way it was intended. But if we keep the implementation the same, I would write
```
        """
        Unit of this frequency band.

        Possible strings for this attribute are:
        'Hz', 'KHz', 'MHz', 'GHz', 'THz'

        Setting this attribute is not case sensitive.

        Returns
        -------
        unit : string
            Upper-case string representing the frequency units
        """
```

PR #879 (2023-04-04): Fix docstring for Frequency.unit
Fixes #877