scikit-rf/scikit-rf #883: Attribute Error with the module from skrf import media

Mariano2nd, 2023-04-07:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**
A clear and concise description of what the bug is.
I am having this problem with this module from skrf import media I keep getting the same error message every time. I run this code
fs = media.Freespace(frequency = ring_slot.frequency,z0=50) and the terminal will give me this error message 'skrf.media.media' has no attribute 'Freespace'. 

**To Reproduce**
Steps to reproduce the behavior:
If you want to try this problem you can follow this video guide -> https://youtu.be/9ZLocBIxOJY or just type these few line of code.
1: Import skrf, skrf.data; ring_slot, and from skrf import media. 
2: fs = media.Freespace(frequency = ring_slot.frequency, z0 = 50)
    fs
(btw I am using VSC jupyter notebook version)

**Expected behavior**
A clear and concise description of what you expected to happen.
What you should expect when you run this code in the editor is 'Freespace Media. 75-110 Ghz. 201 points'
**Data**
If applicable, add the data leading to the issue (a minimum example is appreciated)
For my data I used ring_slot from skrf.data
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
I uploaded screenshots of the library and the snippet of the code.
**System**
 - OS: Windows 10 Pro
 - scikit-rf version 0.26.0

**Additional context**
Add any other context about the problem here if relevant.

cafeclimber, 2023-04-07:
Does the commented code not run?

Not sure what's going on with the other code, but I was just able to run this no problem

```python
import skrf
from skrf.media import Freespace

fs = Freespace()
print(fs)
```

Mariano2nd, 2023-04-07:
@cafeclimber I tried the code and I was able to follow the video using the code. 
[image]
My question is that why doesn't the media class does not work the same way as the video when I tried doing it?
[image]
https://youtu.be/9ZLocBIxOJY?t=420
Can you try it on your side the same way the person in the video did, I wonder if you would get the same error as me.

jhillairet, 2023-04-07:
> My question is that why doesn't the media class does not work the same way as the video when I tried doing it?

The video is probably not totally up to date (we had the project to redo it and add some new ones, but...) with the current version. Currently, not all the media are automatically imported into `media` and some of them must be explicitly imported, like:

```python
import skrf as rf
from skrf.media import Freespace

Freespace(frequency=rf.data.ring_slot.frequency, z0=50)
```

Mariano2nd, 2023-04-07:
@jhillairet ok

jhillairet, 2023-04-12:
Closing, but nead to update the documentation and eventually make new videos

closed 2023-04-12 by jhillairet