scikit-rf/scikit-rf #884: Import timestamp data while reading in a sNp file

mgoulder303, 2023-04-07:
Most (if not all) *.sNp files generated from PNAs contain a timestamp in the header. It would be a great to import this timestamp into the Network object. I came across this as I have a large number of S-parameters I would like to plot over time. The filename does not have a timestamp but the file does in the header.

jhillairet, 2023-04-08:
If the timestamp is in the comments in the header of the Touchstone file, it is available in the `.comments` property of the Network. Then it can be parsed to your need.

cafeclimber, 2023-04-08:
On this, should comments be a `list[str]` instead of `str`? When constructing an mdif you pass a `list[str]` as comments. Might be easier to parse instead of one giant string blob

jhillairet, 2023-04-08:
> On this, should comments be a `list[str]` instead of `str`? When constructing an mdif you pass a `list[str]` as comments. Might be easier to parse instead of one giant string blob

Yes eventually.

mgoulder303, 2023-04-08:
> > On this, should comments be a `list[str]` instead of `str`? When constructing an mdif you pass a `list[str]` as comments. Might be easier to parse instead of one giant string blob
> 
> Yes eventually.

Good to know about the .comments property. Thanks.

As @cafeclimber mentioned, having a list[str] would be nice as a future feature.

closed 2023-04-08 by mgoulder303