scikit-rf/scikit-rf #890: pandas version pinned to "pandas ~=1.1",

TM90, 2023-04-12:
scikit-rf version to 0.23.0 seems to allow pandas 2.0.0 but newer scikit-rf versions do not.

Is pinning the pandas version to 1.1 really necessary?
Is it "safe" to use `scikit-rf 0.23.0`  with `pandas 2.0.0`?

jhillairet, 2023-04-12:
Thanks for reporting. 

> Is it "safe" to use `scikit-rf 0.23.0` with `pandas 2.0.0`?

It should be compatible, at least I don't see any evident reason why it should not (except maybe some warnings than are now errors in in pandas 2.0). To be tested.

@FranzForstmayr do you know if the versions identified in `pyproject.toml` are minimum versions, or pinned ones? (ie. should we replace all the `~=` with `>=` ?)

FranzForstmayr, 2023-04-12:
`pyproject.toml` sticks pandas to `~1.1` which means `>=1.1` and `<2`.
The main reason pandas version 2 is not allowed is simply that we haven't tested this configuration yet, although I don't expect any difficulties.

I can have a look into this topic later today, maybe it's fine to drop the upper bound.

closed 2023-04-13 by FranzForstmayr