scikit-rf/scikit-rf #901: Timeout with NanoVNA_v2

Yannick-R59, 2023-04-26:
On some vna 2 (perhaps all), you can set the ifbw by playing on an "average" terms: Then the measurement is quite long but more accurate. A same program works well with an averaging of 10 but failed with an averaging set to 80. You can set the average on the nanovna before the start of the python program after, you have no more access to the local screen of the device so you can't anymore.
So an update of the VNA drivers NanoVNA_v2.py with a function to set the timeout of the scpi command would be welcome. That is for the bug which a real problem. But an average function in the drivers to be able to set this average with python would be also welcome. the command could look to send with scpi: 0x20 (write) 0x40 (adress of this average parameter) 0x03 (average value=3)

**Expected behavior**
No timeout :)

exemple of program
----------------------
import numpy as np
import matplotlib.pyplot as plt
import math
import skrf as rf
from skrf.vi import vna

nanovna = vna.NanoVNAv2('ASRL4::INSTR')
**#nanovna.timeout = 250000  has no effect** even if it should be according to https://pyvisa.readthedocs.io/en/1.8/resources.html
nanovna.set_frequency_sweep(10e6, 3e9, 1024)
nw_S11 = nanovna.get_snp_network(ports=(0,))
nw_S11.plot_s_smith()
nanovna.close()

---------------------------------------
Error:

Traceback (most recent call last):

  File C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File c:\users\yannick\desktop\litevna\essais_python\essai1.py:33
    nw_S11 = nanovna.get_snp_network(ports=(0,))

  File C:\ProgramData\anaconda3\lib\site-packages\skrf\vi\vna\nanovna_v2.py:428 in get_snp_network
    data_s11, data_s21 = self.get_s11_s21()

  File C:\ProgramData\anaconda3\lib\site-packages\skrf\vi\vna\nanovna_v2.py:204 in get_s11_s21
    data_raw.extend(self.resource.read_bytes(32 * len_segment))

  File C:\ProgramData\anaconda3\lib\site-packages\pyvisa\resources\messagebased.py:371 in read_bytes
    chunk, status = self.visalib.read(self.session, size)

  File C:\ProgramData\anaconda3\lib\site-packages\pyvisa_py\highlevel.py:519 in read
    return data, self.handle_return_value(session, status_code)

  File C:\ProgramData\anaconda3\lib\site-packages\pyvisa\highlevel.py:251 in handle_return_value
    raise errors.VisaIOError(rv)

**VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.**

------------------------------- Pyvisa -------------------------------------
https://pyvisa.readthedocs.io/en/1.8/resources.html
timeout

Very most VISA I/O operations may be performed with a timeout. If a timeout is set, every operation that takes longer than the timeout is aborted and an exception is raised. Timeouts are given per instrument in milliseconds.

**For all PyVISA objects,** a timeout is set with

my_device.timeout = 25000

jhillairet, 2023-09-08:
@Yannick-R59 , is the issue still exist with the latest version?

jhillairet, 2023-09-20:
@cafeclimber is it fixed by #958 ?

cafeclimber, 2023-09-20:
It would be, but there was someone who'd opened a PR on my fork to improve the nanovna v2 class. Without one to test with, I can't be certain

Yannick-R59, 2023-09-22:
Hello,

Thank you to follow this issue. 

so I have
- updated anaconda, 
- updated spyder (5.4.2)
- updated scikit-rf (0.29.0)

but now I have this error:
 nanovna = skrf.vi.vna.NanoVNAv2('ASRL4::INSTR')
AttributeError: module 'skrf.vi.vna' has no attribute 'NanoVNAv2'

So apparently with my version, I have lose the "nanovnav2".

Yannick-R59, 2023-09-22:
(english is not my native langage, so I try to be clear but....)
So to answer to your suggestion, there are many kind of NanoVna.
From my understanding, the original one, called nanovna is a 300MHz VNA. Then as the source is square wave, people begin to play with the harmonics of the square wave to reach higher frequencies. Obviously this is  a little bit less good than using the fundamental. So they have modify a little bit the hardware keeping this way to reach higher frequencies and I believe, on this trend, the final nanovna is nanovna-H limited to around 1.5GHz using harmonics to reach this values. Then, as this is very nice concept, having for less than 100-200$ something that allow hobbyst to do stuff similar to those done with 100k$ equipment, new products appeared. I belive this is so called v2 nanovna (it is complicated because, original hardware was quite open source, then there was many clone not all good and new version where not so open to avoid clones or to make money, and there is no more really a unity) so in this nanovna v2 "trend", there are one called LiteVNA (compatible with v2, that I have and that was working with scikit several month ago). This is a nice hardware with a fundamental up to 6.3 GHz (you can go above with harmonic but there some issues). So this is really a very good equipment, and so somebody, has introduced the nanovna v2 has a new device for scikit rf.
So for exemple we can still find on scikit website in the documentation:
# connect to NanoVNA on /dev/ttyACM0 (Linux)
nanovna = skrf.vi.vna.NanoVNAv2('ASRL/dev/ttyACM0::INSTR')
# for Windows users: ASRL1 for COM1
# nanovna = skrf.vi.vna.NanoVNAv2('ASRL1::INSTR') 
From my understanding, has the hardware is quite different between the original (fun but quite outdated) nanovna and the v2. The way to communicate with the hardware is quite different (I'm not a software guy). So the v2 in the name was important.

So now, what is the reason to do such long introduction:
- few months ago there was a NanoVNAv2 in the skrf.vi.vna. It was working but only if the hardware was set to give a fast enough answer else there was à timeout has described above.
- today there is no more (from my understanding) a NanoVNAv2 in the skrf.vi.vna, so the problem is not solved (or poorly solved, if you consider that there is no more timeout problem as you can't anymore communicate with the hardware)
- 
So I can add this line
from skrf.vi.vna.NanoVNA import NanoVNA
but this is probably for "original nanovna" not for mine NanoVNA v2
Then I have to say on with COM port, I have to communicate with the device
and I can try:
- nanovna = vna.nanovna('ASRL4::INSTR')
TypeError: 'module' object is not callable
- nanovna = vna.NanoVNA('ASRL4::INSTR')
AttributeError: module 'skrf.vi.vna' has no attribute 'NanoVNA'
- nanovna = NanoVNA('ASRL4::INSTR')
this one seems ok, 
but after,
the command that was ok before doesnt work anymore:
-  nanovna.set_frequency_sweep(10e6, 3e9, 1024)
AttributeError: 'NanoVNA' object has no attribute 'set_frequency_sweep'
I don't know if it is my bad Python skills or just because, I try to communicate with a nanovnav2 with command specific for another device which would be "original" nanovna and that has not the same commands

cafeclimber, 2023-09-25:
Hey thanks for the thorough response! 

So the naming was my mistake in writing the updated driver. I based what I called `NanoVNA` on the previous `NanoVNAv2` implementation so that's something I'll try to fix. 

Now to go through the code examples and explain what's happening:

```python
nanovna = vna.nanovna('ASRL4::INSTR')
TypeError: 'module' object is not callable
```
This occurs because you're calling `vna.nanovna` instead of `vna.nanovna.NanoVNA`. `vna.nanovna` is the module in which the `NanoVNA` class is defined. you need to call `vna.nanovna.NanoVNA(...)`

---

```python
nanovna = vna.NanoVNA('ASRL4::INSTR')
AttributeError: module 'skrf.vi.vna' has no attribute 'NanoVNA'
```

Similar to above, `NanoVNA` is not defined in `vna`, it's defined in `vna.nanovna`, so you need to give the full module path.

---

```python
nanovna.set_frequency_sweep(10e6, 3e9, 1024)
AttributeError: 'NanoVNA' object has no attribute 'set_frequency_sweep'
```

With the overhauled virtual instrument implementation, `set_frequency_sweep` is no longer the way to set the frequency. Instead do something like

```python
from skrf import Frequency
from skrf.vi.vna import nanovna

instr = nanovna.NanoVNA("ASRL4::INSTR")

freq = Frequency(start=10e6, stop=3e9, npoints=1024)
instr.frequency = freq
```

This is very likely to not work, but I appreciate your working through it! Please report any errors you encounter (as verbosely as possible! i.e. every bit of error output that gets printed)

Vinc0110, 2024-01-06:
Has this been fixed by PR #968?