scikit-rf/scikit-rf #903: Vector Network Analyzer Sweep Power Measurement，and  create the network!

AceYoung938, 2023-04-27:
I use a program-controlled VNA to perform power sweep measurements. I set the software to measure S-parameters from 0 to -20dbm at a frequency of 2GHZ, with a total of 201 points. 
  The data format read by the software is an array of real and imaginary parts. Now I want to turn these data into network objects through the Scikit-rf library. How to do it? And these data are uncalibrated, how to calibrate?
  please help me

cafeclimber, 2023-04-27:
Just to make sure I understand, you have already generated your .sNp files?

AceYoung938, 2023-04-27:
No,Only the real part and imaginary part of the S parameter, each power point corresponds to an array.

cafeclimber, 2023-04-27:
Oh sorry, just saw the upload txt file. Looking at that file, it IS indeed a .sNp file. Trying to load it myself, it looks like scikit-rf doesn't currently support Network objects with a power sweep instead of a frequency sweep. 

A possible feature request @jhillairet

AceYoung938, 2023-04-29:
> 

thanks for your answer

jhillairet, 2023-04-30:
> I use a program-controlled VNA to perform power sweep measurements. I set the software to measure S-parameters from 0 to -20dbm at a frequency of 2GHZ, with a total of 201 points.
> The data format [...]

A  `Network` is by definition a frequency-dependant data container as mentionned by @cafeclimber. 

It is however possible to represent a power sweep dataset using a `NetworkSet` parametrized with the power values. Here is an example:

```python
import numpy as np
import matplotlib.pyplot as plt
import skrf as rf

#%% Dummy data representing the real and imaginary parts of S11 for a power sweep
# from 0 to -20dbm at a frequency of 2GHZ, with a total of 201 points.
powers = np.linspace(0, -20, 201)
s11_r = np.random.rand(powers.size)
s11_i = np.random.rand(powers.size)

#%% Creating a NetworkSet containing 201 networks, each containing the S11 at 2GHz.
# Power values are defined using a named parameter 'power'
s11 = s11_r + 1j*s11_i
freq = rf.Frequency(2, 2, npoints=1, unit='GHz')

ntwks = [rf.Network(frequency=freq, s=s, params={'power': power}) for (s, power) in zip(s11, powers)]
ns = rf.NetworkSet(ntwks)

#%% Examples on how to access NetworkSet data
ns.sel({'power': [-5, -10]}) # -> NetworkSet for -5 and -10 dBm

#%% plot s vs power
_s = [n.s_db.squeeze() for n in ns]
_powers = [n.params['power'] for n in ns]

fig, ax = plt.subplots()
ax.plot(_powers, _s)
ax.set_xlabel('power [dBm]')
ax.set_ylabel('s11 [dB]')
``` 

Please refer to the NetworkSet Tutorial and this [NetworkSet example](https://scikit-rf.readthedocs.io/en/latest/examples/networksets/Interpolating%20Network%20Sets.html).

> And these data are uncalibrated, how to calibrate?

Please refer first to the [calibration tutorial](https://scikit-rf.readthedocs.io/en/latest/tutorials/Calibration.html) of the documentation and the [calibration examples](https://scikit-rf.readthedocs.io/en/latest/examples/index.html#metrology-and-calibration). Without additional information on your particular setup, it's hard for us to help us more.

AceYoung938, 2023-04-30:
thank you，I will try your way now

AceYoung938, 2023-04-30:
In #905 ， I created the SOLT calibration standard and got the calibration object 'cal'. As an example, the 'cal' object is created in the 10mhz to 6Ghz frequency range, can you calibrate the S-parameters of '_s' in jhillairet's code ？ @jhillairet

closed 2026-02-01 by jhillairet