scikit-rf/scikit-rf #928: Zoom and Pan features for Smith Chart

Sionwage, 2023-06-29:
It would be nice to define a normalized impedance and zoom left to enlarge a region of the smith chart.

[image]

cafeclimber, 2023-06-29:
@jhillairet if we're still interested in separating out the GUI stuff into a separate repo for 1.0, I think it would make sense to move to pyqtgraph. That would help with something like this as well as e.g. fast updates from a VNA (if it's possible to query data that fast). I did a polar plot implementation for PyChamber and think it wouldn't be too much to build on that for a smith chart

Sionwage, 2023-06-30:
Is there a post or document outlining the envisioned modules to utilize for plotting?

cafeclimber, 2023-06-30:
I don't believe there's any concrete plans about changes to plotting. There is a module called qtapps? that contains widgets that can be added to apps using Qt to make a GUI. I think with moving that module to a separate repo provides the opportunity to consider other plotting libraries.

With your original question, are you able to zoom in using matplotlib's zoom to rectangle button?

Sionwage, 2023-06-30:
Actually I didn't know about ipympl but that works perfectly to move and zoom. 

I just installed it and added "%matplotlib ipympl" to the top of the jupyter lab cell I was running.

I do need to figure out how to enhance the axis density on zooming in for Smith Chart plots.

jhillairet, 2023-07-01:
> @jhillairet if we're still interested in separating out the GUI stuff into a separate repo for 1.0, I think it would make sense to move to pyqtgraph. That would help with something like this as well [...]

Eventually, letting users the possibility to use a graphics backend or another.