scikit-rf/scikit-rf #951: Probably a typo in time.py

pigskinsoup, 2023-08-30:
Not 100% sure about this issue, because I'm not a coding person and didn't check the code. If someone can confirm with me about this small issue in the code would be wonderful.  Or correct me if I'm wrong.

**Describe the bug**
This typo is in the description of time-gating method fft. The description below indicated that data gets transformed into time-domain using inverse FFT and the time-domain signal is then transformed back into frequency-domain using inverse FFT too. Which probably a typo because transform into frequency-domain is using FFT right?

Below is the description about the fft method in time.py
        With *'fft'* (default), the data gets transformed into time-domain using inverse FFT and the gating is achieved
        by multiplication with the time-domain gate. The gated time-domain signal is then transformed back into
        frequency-domain using inverse FFT. As only positive signal frequencies are considered for the inverse FFT
        (with or without a dc component), the resulting time-domain signal has the same number of samples as in the
        frequency-domain, but is complex-valued. This method is also know as *time-domain band-pass mode*.

**To Reproduce**
just remove "inverse" in the process of time-domain transformed back into frequency-domain. If someone can confirm the code with me would be great. Or correct me if I'm wrong.

**Expected behavior**
just remove "inverse" in the process of time-domain transformed back into frequency-domain. If someone can confirm the code with me would be great. Or correct me if I'm wrong.

**Data**

**Screenshots**

**System**
 - OS:
 - scikit-rf version

**Additional context**

Vinc0110, 2023-09-01:
Yes, you are right. We should fix this in the documentation. The code should be fine, though. Thanks for reporting!

jhillairet, 2023-09-26:
> Yes, you are right. We should fix this in the documentation. The code should be fine, though. Thanks for reporting!

@Vinc0110 do you intend to fix the doc, or should I do it?

Vinc0110, 2023-09-26:
Done. Thanks for the reminder ;)

commit 2f01000e (2023-09-26): Fix #951
commit 9cf77678 (2023-09-26): Merge pull request #962 from Vinc0110/master

Fix #951
PR #963 (2023-09-26): v0.29.1
Minor release, mostly bug and documentation fixes.

## What's Changed
* Fix network_2_dataframe for more than 10 ports by @FranzForstmayr in https://github.com/scikit-rf/scikit-rf/pull/955
* Check python 3.12 compability by @FranzForstmayr in https://github.com/scikit-rf/scikit-rf/pull/956
* Fix plotting notebook for matplotlib 3.8 by @FranzForstmayr in https://github.com/scikit-rf/scikit-rf/pull/959
* Fix VI documentation issue by @cafeclimber in https://github.com/scikit-rf/scikit-rf/pull/954
* Add timeout argument to VNA constructor by @cafeclimber in https://github.com/scikit-rf/sc [... cut]
closed 2023-09-26 by jhillairet