scikit-rf/scikit-rf #960: I am still seeing this issue : https://github.com/scikit-rf/scikit-rf/issues/207

JG1968, 2023-09-25:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**
I am using skit rf 0.29.0 in vs code. I am experiencing this old issue https://github.com/scikit-rf/scikit-rf/issues/207 again. What am I doing wrong.
**To Reproduce**
Steps to reproduce the behavior:

plot_s.d_time produces x axis is in Frequency in time domain plot
A clear and concise description of what you expected to happen.

**Data**
If applicable, add the data leading to the issue (a minimum example is appreciated)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**System**
 - OS: windows10
 - scikit-rf version 0.29.0

**Additional context**
Add any other context about the problem here if relevant.

FranzForstmayr, 2023-09-25:
Obviously regression after #848. Will try to fix that again!

closed 2023-09-26 by jhillairet

JG1968, 2023-09-26:
Thanks a lot. Really appreciate your work.

PR #961 (2023-09-26): Use correct x-axis for time_mag and time_db plots
Should fix #960.