scikit-rf/scikit-rf #964: Required keyword argument names

FranzForstmayr, 2023-09-26:
Scipy recently started to deprecate positional arguments for non obvious keyword arguments, so a user can only pass these argument with the explicit keyword. 

https://docs.scipy.org/doc/scipy/dev/missing-bits.html#required-keyword-names

This makes it easier to change functionality of functions or methods in a backward compatible way, for example if a new argument is added.

This is mostly beneficial for functions with lot of arguments which have no meaning to the reader if passed positional like:

A few examples:

https://github.com/scikit-rf/scikit-rf/blob/master/skrf/network.py#L4017
```python
def impulse_response(
        self, 
        window: str = 'hamming', 
        n: int = None, 
        pad: int = 1000,
        bandpass: bool = None, 
        squeeze: bool = True) -> 
    Tuple[npy.ndarray, npy.ndarray]:
```

```python
def time_gate(ntwk: 'Network', start: float = None, stop: float = None, center: float = None, span: float = None,
              mode: str = 'bandpass', window=('kaiser', 6),
              method: str ='fft', fft_window: str='cosine', conv_mode: str='wrap', t_unit: str = "") -> 'Network':
```

The first one would look like:

```python
def impulse_response(
        self, 
        *,
        window: str = 'hamming', 
        n: int = None, 
        pad: int = 1000,
        bandpass: bool = None, 
        squeeze: bool = True) -> 
    Tuple[npy.ndarray, npy.ndarray]:
```

The `window` could also stay positional if wanted, but the remaining ones should defenitely be keyword arguments.

This is a breaking change for most users of course, so it would need a deprecation cycle and a warning if still passed positional.
After one or two releases the function could only accept the keyword arguments.

jhillairet, 2023-09-27:
I understand this requirement in scipy is for *new* functions (otherwise it will break a huge number of code). I guess it's same for us: I agree this is a good coding practice, but I am more reserved about the usefulness of changing existing code.

FranzForstmayr, 2023-09-27:
Yes, sometimes they execute this change if any change has to be done anyway.