scikit-rf/scikit-rf #967: z2s with Complex Reference Impedance

spratsch, 2023-10-16:
When generating a network through z parameters, the networks s parameters are incorrect when a complex valued reference impedance is used.
Please note that a perfect conjugate match is employed in this example, which should lead to no return loss (S parameter magnitude equals to 0 or -inf dB)

MWE (wrong result)
```
import skrf as rf
import numpy as np

reference_impedance = 50+1j*50
frequency = 1e9
load_impedance = np.ones((1, 1, 1)) * 50-1j*50  # conjugate match

print(f'Reference impedance is {reference_impedance}')
print(f'Load impedance is {load_impedance}')

ntwk = rf.Network(frequency=frequency, z=load_impedance, name='my 1 port')
print(f'SKRF: Return loss is {10*np.log10(ntwk.s_mag[0, 0, 0])}(dB magnitude)')
```

When calculating the S parameter using skrf.network.z2s, the calculated S parameter is acutally correct.
MWE2 (correct result):
```
import skrf as rf
import numpy as np

reference_impedance = 50+1j*50
frequency = 1e9
load_impedance = np.ones((1, 1, 1)) * 50-1j*50  # conjugate match

print(f'Reference impedance is {reference_impedance}')
print(f'Load impedance is {load_impedance}')

rl2 = rf.network.z2s(load_impedance, reference_impedance)
print(f'SKRF: Return loss is {10 * np.log10(np.abs(rl2))}(dB magnitude)')
```

 - OS: Windows 11, Anaconda 4.12.0, Python 3.9.12
 - scikit-rf version 0.24.1

Ttl, 2023-10-16:
`reference_impedance` is not used in the first case. The generated network will have port impedance of 50 ohms by default and the result is correct for that case. To match the second case you need to also pass `z0=reference_impedance` to the network initializer.

spratsch, 2023-10-16:
Wow, I totally missed that. This resolves my issue. There is acutally no bug.

closed 2023-10-16 by spratsch