scikit-rf/scikit-rf #969: One possible idea to speed up the C attribute of Circuit class

Asachoo, 2023-10-30:
Although the performance of circuit computing S-parameters has been significantly improved, I still want to try whether it is possible to further enhance the calculation performance.
Related to https://github.com/scikit-rf/scikit-rf/issues/574 and https://github.com/scikit-rf/scikit-rf/pull/575
I noticed that the performance of the X property of Circuit class has been improved, so focused on the C property of Circuit class.

In the current [circuit.py](http://circuit.py/), there is a section of for loop code within the C property. Due to the use of the product function, the loop is executed a large number of times. Therefore, I am wondering if it is possible to accelerate it using numpy's broadcasting feature.

Existing code:
```
for (ntw_name, ntw_ports) in ntws_ports_reordering.items():
    # get the port re-ordering indexes (from -> to)
    ntw_ports = np.array(ntw_ports)
    # create the port permutations
    from_port = list(product(ntw_ports[:,0], repeat=2))
    to_port = list(product(ntw_ports[:,1], repeat=2))

    #print(ntw_name, from_port, to_port)
    for (_from, _to) in zip(from_port, to_port):
        #print(f'{_from} --> {_to}')
        S[:, _to[0], _to[1]] = ntws[ntw_name].s_traveling[:, _from[0], _from[1]]

return S  # shape (nb_frequency, nb_inter*nb_n, nb_inter*nb_n)
```

Improved code:
```
for (ntw_name, ntw_ports) in ntws_ports_reordering.items():
    # get the port re-ordering indexes (from -> to)
    ntw_ports = np.array(ntw_ports)
    # create the port permutations
    from_port = ntw_ports[:,0]
    to_port = ntw_ports[:,1]

    #print(ntw_name, from_port, to_port)
    for (_from, _to) in zip(from_port, to_port):
        #print(f'{_from} --> {_to}')
        S[:, _to, to_port] = ntws[ntw_name].s_traveling[:, _from, from_port]

return S  # shape (nb_frequency, nb_inter*nb_n, nb_inter*nb_n)
```

However, it was found that only when the network in the circuit is connected internally, there is an efficiency improvement of about 10%. Therefore, I would like to know if this acceleration method is truly effective and if it may result in unexpected errors.

jhillairet, 2023-11-03:
I don't remember why the `product` was used. Are the tests passing with your modification?

Asachoo, 2023-11-06:
> I don't remember why the `product` was used. Are the tests passing with your modification?

```
pytest test_network.py
```
I'm not sure if my testing method is correct, but when I executed the code above, this was the result I obtained (I believe all tests have passed).
```
=========================================================================================================== test session starts ============================================================================================================
platform linux -- Python 3.8.16, pytest-7.1.2, pluggy-1.0.0
rootdir: /home/lansus/WorkFile/Git/scikit-rf, configfile: pyproject.toml
plugins: cov-4.1.0, anyio-3.6.1
collected 107 items                                                                                                                                                                                                                        

test_network.py .......s...................................................................................................                                                                                                          [100%]

---------- coverage: platform linux, python 3.8.16-final-0 -----------
Name                                                                                                 Stmts   Miss  Cover
------------------------------------------------------------------------------------------------------------------------
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/__init__.py                        67      9    87%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/calibration/__init__.py             6      0   100%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/calibration/calibration.py       2680   2413    10%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/calibration/calibrationSet.py      52     36    31%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/calibration/deembedding.py        851    777     9%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/circuit.py                        344    262    24%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/constants.py                       25      2    92%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/data/__init__.py                   64     14    78%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/frequency.py                      223     69    69%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/instances.py                      235     76    68%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/io/__init__.py                      5      0   100%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/io/citi.py                        119    100    16%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/io/csv.py                         326    288    12%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/io/general.py                     246    198    20%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/io/mdif.py                        194    169    13%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/io/metas.py                        23     23     0%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/io/touchstone.py                  283    103    64%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/mathFunctions.py                  211     71    66%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/__init__.py                   9      0   100%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/circularWaveguide.py         97     54    44%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/coaxial.py                   92     61    34%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/cpw.py                      124     39    69%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/definedAEpTandZ0.py          63     36    43%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/device.py                    98     98     0%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/distributedCircuit.py        53     29    45%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/freespace.py                 75     25    67%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/media.py                    395    205    48%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/mline.py                    223    195    13%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/media/rectangularWaveguide.py     104     56    46%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/network2.py                       462    462     0%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/network.py                       1978    505    74%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/networkSet.py                     379    225    41%
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/notebook/__init__.py                0      0   100%
[... cut ...]
/home/lansus/Applications/anaconda3/lib/python3.8/site-packages/skrf/vi/vna/vna.py                     180    180     0%
------------------------------------------------------------------------------------------------------------------------
TOTAL                                                                                                13758   9740    29%

====================================================================================================== 106 passed, 1 skipped in 4.36s ======================================================================================================
```

jhillairet, 2024-01-27:
Dear @Asachoo ,

Sorry for the delay! Your suggestion is perfect. I've implemented it in PR #1013. Many thanks!