scikit-rf/scikit-rf #972: darwin aarch64 skrf/calibration/tests/test_calibration.py failing with devide by zero (determinate calcualations)

robert-manchester, 2023-11-09:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**
updating to scikit-rf version 0.29.1 PR https://github.com/NixOS/nixpkgs/pull/266389 fails in the unit `skrf/calibration/tests/test_calibration.py` fails with divide by zero errors.

[build failure](https://github.com/scikit-rf/scikit-rf/files/13312763/unitfail.txt)

one failure pasted inline.
```
=================================== FAILURES ===================================
__________ DetermineTest.test_determine_reflect_matched_thru_and_line __________
self = <test_calibration.DetermineTest testMethod=test_determine_reflect_matched_thru_and_line>
    def test_determine_reflect_matched_thru_and_line(self):
        freq = rf.F(.25, .7, 40, unit = 'GHz')
        medium = Coaxial.from_attenuation_VF(freq, att = 3.0, VF = .69)
    
        thru = medium.line(0, 'm')
        line = medium.line(0.12, 'm')
        short = medium.short()
    
        rng = npy.random.default_rng(12)
        short.s[:, 0, 0] = short.s[:, 0, 0]+rng.uniform(-.02, 0.02, freq.f.size) + rng.uniform(-.02, 0.02, freq.f.size)*1j
    
>       r = determine_reflect(thru, rf.two_port_reflect(short, short), line)
skrf/calibration/tests/test_calibration.py:101: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
skrf/calibration/calibration.py:6406: in determine_reflect
    d = -det(thru_m.s)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
a = array([[[5.e-08+0.j, 1.e+00+0.j],
        [1.e+00+0.j, 5.e-08+0.j]],
       [[5.e-08+0.j, 1.e+00+0.j],
        [1.e+0... 1.e+00+0.j],
        [1.e+00+0.j, 5.e-08+0.j]],
       [[5.e-08+0.j, 1.e+00+0.j],
        [1.e+00+0.j, 5.e-08+0.j]]])
    @array_function_dispatch(_unary_dispatcher)
    def det(a):
        """
        Compute the determinant of an array.
    
        Parameters
        ----------
        a : (..., M, M) array_like
            Input array to compute determinants for.
    
        Returns
        -------
        det : (...) array_like
            Determinant of `a`.
    
        See Also
        --------
        slogdet : Another way to represent the determinant, more suitable
          for large matrices where underflow/overflow may occur.
        scipy.linalg.det : Similar function in SciPy.
    
        Notes
        -----
    
        .. versionadded:: 1.8.0
    
        Broadcasting rules apply, see the `numpy.linalg` documentation for
        details.
    
        The determinant is computed via LU factorization using the LAPACK
        routine ``z/dgetrf``.
    
        Examples
        --------
        The determinant of a 2-D array [[a, b], [c, d]] is ad - bc:
    
        >>> a = np.array([[1, 2], [3, 4]])
        >>> np.linalg.det(a)
        -2.0 # may vary
    
        Computing determinants for a stack of matrices:
    
        >>> a = np.array([ [[1, 2], [3, 4]], [[1, 2], [2, 1]], [[1, 3], [3, 1]] ])
        >>> a.shape
        (3, 2, 2)
        >>> np.linalg.det(a)
        array([-2., -3., -8.])
    
        """
        a = asarray(a)
        _assert_stacked_2d(a)
        _assert_stacked_square(a)
        t, result_t = _commonType(a)
        signature = 'D->D' if isComplexType(t) else 'd->d'
>       r = _umath_linalg.det(a, signature=signature)
E       RuntimeWarning: divide by zero encountered in det
```

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Data**
If applicable, add the data leading to the issue (a minimum example is appreciated)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - OS: macos aarch64
 - scikit-rf version 0.29.1

**Additional context**
Add any other context about the problem here if relevant.

robert-manchester, 2023-11-10:
on macos simplified code produces a divide by zero warning.
```py
#!/usr/bin/env python3
import numpy as np
print(np.__version__)
a = np.array([[5.e-8+0.j, 1+0.j], [1+0.j, 5.e-8+0.j]])
print(a)
print(np.linalg.det(a))
```
```console
./test.py
1.25.2
[[5.e-08+0.j 1.e+00+0.j]
 [1.e+00+0.j 5.e-08+0.j]]
/nix/store/rpdx49knxarbr4lmkwfi5l0461lyp6bd-python3-3.11.6-env/lib/python3.11/site-packages/numpy/linalg/linalg.py:2180: RuntimeWarning: divide by zero encountered in det
  r = _umath_linalg.det(a, signature=signature)
/nix/store/rpdx49knxarbr4lmkwfi5l0461lyp6bd-python3-3.11.6-env/lib/python3.11/site-packages/numpy/linalg/linalg.py:2180: RuntimeWarning: invalid value encountered in det
  r = _umath_linalg.det(a, signature=signature)
```

pbsds, 2023-11-10:
note to future self: re-enable the failing test in nixpkgs if a fix is found

FranzForstmayr, 2023-11-13:
This should be reported to https://github.com/numpy/numpy itself, but this probably related https://github.com/numpy/numpy/issues/22025.

FranzForstmayr, 2023-11-13:
Nevertheless we could modify the unit test slightly to avoid this issue. I'm afraid we are not able to find the root cause here.

emilazy, 2025-08-22:
This same test is now failing in a different manner on Linux for us, after updates to other packages: https://hydra.nixos.org/build/303689719/nixlog/1

biergaizi, 2025-09-26:
> @emilazy: This same test is now failing in a different manner

Please open a new issue. This is not the "same test" and is unrelated to the original problem, the file `test_calibration.py` contains over 600 subtests.

closed 2025-09-26 by jhillairet