scikit-rf/scikit-rf #974: How to perform port cropping?

Asachoo, 2023-11-15:
When we read a TouchStone file, its S parameters and port number will be determined by the TouchStone file.

We can change the order of ports (renumber) without changing the number of ports through the _network.renumber_ method, but when I want to obtain a network with **only some ports** (submatrix) by cutting, is there a build-in method to achieve it?

For example, read an s4p file, and then select ports 1 and 3 to splice it into a new s2p file.

Consider that the _network.renumber_ method is performed in-place and is called widely. So I wonder if we have to come up with a new method or introduce some additional mechanisms in the _network.renumbered_ method to implement the port cropping?

Asachoo, 2023-11-15:
An ugly attempt to achieve this

```
def renumbered(self, from_ports: Sequence[int], to_ports: Sequence[int] = None) -> 'Network':
    """
    Return a renumbered Network, leave self alone.

    Parameters
    ----------
    from_ports : list-like
        List of port indices to change. Size between 1 and N_ports.
    to_ports : can be None or list-like
        List of desired port indices. Size between 1 and N_ports.
        None to use automatically generate a List of desired port indices. Size is between 1 and from_ports' size.

    NB: from_ports and to_ports must have same size.

    Returns
    -------
    ntwk : :class:`Network` object
        Resulting renumbered Network

    See Also
    --------
    renumber
    flip
    flipped

    """
    out = self.copy()
    if to_ports:
        out.renumber(from_ports, to_ports)
    else:
        to_ports_idx = npy.ix_(npy.arange(out.s.shape[0], from_ports, from_ports)
        out.s = out.s[to_ports_idx]
        out.z0 = out.z0[:, from_ports]
    return out
        
```

denzchoe, 2023-11-15:
Will this help ? I think there is already a built-in function already for your request.

https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.subnetwork.html#skrf.network.Network.subnetwork

<br>

Also, should this be in the `Discussion` thread ?

<br>

https://github.com/scikit-rf/scikit-rf/blob/b32ff9c86e8829f43c71efe706c6db1c042e8424/skrf/network.py#L5442-L5480

Asachoo, 2023-11-15:
> Will this help ? I think there is already a built-in function for your request.
> 
> https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.subnetwork.html#skrf.network.Network.subnetwork
> 
> 
> Also, should this be in the `Discussion` thread rather than `Issues` ? 🤔 We can always convert _Discussions_ to _Issues_ once it is confirmed
> 
> 
> https://github.com/scikit-rf/scikit-rf/blob/b32ff9c86e8829f43c71efe706c6db1c042e8424/skrf/network.py#L5442-L5480

Thanks for your tip, your reply solved my problem very well

closed 2023-11-15 by Asachoo