scikit-rf/scikit-rf #977: Relation of Z-Parameters to Y-Parameters: To be, or not to be

zolabar, 2023-11-23:
**Describe the bug**

Y-Parameters and Z-Parameters of networks are not inverse to each other, and still "exist".

If Y-Parameters and Z-Parameters of networks are not inverse to each other, one of the two should not exist. 

**To Reproduce**

`>>> nw = rf.Network('designer_capacitor_30_80MHz_simple.s2p')`

`>>>nw.z[0]`

array([[3.06987185e+10-1046.28673594j, 3.06987185e+10-1010.65464887j],
       [3.06987185e+10-1010.65464367j, 3.06987185e+10-1046.28673071j]])
       
`>>>nw.y[0]`

array([[-1.73472348e-18+0.0140323j,  0.00000000e+00-0.0140323j],
       [ 8.67361738e-19-0.0140323j, -1.73472348e-18+0.0140323j]])

But if Y exsits, it should be equal to Z, and vise versa

$$Z=Y^{⁻1}$$

$$Y=Z^{⁻1}$$

I understand, that do to the modelling, one of the two may exist and the other not, if the user constructs such a network.
But then, scikit-rf should warn about that or don't compute the non existent parameter matrix.

Instead of giving Z and Y out, without them beeing inverse to each other:

`np.linalg.inv(nw.z)[0]`

array([[ 7.51134248e-10+0.0140323j, -7.34846925e-10-0.0140323j],
       [-7.34846923e-10-0.0140323j,  7.51134250e-10+0.0140323j]]) 

`np.linalg.inv(nw.y)[0]`

---------------------------------------------------------------------------
LinAlgError                               Traceback (most recent call last)
Cell In[16], line 1
----> 1 np.linalg.inv(nw.y)[0]

File ~/PROJECTS/Create_PDN_Lumped_model/PY/pdn_fit/.venv/lib/python3.11/site-packages/numpy/linalg/linalg.py:561, in inv(a)
    559 signature = 'D->D' if isComplexType(t) else 'd->d'
    560 extobj = get_linalg_error_extobj(_raise_linalgerror_singular)
--> 561 ainv = _umath_linalg.inv(a, signature=signature, extobj=extobj)
    562 return wrap(ainv.astype(result_t, copy=False))

File ~/PROJECTS/Create_PDN_Lumped_model/PY/pdn_fit/.venv/lib/python3.11/site-packages/numpy/linalg/linalg.py:112, in _raise_linalgerror_singular(err, flag)
    111 def _raise_linalgerror_singular(err, flag):
--> 112     raise LinAlgError("Singular matrix")

LinAlgError: Singular matrix

I think that here, Z does not exist.

This could be checked internally with the powerful singular value decompostion based `np.linalg.matrix_rank` function.

**Expected behavior**

A rf.Network should give out warnings when its Y-Parameters and Z-Parameters are not inverse to each other.
Alternatively, the non-existent parameters could not be given out, and instead, an error could be generated.

**Data**
The scikit-rf project example data designer_capacitor_30_80MHz_simple.s2p was taken.

**Screenshots**

[image]

**System**
 - OS: Linux
 - scikit-rf version: 0.29.1

Vinc0110, 2023-11-24:
Not sure if these are just typos, but you made contrary statements about Z and Y parameters being inverse to each other (or not).
To be clear, they are theoretically inverse:
>Conversion from [Y-parameters](https://en.wikipedia.org/wiki/Y-parameters) to Z-parameters is much simpler, as the Z-parameter matrix is just the [inverse](https://en.wikipedia.org/wiki/Matrix_inverse) of the Y-parameter matrix.
https://en.wikipedia.org/wiki/Impedance_parameters#Relation_to_Y-parameters

But you are right, the inversion is problematic in some cases, such as a short circuit (Z11=0), or an open circuit (Y11=0). One way around that is the usage of scattering parameters, as done in case of `doc/source/examples/circuit/designer_capacitor_30_80MHz_simple.s2p`.

In scikit-rf, the conversion between different types of network parameters is done with various dedicated functions, which also handle problematic cases:
- `s2z()`: https://github.com/scikit-rf/scikit-rf/blob/b32ff9c86e8829f43c71efe706c6db1c042e8424/skrf/network.py#L5838
- `s2y()`: https://github.com/scikit-rf/scikit-rf/blob/b32ff9c86e8829f43c71efe706c6db1c042e8424/skrf/network.py#L5930
- ... many more

For example, matrix singularities are removed by "nudging" the matrix before the inversion:
https://github.com/scikit-rf/scikit-rf/blob/b32ff9c86e8829f43c71efe706c6db1c042e8424/skrf/network.py#L5908

See `mathFunctions.nudge_eig()` for details:
https://github.com/scikit-rf/scikit-rf/blob/b32ff9c86e8829f43c71efe706c6db1c042e8424/skrf/mathFunctions.py#L1284

This is why scikit-rf is still able to provide Z parameters for the problematic capacitor example. I don't think this is incorrect.

zolabar, 2023-11-24:
@Vinc0110  thank you for the fast and detailled answer.

Which typo do you mean?

$$Z=Y^{⁻1}$$

$$Y=Z^{⁻1}$$

is what I see in wikpedia

[image]

and since inversion (if it exists) goes both ways that should be  OK.

I understand, that the "expected behaviour" is not clear.

But since inversion is symmetric, i.e.  $A=B^{-1}$ is equivalent to $A^{-1}=B$ one could expect a warning/error message from scikit-rf, too, if that is not case for Y and Z.

But this reminds me on discussion about "zero" in sympy, it depends on each user what he would expect ;)

Vinc0110, 2023-11-24:
Maybe I just misinterpreted your first sentence:
> Y-Parameters and Z-Parameters of networks are not inverse to each other, and still "exist".

Without context, that's just wrong, as you stated yourself. 

But I guess you are referring to special cases where they *should* be invertible, but the inverse cannot be calculated exactly (without nudging or other tricks). However, the values calculated in scikit-rf (with the help of nudging) should be very close to the exact values. 

Your own example is showing it:
The difference between `nw.y[0]` and `np.linalg.inv(nw.z[0])` is very small. In both cases, we basically get `0 + 0.0140323j` or `0 - 0.0140323j` for the respective parameters (the real parts are slightly different, but basically zero).

I'm still not sure why you want a warning. Are you worried about those tiny differences?

zolabar, 2023-11-24:
> Maybe I just misinterpreted your first sentence:
>
>
> Without context, that's just wrong, as you stated yourself.
>
> But I guess you are referring to special cases where they _should_ be invertible, but [...]

you are right, in my example (which is from scikit-rf) $Y \approx Z^{-1}$.  What made me write the comment was, that then I would expect also $Z \approx Y^{-1}$, but $Y^{-1}$ (see the singular matrix error message) did not exist.

Which from basic circuit theory and from the mathematical point of view is difficult to understand ;)

And since one can see such contradictory cases a priori with the help of np.linalg.matrix_rank I thought that warnings can be issued.

In the example
`np.linalg.matrix_rank(nw.z[0])` gave 2 and `np.linalg.matrix_rank(nw.y[0])` gave 1 which already tells us that  $Y^{-1}$ will lead to an error and that there is "problem",  which is confirmed by the fact that $Y^{-1}$ is not "at all" close to Z (`nw.z[0]` is not close to `rf.y2z(nw.y[0])`).

I could imagine, for instance, that scikit-rf computes the rank of Y, Z each time one uses on of both and gives a warning when the rank is not full, so the user can be aware of the risky terrain he is on.

This was just something that came to my head when I saw the non symmetry of inversions. It might not be of interest for the core user community.

Vinc0110, 2023-11-24:
Now I understand. I was thinking this is only about `np.linalg.inv()` failing, but we also get an error from `network.y2z()` despite the nudging.

Summary with 
`nw = skrf.Network('designer_capacitor_30_80MHz_simple.s2p')`:
- This does work: `skrf.network.z2y(nw.z)`
- This does **not** work:  `skrf.network.y2z(nw.y)`
- This does work: `skrf.network.s2y(skrf.network.z2s(nw.z))`
- This does work: `skrf.network.s2z(skrf.network.y2s(nw.y))`

I agree with your proposal to check the matrix rank before the conversions in `y2z()` and `z2y()`. But as you wrote, this is probably only an issue for power users. Usually, the network gets created from S parameters and is then converted to Z or Y.

zolabar, 2023-11-24:
👍🏽