scikit-rf/scikit-rf #980: scikit-rf port to .Net C#. Help Needed!

Zinc999, 2023-11-28:
Hi ,

I have a UseCase where I would like to use scikit-rf functionality but... python is not allowed on the system. 

Hence, looking out for any ongoing effort where sckit-rf port to .Net C# is planned or ongoing.

If not, I would like to do that. To get started I have following queries:
1. Is it legally okay to start a port?
2. If yes then can we create a repo here under scikit-rf umbrella?
3. If yes for the above, then who all are interested in helping me :)

Looking out for some helping hands. Thanks in advance!

arsenovic, 2023-12-03:
> Is it legally okay to start a port?
> The project is BSD licensed, which means you can do basically everything
> you like with the code.
>
> If yes then can we create a repo here under scikit-rf umbrella?
> I don't know how the users in this group can help in C#, so probably
> starting your own repo outside skrf would be the easiest option. The
> project could be added afterwards anyway.
>
> If yes for the above, then who all are interested in helping me :)
> I'm out as I'm not familiar with C#

just tell chatgpt to do it.

jhillairet, 2024-01-23:
I'm closing for now, as a port in C# is not foreseen. Anyway, anybody can use the code in this repo for making a C# version of scikit-rf.

closed 2024-01-23 by jhillairet