scikit-rf/scikit-rf #1001: vector fit crashes when npoles > npoints/2

PinkMushroom, 2024-01-19:
**Describe the bug**
if your data set has npoints, then when you specify more than npoints/2 complex poles, vector fit crashes

**To Reproduce**
            vf1.vector_fit(n_poles_real=1, n_poles_cmplx=ncomplex)
where ncomplex is >= number of data points /2

has nothing to do with the data itself

**Expected behavior**
it shouldn't crash

**Screenshots**

In this particular example npoints=21 and number of poles = 10

  File "/home/user/.local/lib/python3.8/site-packages/skrf/vectorFitting.py", line 366, in vector_fit
    R[i] = np.linalg.qr(A_ri[i], mode='r')
ValueError: could not broadcast input array from shape (42,44) into shape (44,44)

**System**
 - OS:
 - ubuntu
 - scikit-rf version
 - latest

**Additional context**
Really strange nobody has run into this problem yet...

Vinc0110, 2024-01-19:
Thanks for reporting this. Of course, this case should be caught and the code should not crash.

I guess the reason nobody has run into this is that this setup does not make much sense. Each complex conjugate pole introduces 2 unknowns to solve for. So, for `2 * n_poles_cmplx > npoints`, the system is underdetermined.

PinkMushroom, 2024-01-19:
yse, but the samples are complex, so it should all work out.

Intuitively shouldn't the order of my polynomial be valid up to the number of samples points ?

If were to use something like Levy/S-K (the original Levy/S-K) I'm pretty sure you could compute a polynomial with an order equal to the number of sample points.

I don't have access to my Levy code right now, but I'll give that a try when I do.

Unless you convince me I'm wrong. and then I'll still give it a try ;-)

Vinc0110, 2024-01-20:
> Intuitively shouldn't the order of my polynomial be valid up to the number of samples points ?

Your intuition might the right. It also does not work if I add multiple frequency responses with `2 * n_poles_cmplx` samples each. Then, the system is definitely not underdetermined anymore. 

This seems to be a real bug in the code. I'll check it in the next days.

closed 2024-01-23 by Vinc0110