scikit-rf/scikit-rf #1011: Analytical Matrix Inversion in subnetwork growth (Filipsson/Compton) vs General Numerical Inversion following Hallbjörner

zolabar, 2024-01-25:
The Circuit object is very flexible und thanks to networkx one can see what one builts!

I have a question and since I need some space and formulas I choose this medium.

I saw that scikit.rf Network objects can be connnected effciently using the connect method, wich uses the analytical connection formulae from Filipsson/Compton.

Wouldn't it be possible to use this connection idea when assembling general circuits?

Instead of using the general matrix inversion proposed by Hallbjörner

$$\boldsymbol{S}=\boldsymbol{X}\left(\boldsymbol{I}-\boldsymbol{C}\boldsymbol{X}\right)^{-1}$$

It seems to me, that the formulae in Filipsson/Compton describe explicitly how to connect s parameters without having to solve any systems of equations, they did that implicitely already.

Is the problem and motiviation for Hallbjörners formuale in the circuit assembly that the networks (of the lumped elements) that would need to be connected in a circuit would have to found first?

If so, I think it would be possible to analytically build 1-Port S-Parameter networks at least for resistors, capacitances, inductivities, SeriesImpedance... and then connect them with Filipsson/Compton's results.

I think, that avoiding the numerical inversion of large matrices can save time, in particular, when optimizations or sensitivity analysis are carried out.

Maybe I missed the crucial point, that motivates the usage of Hallbjörner's numerical matrix innversions.

Compton, R.C.; , “Perspectives in microwave circuit analysis,” Circuits and Systems, 1989., Proceedings of the 32nd Midwest Symposium on , vol., no., pp.716-718 vol.2, 14-16 Aug 1989. URL: http://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=101955&isnumber=3167

Filipsson, Gunnar; , “A New General Computer Algorithm for S-Matrix Calculation of Interconnected Multiports,” Microwave Conference, 1981. 11th European , vol., no., pp.700-704, 7-11 Sept. 1981. URL: http://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=4131699&isnumber=4131585

Hallbjörner, Paul. "Method for calculating the scattering matrix of arbitrary microwave networks giving both internal and external scattering." Microwave and Optical Technology Letters 38.2 (2003): 99-102.
https://onlinelibrary.wiley.com/doi/abs/10.1002/mop.10983

jhillairet, 2024-01-26:
> scikit.rf Network objects can be connnected effciently using the connect method, wich uses the analytical connection formulae from Filipsson/Compton.\n\nWouldn't it be possible to use this connection idea when assembling general circuits?

Maybe. To be honest I did not try. Using only `connect` function is not that simple when connecting N-ports with N>2, which leads me to the Hallbjörner method. Its implementation was relatively straightforward from the paper. 

One other great advantage is that this method allows a direct access to the internal S parameters, and so, to internal voltages and currents at all nodes.

arsenovic, 2024-01-28:
i have used sympy (symbolic  library) and the cascade methods to do as you
describe  in the past. it worked great.

arsenovic, 2024-01-28:
also i just saw your conformal maps library. looks cool!
i have in the past tried to visualize how a two-port warps the smithchart
over frequency, as an animation of conformal maps. but i did it
numerically. the idea was to try and get a feel for what the 'error
networks' in the calibrations were doing.

zolabar, 2024-01-29:
> Also 2-Port network S-Parameters could be obtained using ABCD and then a2s, I think. And then, the 1-Port and 2-Port networks could be connected via Filipsson/Compton with each other and any existing multiport network.

@jhillairet , thank you for the quick response. I think I see the practical part of the general formula. I just started thinking about this topic, because of speed of the circuit building (I think https://github.com/scikit-rf/scikit-rf/issues/969 also deals with that) ;)

zolabar, 2024-01-29:
> i have used sympy (symbolic library) and the cascade methods to do as you describe in the past. it worked great.
> […](#)

@arsenovic , thank you for the quick reply. That experiment with sympy sounds great. I forgot to write in the original potst (updated now) that besides 1-Ports networks, also 2-Port network S-Parameters could be obtained using ABCD and then a2s, I think. And then, the 1-Port and 2-Port networks could be connected via Filipsson/Compton with each other and any existing multiport network. I think, that circuit networks assembled in this way could have a faster assembly than the general Hallbjörner version. Of course, I don't know if there are exceptions to my hypothesis ;)

zolabar, 2024-01-29:
> also i just saw your conformal maps library. looks cool! i have in the past tried to visualize how a two-port warps the smithchart over frequency, as an animation of conformal maps. but i did it [...]

@arsenovic thank you for the feedback on the conformal mapping project. The application with respect to the smith chart sound interesting (the smith chart as a Möbius Transform ;)). Until now, I had mainly fluid dynamic and mechanical applications of confromal mappings in mind.
But rf theory and practice offer many more applications of these!

Asachoo, 2024-06-15:
Perhaps these two PRs https://github.com/scikit-rf/scikit-rf/pull/1069 and https://github.com/scikit-rf/scikit-rf/pull/1080 have somewhat alleviated your concerns regarding the speed of circuit building in the general Hallbjörner version.
By connecting and merging series `Network`s in the circuit connections using `rf.connect()` method, and then calculated the reduced circuit using Hallbjörner's method (retaining non-series `Network`s and `Ports`). These changes resulted in significant performance improvements in my testing..
I hope you find these updates beneficial ;)

zolabar, 2024-06-17:
> Perhaps these two PRs #1069 and #1080 have somewhat alleviated your concerns regarding the speed of circuit building in the general Hallbjörner version. By connecting and merging series `Network`s [...]

@Asachoo, thank you for the update!  In my next sensitivity analysis I could use this improvement :)

closed 2026-02-01 by jhillairet