scikit-rf/scikit-rf #1014: De-embedding procedure help request

shabaz123, 2024-01-28:
Hello,

I am using the latest 0.31.0 version of scikit-rf (but I have also tested on the earlier 0.30.0, and the results are identical).

I'm not sure if this is a bug or not. I'm not getting expected results when doing a de-embed. It may be a mistake in my understanding, I'd appreciate any help.
Here's the workflow which I did:
1. I used an R&S FPC1500 VNA. It is attached to a semi-rigid coax, and I attached an OSL T-piece on the end (for which I have the calibration coefficients stored inside the VNA).
2. I calibrated the VNA
3. I removed the OSL T-piece, and attached a cable to the end of the existing cable.
4. I soldered a short, and captured a touchstone file, called ns-short.s1p for the range 10MHz-1GHz.
5. I removed the short, and captured a touchstone file called ns-open.s1p
6. I attached an 0603-sized 150 ohm resistor (measures 148 ohm to be more precise) on the end and captured a touchstone file called ns-dut-148.s1p
Next, in my Python code, I had the following:

```
short_fname = 'ns-short.s1p'
open_fname = 'ns-open.s1p'
dut_fname = 'ns-dut-148.s1p'
corrected_dut_fname = 'ns-dut-corrected.s1p'

# read the two files
short_ntwk = rf.Network(short_fname)
open_ntwk = rf.Network(open_fname)
dut_ntwk = rf.Network(dut_fname)

# perform the de-embedding
dm = OpenShort(dummy_short=short_ntwk, dummy_open=open_ntwk)
corrected_dut_ntwk = dm.deembed(dut_ntwk)

# save the corrected DUT network
corrected_dut_ntwk.write_touchstone(corrected_dut_fname)

# plot the results on a Smith chart
dut_ntwk.plot_s_smith(lw=2, label='Uncorrected')
corrected_dut_ntwk.plot_s_smith(lw=2, label='Corrected')
```

The results look pretty ugly:
[image]

Have I misunderstood the procedure? I have double-checked that the VNA was always in Cal status, i.e. the coefficients for the OSL were loaded, and I had performed the OSL calibration.

Any ideas where I am going wrong?

I have placed the open/short/DUT touchstone files zip folder here:
[ns_s1p.zip](https://github.com/scikit-rf/scikit-rf/files/14078058/ns_s1p.zip)

Many thanks for all your help in advance. I apologise if I've made some obvious mistake : (

keikawa, 2024-01-29:
Hi @shabaz123 ,
The code you provided looks good to me.
The OpenShort de-embedding is based on the assumption that the embedding network (here the connected cables) is shorter than the wavelength of the RF signal. Therefore, longer cables result in lower accuracy.
In this case, if the VNA has an electrical length correction function, it is recommended to use it.
Alternatively, the following methods are available for electrical length correction using scikit-rf.

```
import skrf as rf
import matplotlib.pyplot as plt

dut_fname = './ns-dut-148.s1p'
open_fname = './ns-open.s1p'
corrected_dut_fname = './ns-dut-corrected.s1p'

# read the two files
dut_ntwk = rf.Network(dut_fname)
open_ntwk = rf.Network(open_fname)

# Electrical length compensation
lossless_tline = rf.media.DefinedGammaZ0(frequency=dut_ntwk.frequency, z0_port=None, z0=50, Z0=None, gamma=1j)
electrical_length = open_ntwk.s_deg[:,0,0] / 2
corrected_dut_ntwk = lossless_tline.line(electrical_length, unit='deg') ** dut_ntwk

# save the corrected DUT network
corrected_dut_ntwk.write_touchstone(corrected_dut_fname)

# plot the results on a Smith chart
dut_ntwk.plot_s_smith(lw=2, label='Uncorrected')
corrected_dut_ntwk.plot_s_smith(lw=2, label='Corrected')

plt.show()
```

shabaz123, 2024-01-29:
Hi @keikawa 

Thank you for this! I just tried your code, and the corrected plot result is superb : )
The diagram here shows the setup I had used and the approximate lengths of cables:
[image]
Since I was only interested in 10MHz to 1GHz, I thought the lengths may have been fine. In any case, I'm very happy that the electrical length compensation method of de-embedding works, and I can run with this. My VNA is a very entry-level product, it doesn't have an automatic way of doing this.

Incidentally, to hopefully contribute back to the project a bit, I have forked scikit-rf and created a branch and hope to add code for the FPC1500 to become a supported VNA for control. I have made a start, but the command seems to be missing in the VNA user manual to retrieve vector measurements (I can retrieve scalar!), so I've raised a case with R&S, and will continue with that once they reply.

Thanks again for your help.

shabaz123, 2024-01-29:
Closing, thank you for the help!

closed 2024-01-29 by shabaz123