scikit-rf/scikit-rf #1022: S-parameter calculation incorrect for 3-port circuit

Fabdef, 2024-02-03:
**Bug**
S-parameter calculation is incorrect for a 3-port circuit

**Procedure to reproduce the bug**
Take the Wilkinson circuit available on scikit-rf website:
https://scikit-rf.readthedocs.io/en/latest/examples/circuit/Wilkinson%20Power%20Splitter.html

Copy the code and remove the resistor, which gives a Tee circuit with impedance matching:

```
# standard imports
import numpy as np
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0, stop=2, npoints=501, unit='GHz')

# characteristic impedance of the ports
z0_ports = 50

# resistor
R = 100
line_resistor = rf.media.DefinedGammaZ0(frequency=freq, z0=R)
resistor = line_resistor.resistor(R, name='resistor')

# branches
z0_branches = np.sqrt(2)*z0_ports
beta = freq.w/rf.c
line_branches = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_branches, gamma=0+beta*1j)

d = line_branches.theta_2_d(90, deg=True)  # @ 90°(lambda/4)@ 1 GHz is ~ 75 mm
branch1 = line_branches.line(d, unit='m', name='branch1')
branch2 = line_branches.line(d, unit='m', name='branch2')

# ports
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)
port3 = rf.Circuit.Port(freq, name='port3', z0=50)

# Connection setup
#┬Note that the order of appearance of the port in the setup is important
connections = [
           [(port1, 0), (branch1, 0), (branch2, 0)],
           [(port2, 0), (branch1, 1)],
           [(port3, 0), (branch2, 1)]
        ]

# Building the circuit
C = rf.Circuit(connections)

fig, (ax1,ax2) = plt.subplots(2, 1, sharex=True)
C.network.plot_s_db(ax=ax1, m=0, n=0,  lw=2)  # S11
C.network.plot_s_db(ax=ax1, m=1, n=1,  lw=2)  # S22
ax1.set_ylim(-90, 0)
C.network.plot_s_db(ax=ax2, m=1, n=0,  lw=2)  # S21
C.network.plot_s_db(ax=ax2, m=2, n=0,  ls='--', lw=2)  # S31
ax2.set_ylim(-4, 0)
fig.suptitle('Ideal Wilkinson Divider @ 1 GHz')
```
Simply execute the code and plot the S-parameters

**Expected behavior**
S21 and S31 should be at -3dB at 1 GHz (middle of the pass-band) because the circuit is now a simple tee, so the power from port 1 is equally split between ports 2 and 3.
This behavior was confirmed with a simulation on ADS. 

**Observed behavior**
S21 and S31 are -1.5 dB at 1 GHz.

**System**
 - OS: Windows
 - scikit-rf version: 0.31.0

**Additional context**
I am trying to design Wilkinsons with different port impedances. I obtained various strange/wrong S-parameter results with scikit-rf (sometimes S21 and S31 are at -0.8 dB, sometimes at -4.1 dB, for designs where ADS gives -3 dB). 
The procedure given here is the simplest I found to show the problem.

Vinc0110, 2024-02-03:
I can confirm the issue. It seems the line impedances are normalized incorrectly due to a missing specification of `z0_port`.
With this modification of `line_branches`, the results look correct:
```python
line_branches = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_branches, gamma=0+beta*1j, z0_port=z0_ports)
```
[image]

@jhillairet, @mhuser: Is this incorrect tutorial something we missed during the changes in PR #727?

Vinc0110, 2024-02-03:
Playing around with `z0_port` for both `line_branches` and `line_resistor` I noticed there is no problem when the resistor is included. That makes sense because in this case, there is no impedance mismatch at the ports to be treated one way or another.

jhillairet, 2024-02-03:
> @jhillairet, @mhuser: Is this incorrect tutorial something we missed during the changes in PR #727?

very probably....can you fix it?

Vinc0110, 2024-02-03:
Sure, I will take care of it. 

Before I do, I'd like to understand why we have `z0_port=None` by default. All those previous changes related to the `embed`  parameter were meant to make things less confusing, but it still confuses me. It seems like some (most?) users expect t-lines to behave like `z0_port=50` or some other value, but not like `z0_port=None`. What is the use case for the "raw" (not renormalized) characteristic impedance? Don't you always renormalize when t-lines or ports of different impedance connect?

Keep in mind that tools like ADS don't ask any such questions to the user, which is probably why this issue here arose.

https://scikit-rf.readthedocs.io/en/latest/api/media/generated/skrf.media.Media.html

> z0_port (number, array-like, or None) – z0_port is the port impedance for networks generated by the media. If z0_port is not None, the networks generated by the media are renormalized (or in other words embedded) from the characteristic impedance z0 of the media to z0_port. Else if z0_port is None, the networks port impedances will be the raw characteristic impedance z0 of the media. (Default is None)

mhuser, 2024-02-03:
I think the `z0_port=None` provided a kind of backward compatibility.  In #651 , a group of users who did electromagnetic simulations were interested in raw impedance. Most users however probably think in terms of 50-ohm measurements. 

What is strange to me in the above example is that the transmission line networks are not renormalized when connected to the `rf.Circuit.Port` impedances. In ADS, the "port" component explicitly solves the impedance problem in S-Parameter simulations.

The issue that we mostly face in scikit-rf is that we want to plot a line network as if has been measured, without cascading explicitly with ports. This is not the case in this example with `rf.Circuit`.

There is two open questions: what should  `z0_port` default be and why the above example does not renormalize when  `z0_port=None`.

jhillairet, 2024-02-04:
> There is two open questions: what should `z0_port` default be and why the above example does not renormalize when `z0_port=None`.

Both are tricky, are some users may use the specific characteristic impedance of certain transmission line (like waveguide) and not 50 ohm per default...

Fabdef, 2024-02-05:
Thank you very much for answering so fast. I am glad you were able to reproduce and understand the problem.

On my side I am not sure I really get how `rf.media.DefinedGammaZ0` should be defined. 
It seems counter-intuitive to me to have to define the value of the port connected to the line as one of the line attributes, because it has nothing to do with the line specs. As mentioned by @Vinc0110,  I was also expecting the normalization with the port impedance to happen when performing `rf.Circuit(connections)`. And yes, none of the other electronic/EM software I have used in the past (ADS and others) needs that kind of information.

Finally, to go back to my model, what I gave you was only a simplified version to show the problem. The real model is a Wilkinson with different port impedances (38 ohms for port 1 and 30 ohms for ports 2 and 3), having anywhere from 2 to 12 stages. Therefore, my current question being, how should 
`line_branches = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_branches, gamma=0+beta*1j, z0_port=z0_ports)`
be defined? 
Most of the 2 to 12 consecutive lines are not connected to any port, and the 3 ports of the circuit do not all have the same impedance.

Subsidiary question, even in the very simple case of a microstripline connecting 2 ports of different impedances, what should be defined as z0_port in 
`line_branches = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_branches, gamma=0+beta*1j, z0_port=z0_ports)`
?

mhuser, 2024-02-05:
> And yes, none of the other electronic/EM software I have used in the past (ADS and others) needs that kind of information.

Actually it does, but trough the port component of the simulation and not in the transmission lines block itself. This is one reason the default is `z0_port=None` which is better in the case of multiple port impedances.

The `z0_port=50` is only a trick if you want to plot the S-Parameters magnitude of the transmission line network as if you did measure it with a VNA and store it in a S-Parameter file. See [this tutorial](https://scikit-rf.readthedocs.io/en/latest/tutorials/Media.html) for more information and a drawing.

Currently I think that the fact it is not working with `rf.Circuit` is not the expected behaviour. I will try to show you a workaround. Maybe temporarily without using the very convenient `rf.Circuit`.

mhuser, 2024-02-05:
As a note, even adding a thru in `rf.Circuit` does not renormalize (S21 stick to -1.5 dB @ 1 GHz), event if `C.network` advertise the correct port impedance.
I will try to connect it with tee instead, which is a terrible thing to do.
```Python
# Does not works
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_ports)
# force the mismatch
thru1 = media50.thru(name='thru1');
thru2 = media50.thru(name='thru2');
thru3 = media50.thru(name='thru3');

# Connection setup
# Note that the order of appearance of the port in the setup is important
connections = [
           [(port1, 0), (thru1, 0)],
           [(thru1, 1),(branch1, 0), (branch2, 0)],
           [(port2, 0), (thru2, 0)],
           [(thru2, 1),(branch1, 1)],
           [(port3, 0), (thru3, 0)],
           [(thru3, 1),(branch2, 1)]
        ]
```

mhuser, 2024-02-05:
Ok, this is ugly but works:
```Python
# standard imports
import numpy as np
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0, stop=2, npoints=501, unit='GHz')

# characteristic impedance of most ports
z0_ports = 50

# resistor
R = 100
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_ports)
media30 = rf.media.DefinedGammaZ0(frequency=freq, z0=30)
media38 = rf.media.DefinedGammaZ0(frequency=freq, z0=38)
resistor = media50.resistor(R, name='resistor')

# branches
z0_branches = np.sqrt(2)*z0_ports
beta = freq.w/rf.c
line_branches = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_branches, gamma=0+beta*1j)

d = line_branches.theta_2_d(90, deg=True)  # @ 90°(lambda/4)@ 1 GHz is ~ 75 mm
branch1 = line_branches.line(d, unit='m', name='branch1')
branch2 = line_branches.line(d, unit='m', name='branch2')

# ports
port1 = media50.thru(name='port1');
port2 = media50.thru(name='port2');
port3 = media50.thru(name='port3');

# tees
tee1 = media50.tee(name='tee1');
tee2 = media50.tee(name='tee2');
tee3 = media50.tee(name='tee3');

# Connection setup
'''
           --branch1--tee2+--port2
           |              |
Port1--tee1+            resistor
           |              |
           --branch2--tee3+--port3
'''
C = rf.network.connect(port1, 1, tee1, 0, 1)
# new ports are at the end of network
# wire TL to tee
C = rf.network.connect(C, 1, branch1, 0, 1)
C = rf.network.connect(C, 2, branch2, 0, 1)
# tee at the end of branch1
C = rf.network.connect(C, 1, tee2, 0, 1)
C = rf.network.connect(C, 3, resistor, 0, 1)
C = rf.network.connect(C, 2, port2, 0, 1)
# tee at the end of branch 2
C = rf.network.connect(C, 1, tee3, 0, 1)
C = rf.network.connect(C, 4, port3, 0, 1)
# connect the resistor
C = rf.network.innerconnect(C, 2, 3, 1)

fig, (ax1,ax2) = plt.subplots(2, 1, sharex=True)
C.plot_s_db(ax=ax1, m=0, n=0,  lw=2)  # S11
C.plot_s_db(ax=ax1, m=1, n=1,  lw=2)  # S22
ax1.set_ylim(-90, 0)
C.plot_s_db(ax=ax2, m=1, n=0,  lw=2)  # S21
C.plot_s_db(ax=ax2, m=2, n=0,  ls='--', lw=2)  # S31
ax2.set_ylim(-4, 0)
fig.suptitle('Ideal Wilkinson Divider @ 1 GHz')
```

mhuser, 2024-02-05:
[image]

mhuser, 2024-02-05:
You can now change your port impedance such as
```Python
media30 = rf.media.DefinedGammaZ0(frequency=freq, z0=30)
media38 = rf.media.DefinedGammaZ0(frequency=freq, z0=38)
# ports
port1 = media38.thru(name='port1');
port2 = media30.thru(name='port2');
port3 = media30.thru(name='port3');
```
There should be no problem keeping tees and resistor media50.

mhuser, 2024-02-05:
> As a note, even adding a thru in `rf.Circuit` does not renormalize (S21 stick to -1.5 dB @ 1 GHz), event if `C.network` advertise the correct port impedance.

@Vinc0110 , @jhillairet Should'nt `rf.Circuit` care about mismatch in connected networks ?

jhillairet, 2024-02-05:
It should take care of (real) impedance mismatch.

mhuser, 2024-02-05:
> It should take care of (real) impedance mismatch.

This simple circuit below with two ports and one 71-ohm line works properly. Maybe the issue is when a node has more than two connection with different impedances.

[image]

```Python
# standard imports
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0.01, stop=2, npoints=200, unit='GHz')

# line
beta = freq.w/rf.c
tl = rf.media.DefinedGammaZ0(frequency=freq, z0=71, gamma=0+beta*1j)
tl1 = tl.line(100, unit='mm', name='tl1')

# ports
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)

# Connection setup
connections = [
           [(port1, 0), (tl1, 0)],
           [(port2, 0), (tl1, 1)],
        ]

# Building the circuit
C = rf.Circuit(connections)

fig = plt.figure()
C.network.plot_s_db()
fig.suptitle('Circuit impedance mismatch issue')
```

mhuser, 2024-02-05:
Hum I was wrong. If I compare two 90-ohm lines in parallel, which should gives an acceptable 50-ohm match, both `rf.Circuit` and network built with `rf.network.connect` shows a good agreement. Even if the node has 3 connection.

[image]

```Python
# standard imports
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0.01, stop=2, npoints=200, unit='GHz')

# tee
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)
tee1 = media50.tee(name='tee1');
tee2 = media50.tee(name='tee2');

# line
beta = freq.w/rf.c
tl = rf.media.DefinedGammaZ0(frequency=freq, z0=90, gamma=0+beta*1j)
tl1 = tl.line(100, unit='mm', name='tl1')
tl2 = tl.line(100, unit='mm', name='tl2')

# ports
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)

# Connection setup
'''   
       +--tl1--+
port1--+       +--port2
       +--tl2--+
'''
connections = [
           [(port1, 0), (tl1, 0), (tl2, 0)],
           [(port2, 0), (tl1, 1), (tl2, 1)],
        ]

# Building the circuit
C = rf.Circuit(connections)

#reference circuit with tees
A = rf.network.connect(tee1, 1, tl1, 0, 1)
A = rf.network.connect(A, 2, tl2, 0, 1)
A = rf.network.connect(A, 2, tee2, 0, 1)
A = rf.network.innerconnect(A, 1, 2, 1)
A.name = "reference"

fig = plt.figure()
C.network.plot_s_db(n=0, m=0)
A.plot_s_db(n=0, m=0)
fig.suptitle('Circuit impedance mismatch issue')
```

mhuser, 2024-02-05:
But if we test the simple tee splitter, there is indeed a difference between `rf.circuit` and `rf.network.connect` results in S21. Or I am wrong?
```
       +--tl1--port2
port1--+
       +--tl2--port3
```
[image]

```Python
# standard imports
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0.01, stop=2, npoints=200, unit='GHz')

# tee and thru
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)
tee1 = media50.tee(name='tee1');
tee2 = media50.tee(name='tee2');
thru1 = media50.thru(name='thru1');
thru2 = media50.thru(name='thru2');

# line
beta = freq.w/rf.c
tl = rf.media.DefinedGammaZ0(frequency=freq, z0=71, gamma=0+beta*1j)
tl1 = tl.line(100, unit='mm', name='tl1')
tl2 = tl.line(100, unit='mm', name='tl2')

# ports
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)
port3 = rf.Circuit.Port(freq, name='port3', z0=50)

# Connection setup
'''   
       +--tl1--port2
port1--+
       +--tl2--port3
'''
connections = [
           [(port1, 0), (tl1, 0), (tl2, 0)],
           [(port2, 0), (tl1, 1)],
           [(port3, 0), (tl2, 1)],
        ]

# Building the circuit
C = rf.Circuit(connections)

#reference circuit with tees
A = rf.network.connect(tee1, 1, tl1, 0, 1)
A = rf.network.connect(A, 2, tl2, 0, 1)
A = rf.network.connect(A, 1, thru1, 0, 1)
A = rf.network.connect(A, 2, thru2, 0, 1)
A.name = "reference"

fig = plt.figure()
C.network.plot_s_db(m=1, n=0)
A.plot_s_db(m=1, n=0)
fig.suptitle('Circuit impedance mismatch issue')
```

mhuser, 2024-02-05:
I found a simpler workaround that is to use tees within `rf.Circuit`. This gives similar results to `connect` and `innerconnect` methods. This narrows the issue on a wrong `rf.Circuit` behaviour instead of a `Media` issue and `z0_port` is not needed tough it still may be confusing by default.

[image]

```Python
# standard imports
import numpy as np
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0, stop=2, npoints=501, unit='GHz')
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)

# resistor
R = 100
resistor = media50.resistor(R, name='resistor')

# tees
tee1 = media50.tee(name='tee1');
tee2 = media50.tee(name='tee2');
tee3 = media50.tee(name='tee3');

# branches
z0_branches = np.sqrt(2) * 50
beta = freq.w/rf.c
line_branches = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_branches, gamma=0+beta*1j)
d = line_branches.theta_2_d(90, deg=True)
branch1 = line_branches.line(d, unit='m', name='branch1')
branch2 = line_branches.line(d, unit='m', name='branch2')

# ports
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)
port3 = rf.Circuit.Port(freq, name='port3', z0=50)

# Connection setup
'''
           --branch1--tee2+--port2
           |              |
Port1--tee1+            resistor
           |              |
           --branch2--tee3+--port3
'''
connections = [
           [(port1, 0), (tee1, 0)],
           [(tee1, 1), (branch1, 0)],
           [(tee1, 2), (branch2, 0)],
           [(branch1, 1), (tee2, 0)],
           [(branch2, 1), (tee3, 0)],
           [(resistor, 0), (tee2, 2)],
           [(resistor, 1), (tee3, 2)],
           [(port2, 0), (tee2, 1)],
           [(port3, 0), (tee3, 1)],
        ]

# reference network with connect and innerconnect
# new ports are at the end of network
# wire TL to tee
ref = rf.network.connect(tee1, 1, branch1, 0, 1)
ref = rf.network.connect(ref, 2, branch2, 0, 1)
# tee at the end of branch1
ref = rf.network.connect(ref, 1, tee2, 0, 1)
ref = rf.network.connect(ref, 3, resistor, 0, 1)
# tee at the end of branch 2
ref = rf.network.connect(ref, 1, tee3, 0, 1)
# connect the resistor
ref = rf.network.innerconnect(ref, 2, 3, 1)
ref.name = "reference"

# Building the circuit
C = rf.Circuit(connections)

fig, (ax1,ax2) = plt.subplots(2, 1, sharex=True)
C.network.plot_s_db(ax=ax1, m=0, n=0,  lw=2)  # S11
C.network.plot_s_db(ax=ax1, m=1, n=1,  lw=2)  # S22
ref.plot_s_db(ax=ax1, m=0, n=0,  lw=2)  # S11
ref.plot_s_db(ax=ax1, m=1, n=1,  lw=2)  # S22
ax1.set_ylim(-90, 0)
C.network.plot_s_db(ax=ax2, m=1, n=0,  lw=2)  # S21
C.network.plot_s_db(ax=ax2, m=2, n=0,  ls='--', lw=2)  # S31
ref.plot_s_db(ax=ax2, m=1, n=0,  lw=2)  # S21
ref.plot_s_db(ax=ax2, m=2, n=0,  ls='--', lw=2)  # S31
ax2.set_ylim(-4, 0)
fig.suptitle('Ideal Wilkinson Divider @ 1 GHz')
```

Fabdef, 2024-02-06:
The latest answer from @mhuser works for my model of multi-stage Wilkinson with different port impedances.
(S21 and S31 are near -3dB as they should, and not around -4.1 dB as they used to)

I just notice that for tees it is also necessary to provide the impedance. However tees having different linewidth will have a different impedance on each leg (which is typically the case for the wilkinson). Since the simulation works, I guess that the impedance we set for the tee in `media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)` gets overwritten or ignored? 
When entering `tee1.z0` I get 
```
array([[50+0.j, 50+0.j, 50+0.j],
[50+0.j, 50+0.j, 50+0.j],
[50+0.j, 50+0.j, 50+0.j]

x N_freq

]
```
Indicating that the tee assumes an impedance of 50 ohms on each leg, which is wrong.
Is it the intended behavior?

Anyways, thank you very much for the help on my case! :)

mhuser, 2024-02-06:
@Fabdef Thanks you a lot for reporting this.
Indeed each branch of the tee are 50-ohm, but they also are zero-length (possible only in simulation of course). This is why it still works, the impedance of the branches get normalized to the tee 50-ohm and then to the port characteristic impedance, but outside the excedentary computation step, the results should not change much.
This is only a temporarily workaround because it should have worked without the tees trick.

mhuser, 2024-02-06:
Is it related to this ?
The behaviour of circuit change if more than two connection.

https://github.com/scikit-rf/scikit-rf/blob/b4a65676c05b596063d13098a964d7de620a330b/skrf/circuit.py#L679-L696

mhuser, 2024-02-06:
@jhillairet if we think of a 3-port connection like a 2-port that is later connected to a 3rd port, the case when cnx_k sounds wrond. However I am not sure of my understanding. I would see something like:

$$ Z_{eq} = \sqrt{\sqrt{Z_1 \cdot Z_2} \cdot Z_3} $$

jhillairet, 2024-02-09:
OK... Sorry for the delay. I have to come back to the original paper to understand the issue.

jhillairet, 2024-02-10:
> @jhillairet if we think of a 3-port connection like a 2-port that is later connected to a 3rd port, the case when cnx_k > 2 sounds wrong. However I am not sure of my understanding. I would see something like the equ below for 3-port:

Interestingly, if we suppress the condition for two ports in `_Xmn_k` to only uses the equation (3), it seems to work with the following case you made above :

```
>        +--tl1--port2
> port1--+
>        +--tl2--port3
> ```
```

however, many tests are failing, some linked to complex values, but not all...

mhuser, 2024-02-10:
@jhillairet I did some testing comparing 2-port and 3-port:
```Python
# standard imports
import numpy as np
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0, stop=2, npoints=501, unit='GHz')

# tee and thru
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)
tee1 = media50.tee(name='tee1');
thru1 = media50.thru(name='thru1');
thru2 = media50.thru(name='thru2');

# branches
z0_branches = np.sqrt(2) * 50
beta = freq.w/rf.c
tl = rf.media.DefinedGammaZ0(frequency=freq, z0=90, gamma=0+beta*1j)
d = tl.theta_2_d(90, deg=True)
tl1 = tl.line(d, unit='m', name='tl1')
tl2 = tl.line(d, unit='m', name='tl2')

# ports
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)
port3 = rf.Circuit.Port(freq, name='port3', z0=50)

# Connection setup A
'''   
       +--tl1--port2
port1--+
       +--tl2--port3
'''
connections_A = [
           [(port1, 0), (tl1, 0), (tl2, 0)],
           [(port2, 0), (tl1, 1)],
           [(port3, 0), (tl2, 1)],
        ]
ACircuit = rf.Circuit(connections_A)

#reference circuit with tee
AConnect = rf.network.connect(tee1, 1, tl1, 0, 1)
AConnect = rf.network.connect(AConnect, 2, tl2, 0, 1)
AConnect = rf.network.connect(AConnect, 1, thru1, 0, 1)
AConnect = rf.network.connect(AConnect, 2, thru2, 0, 1)
AConnect.name = "Aref"

# Connection setup B
'''   
port1--tl1--port2
'''
connections_B = [
           [(port1, 0), (tl1, 0)],
           [(port2, 0), (tl1, 1)],
        ]
BCircuit = rf.Circuit(connections_B)
BConnect = thru1 ** tl1 ** thru2
BConnect.name = "Bref"

fig, axs = plt.subplots(2, 2)
ACircuit.network.plot_s_db(m=1, n=0, ax=axs[0,0])
AConnect.plot_s_db(m=1, n=0,ax=axs[0,0], ls='--')
axs[0,0].set_title('A 3-port IL')
ACircuit.network.plot_s_db(m=0, n=0, ax=axs[1,0])
ACircuit.network.plot_s_db(m=1, n=1, ax=axs[1,0])
AConnect.plot_s_db(m=0, n=0,ax=axs[1,0], ls='--')
AConnect.plot_s_db(m=1, n=1,ax=axs[1,0], ls='--')
axs[1,0].set_title('A 3-port RL')
BCircuit.network.plot_s_db(m=1, n=0, ax=axs[0,1])
BConnect.plot_s_db(m=1, n=0,ax=axs[0,1], ls='--')
axs[0,1].set_title('B 2-port IL')
ACircuit.network.plot_s_db(m=0, n=0, ax=axs[1,1])
AConnect.plot_s_db(m=0, n=0,ax=axs[1,1], ls='--')
axs[1,1].set_title('B 2-port RL')
fig.tight_layout()
```

mhuser, 2024-02-10:
Scikit-rf current

[image]

Scikit-rf  `_Xmn_k` to only uses the equation (3),

[image]

Scikit-rf  `_Xmn_k` to only geometrical mean of impedances

[image]

mhuser, 2024-02-10:
In these cases, `_Xmn_k` to only uses the equation (3) looks better.

jhillairet, 2024-02-10:
I'm searching for the S parameters of an ideal N-port mismatch, that is, the generalization of:

[image]
source : https://cds.cern.ch/record/1415639/files/p67.pdf

for an N-port with N different characteristic impedances...

If didn't find in book. ChatGPT, who always answer something (correct or not), proposes for reflection coefficients $S_{ii}$:
$S_{ii} = \frac{Z_i - Z_{\text{total}}}{Z_i + Z_{\text{total}}} $
with for port $i$:
$Z_{\text{total}} = Z_1 +  \ldots + Z_{i-1} + Z_{i+1} + \ldots + Z_N $

and for transmission coefficients $S_{ij}$ ($i \neq j$):
$S_{ij} = S_{ji} = \frac{2\sqrt{Z_iZ_j}}{Z_i + Z_j + \sum_{k \neq i,j} Z_k} $

but I can't check it with another soft from where i am

mhuser, 2024-02-10:
I wish to test these expressions on a 4-port mismatch when I have more time:

```Python
import skrf as rf

freq = rf.Frequency(start=1, stop=1, npoints=1, unit='GHz')
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)
splitter = media50.splitter(nports=4);
splitter.renormalize(z_new=[10, 20, 30, 40])

print(splitter.z0)
print(splitter.s.real)
```
Results:
```
[[10.+0.j 20.+0.j 30.+0.j 40.+0.j]]
[[[-0.04        0.67882251  0.55425625  0.48      ]
  [ 0.67882238 -0.51999991  0.39191842  0.33941132]
  [ 0.5542562   0.39191836 -0.67999992  0.27712813]
  [ 0.47999992  0.33941131  0.27712817 -0.75999996]]]
```

mhuser, 2024-02-10:
> I wish to test these expressions on a 4-port mismatch when I have more time:

The ChatGPT equation is coherent with CERN source for 2-port. It works well when there is no mismatch (what is rather useless for our question)
```
n_port = 2
z_port
[[50.+0.j 50.+0.j]]
s spliter
[[0. 1.]
 [1. 0.]]
s ChatGPT equation
[[0. 1.]
 [1. 0.]]
n_port = 3
z_port
[[50.+0.j 50.+0.j 50.+0.j]]
s spliter
[[-0.33333333  0.66666667  0.66666667]
 [ 0.66666667 -0.33333333  0.66666667]
 [ 0.66666667  0.66666667 -0.33333333]]
s ChatGPT equation
[[-0.33333333  0.66666667  0.66666667]
 [ 0.66666667 -0.33333333  0.66666667]
 [ 0.66666667  0.66666667 -0.33333333]]
```

But it does not seems to goes well with a mismatch (of course there can be an implementation error)
```
n_port = 2
z_port
[[10.+0.j 20.+0.j]]
s spliter
[[ 0.33333336  0.94280901]
 [ 0.94280901 -0.33333329]]
s ChatGPT equation
[[-0.33333333  0.94280904]
 [ 0.94280904  0.33333333]]
n_port = 3
z_port
[[10.+0.j 20.+0.j 30.+0.j]]
s spliter
[[ 0.0909091   0.77138921  0.62983664]
 [ 0.77138929 -0.45454547  0.44536166]
 [ 0.62983675  0.44536169 -0.63636371]]
s ChatGPT equation
[[-0.66666667  0.47140452  0.57735027]
 [ 0.47140452 -0.33333333  0.81649658]
 [ 0.57735027  0.81649658  0.        ]]
```

```Python
import numpy as np
import skrf as rf

freq = rf.Frequency(start=1, stop=1, npoints=1, unit='GHz')

for n in range(2, 4):
    n_port = n
    z_port = np.arange(1, n_port + 1) * 10
    #z_port = np.ones((n_port)) * 50
    
    media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)
    splitter = media50.splitter(nports=n_port);
    splitter.renormalize(z_new=z_port)
    
    print(f'n_port = {n_port}')
    print('z_port')
    print(splitter.z0)
    print('s spliter')
    print(splitter.s[0].real)
    
    # manual computation
    z_total = np.sum(z_port)
    s = np.zeros((n_port, n_port), dtype = complex)
    
    #sii
    for i in range(n_port):
        z_total_p = z_total - z_port[i]
        s[i,i] = (z_port[i] - z_total_p) / (z_port[i] + z_total_p)
        #sji
        for j in range(n_port):
            if j != i:
                s[j,i] = 2 * np.sqrt(z_port[i]*z_port[j]) / z_total
                s[i,j] = s[j,i]
    print('s ChatGPT equation')
    print(s.real)
```

mhuser, 2024-02-10:
As a side note, the lossless n-port splitter is defined in `Media` as

$$
Splitter(n) = 
\begin{bmatrix}
\frac{2}{n}-1 & \sqrt{\frac{1-\left(\frac{2-n}{n}\right)^2}{n-1}}  & \cdots & \sqrt{\frac{1-\left(\frac{2-n}{n}\right)^2}{n-1}} \\
\sqrt{\frac{1-\left(\frac{2-n}{n}\right)^2}{n-1}} & \frac{2}{n}-1 & \cdots & \sqrt{\frac{1-\left(\frac{2-n}{n}\right)^2}{n-1}} \\
\vdots  & \vdots  & \ddots & \vdots  \\
\sqrt{\frac{1-\left(\frac{2-n}{n}\right)^2}{n-1}} & \sqrt{\frac{1-\left(\frac{2-n}{n}\right)^2}{n-1}} & \cdots & \frac{2}{n}-1 
\end{bmatrix}
$$

jhillairet, 2024-02-14:
The method `_Xk` could be replaced by something like

```python
    def _Xk(self, cnx_k: list[tuple]) -> np.ndarray:
        z0s = np.array([ntw.z0[:,ntw_port] for (ntw, ntw_port) in cnx_k]).T
        
        med = media.DefinedGammaZ0(frequency=self.frequency)
        return renormalize_s(med.splitter(len(cnx_k)).s, z_old=50, z_new=z0s)
```
The tests associated to the X matrix elements will fail, but this expected as the formula (3) in the paper seems actually incorrect in case of different characteristic impedances. 

Some remaining tests also fail, some associated to a loss of precision (unclear why), and some due to complex char impedances (playing with s_def can solve some, but I'm convinced why it works)

mhuser, 2024-02-16:
> The method `_Xk` could be replaced by something like

I also did try something like that (but the code was less compact 😉 and observed the same tests failing. We could dodge this by renormalizing all the inner networks to a real reference impedance and then renormalizing to the external ports, but this sounds unsatisfactory (though it would probably works, like when @Vinc0110 specified `z_port=50` for the line.

I will try to think some more on it.

mhuser, 2024-02-16:
Let's take the 50 ohm 2-port splitter. It is lossless, reciprocal and matched. As a matter of fact, it is just a zero-length thru. The scattering parameters are:

$$
S = \begin{bmatrix}
 0 & 1\\
1 & 0\\
\end{bmatrix}
$$

Which are normalized to the port impedance:

$$Z_1 = 50, Z_2 = 50$$

The impedance parameters do not need to normalize to a port impedance, we can convert to them like [`skrf.network.s2z`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.s2z.html). Behind the hood, when using [`renormalize_s`](https://scikit-rf.readthedocs.io/en/latest/_modules/skrf/network.html#renormalize_s), scikit-rf just do `z2s(s2z())` with `z_old` and `z_new`.

$$
Z = 
\begin{bmatrix} \frac{1}{2\sqrt{|\Re(Z_1)|}} & 0 \\
0 & \frac{1}{2\sqrt{|\Re(Z_2)|}} \end{bmatrix}^{-1}
\left(1 - S \right)^{-1}
\left(S
\begin{bmatrix}Z_1 & 0 \\
0 & Z_2 \end{bmatrix} + \begin{bmatrix}Z_1 & 0 \\
0 & Z_2 \end{bmatrix}^{*} \right)
\begin{bmatrix} \frac{1}{2\sqrt{|\Re(Z_1)|}} & 0 \\
0 & \frac{1}{2\sqrt{|\Re(Z_2)|}} \end{bmatrix}
$$

Which gives

$$
Z = 
\left(\left(\begin{bmatrix} 1 & 0 \\
0 & 1 \end{bmatrix} - \begin{bmatrix} 0 & 1 \\
1 & 0 \end{bmatrix} \right)
\begin{bmatrix} 2\sqrt{50} & 0 \\
0 & 2\sqrt{50} \end{bmatrix}\right)^{-1}
\left(\begin{bmatrix} 0 & 1 \\
1 & 0 \end{bmatrix}
\begin{bmatrix}50 & 0 \\
0 & 50 \end{bmatrix} + \begin{bmatrix}50 & 0 \\
0 & 50 \end{bmatrix} \right)
\begin{bmatrix} \frac{1}{2\sqrt{50}} & 0 \\
0 & \frac{1}{2\sqrt{50}} \end{bmatrix}
$$

$$
Z = 
\begin{bmatrix} 2\sqrt{50} & -2\sqrt{50} \\
-2\sqrt{50} & 2\sqrt{50} \end{bmatrix}^{-1}
\begin{bmatrix} \frac{\sqrt{50}}{2} & \frac{\sqrt{50}}{2} \\
\frac{\sqrt{50}}{2} &\frac{\sqrt{50}}{2} \end{bmatrix}
$$

And here tadaaa, the left matrix is singular and cannot be inverted ! `s2z` will still return a result because this matrix is added a small offset by `nudge` to avoid singularity and gives a matrix with big numbers. Hence probably the loss of precision when using `z2s` on this result.

mhuser, 2024-02-16:
This dirty fix removes most failures by using legacy code instead of 2-port splitter.
I think the 3-port splitter has no singularity but I need to check.
```Python
    def _Xk(self, cnx_k: list[tuple]) -> np.ndarray:
        """
        Return the scattering matrices [X]_k of the individual intersections k.

        Parameters
        ----------
        cnx_k : list of tuples
            each tuple contains (network, port)

        Returns
        -------
        Xs : :class:`numpy.ndarray`
            shape `f x n x n`
        """

        # TEST : Could we use media.splitter() instead ? -> yes
        if len(cnx_k) == 2:
            Xnn = self._Xnn_k(cnx_k)  # shape: (nb_freq, nb_n)
            Xmn = self._Xmn_k(cnx_k)  # shape: (nb_freq, nb_n)
            # # for loop version
            # Xs = []
            # for (_Xnn, _Xmn) in zip(Xnn, Xmn):  # for all frequencies
            #       # repeat Xmn along the lines
            #     _X = np.tile(_Xmn, (len(_Xmn), 1))
            #     _X[np.diag_indices(len(_Xmn))] = _Xnn
            #     Xs.append(_X)

            # return np.array(Xs) # shape : nb_freq, nb_n, nb_n
            # vectorized version
            nb_n = Xnn.shape[1]
            Xs = np.tile(Xmn, (nb_n, 1, 1)).swapaxes(1, 0)
            Xs[:, np.arange(nb_n), np.arange(nb_n)] = Xnn

            return Xs # shape : nb_freq, nb_n, nb_n
        else:
            med = media.DefinedGammaZ0(frequency=self.frequency)
            return renormalize_s(med.splitter(len(cnx_k)).s,
                                         z_old=50, z_new=self._cnx_z0(cnx_k))
```

mhuser, 2024-02-16:
Let's think of a 50 ohm 3-port splitter. It is lossless and reciprocal but not matched. If a power wave is injected on one port, it is reflected by the 2 other 50 ohm port in parallel. Thus

$S_{11} = S_{22} = S_{33} = -\frac{50-(50/2)}{50+(50/2)} = -\frac{1}{3}$

Because it is lossless and reciprocal, the remaining power is equally split between the other ports and satisfy

$1 = S_{11} ^2 + S_{21}^2 + S_{31}^2$

If this is propagated to all ports combinations, this gives:

$$
S = \begin{bmatrix} -1/3 & 2/3 & 2/3 \\
2/3 & -1/3 & 2/3 \\
2/3 & 2/3 & -1/3\end{bmatrix}
$$

mhuser, 2024-02-16:
`s2z` of 50 ohm 3-port splitter gives:

$$
Z = 
\left(2\sqrt{50} \begin{bmatrix} 1-1/3 & -2/3 & -2/3 \\
-2/3 & 1-1/3 & -2/3 \\
-2/3 & -2/3 & 1-1/3\end{bmatrix}
\right)^{-1}
\frac{1}{2\sqrt{50}} \left(50 \begin{bmatrix} -1/3 & 2/3 & 2/3 \\
2/3 & -1/3 & 2/3 \\
2/3 & 2/3 & -1/3\end{bmatrix} + 50 \right)
$$

Oh no, singular matrix again and big numbers thank to `nudge` ! I guess the dirty fix above seemed to works only because node with 3 connexions with different impedances are not tested 🤔

mhuser, 2024-02-16:
Either I did some error or it is not suitable to renormalize the splitter with the `z2s(s2z())` method that is used by `renormalize_s`

mhuser, 2024-02-17:
I think I got the equation of the arbitrary port impedance splitter! 🎉 

Or at least in the example below with port of different impedances, it agrees with 10e-32 to the sum of least square with a model built with 50-ohm splitter connected to thrus of distinct impedances.
While the renormalize method is only close to 10E-14 (which is the square of the error).

```
Output:
n_port = 2
z_port
[20 40]
connect (reference)
[[ 0.33333333+0.j  0.94280904+0.j]
 [ 0.94280904+0.j -0.33333333+0.j]]
manual
[[ 0.33333333+0.j  0.94280904+0.j]
 [ 0.94280904+0.j -0.33333333+0.j]]
sum((manual - reference)^2) = (1.3866695599588098e-31+0j)
sum((renormalize - reference)^2) = (1.4047407312349626e-17+0j)
n_port = 3
z_port
[20 40 60]
connect (reference)
[[ 0.09090909+0.j  0.77138922+0.j  0.62983666+0.j]
 [ 0.77138922+0.j -0.45454545+0.j  0.44536177+0.j]
 [ 0.62983666+0.j  0.44536177+0.j -0.63636364+0.j]]
manual
[[ 0.09090909+0.j  0.77138922+0.j  0.62983666+0.j]
 [ 0.77138922+0.j -0.45454545+0.j  0.44536177+0.j]
 [ 0.62983666+0.j  0.44536177+0.j -0.63636364+0.j]]
sum((manual - reference)^2) = (6.644458308135964e-32+0j)
sum((renormalize - reference)^2) = (2.7950731588670355e-14+0j)
n_port = 4
z_port
[20 40 60 80]
connect (reference)
[[-0.04      +0.j  0.67882251+0.j  0.55425626+0.j  0.48      +0.j]
 [ 0.67882251+0.j -0.52      +0.j  0.39191836+0.j  0.33941125+0.j]
 [ 0.55425626+0.j  0.39191836+0.j -0.68      +0.j  0.27712813+0.j]
 [ 0.48      +0.j  0.33941125+0.j  0.27712813+0.j -0.76      +0.j]]
manual
[[-0.04      +0.j  0.67882251+0.j  0.55425626+0.j  0.48      +0.j]
 [ 0.67882251+0.j -0.52      +0.j  0.39191836+0.j  0.33941125+0.j]
 [ 0.55425626+0.j  0.39191836+0.j -0.68      +0.j  0.27712813+0.j]
 [ 0.48      +0.j  0.33941125+0.j  0.27712813+0.j -0.76      +0.j]]
sum((manual - reference)^2) = (1.2402988841853799e-31+0j)
sum((renormalize - reference)^2) = (4.5260197086374836e-15+0j)
``` 

```Python
import numpy as np
from numpy import sum, sqrt
import skrf as rf
from skrf.network import concat_ports, connect

freq = rf.Frequency(start=1, stop=1, npoints=1, unit='GHz')

for n in range(2, 5):
    n_port = n
    z_port = np.arange(1, n_port + 1) * 20
    #z_port = np.ones((n_port)) * 50
    
    # reference with connect
    media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)
    splitter = media50.splitter(nports=n_port);
    thrus = []
    for k in range(n):
        thrus.append(media50.thru(z0 = z_port[k]))
    
    # connect
    thru = concat_ports(thrus, port_order='second')
    nconnect = connect(splitter, 0, thru, 0, n)
    
    # renormalize
    splitter.renormalize(z_new=z_port)
    
    print(f'n_port = {n_port}')
    print('z_port')
    print(z_port)
    print('connect (reference)')
    print(nconnect.s[0])
    
    # manual computation
    y_total = sum(1. / z_port)
    s = np.zeros((n_port, n_port), dtype = complex)
    
    # sii
    for i in range(n_port):
        s[i,i] = 2. / (z_port[i] * y_total) - 1
        #sji
        for j in range(n_port):
            if j != i:
                y_eq = sqrt((1./z_port[i]) * (1./z_port[j]))
                s[j,i] = 2. *y_eq / y_total
                s[i,j] = s[j,i]
    print('manual')
    print(s)
    print(f'sum((manual - reference)^2) = {sum((s - nconnect.s[0])**2)}')
    print(f'sum((renormalize - reference)^2) = {sum((splitter.s[0] - nconnect.s[0])**2)}')
```

mhuser, 2024-02-17:
The case of $S_{ii}$ is quite trivial, the port has a mismatch with the equivalent impedance of the other port in parallel, which is more easily computed in terms of admittance.

$$
S_{ii} = \frac{Y_i - \sum_{j\neq i} Y_j}{Y_i + \sum_{j\neq i} Y_j} = \frac{2Y_i}{\sum_{j=1...n} Y_j} - 1
$$

Which is coherent with `rf.Circuit` # formula (1)

mhuser, 2024-02-17:
For $S_{ij}$ it should be remembered that it is a power divider. The remaining power that go through after the reflection is split according to the port's impedances (e.g. if they are equal the power is split in equal parts).

$$
1 = S_{ii}^2 + \sum_{j=1...n} S_{ij}^2
$$

The power that remains to be split is

$$
\sum_{j=1...n} S_{ij}^2 = 1 - S_{ii}^2 
$$

Which is satisfied by

$$
S_{ij} = \frac{2\sqrt{Y_i \cdot Y_j}}{\sum_{k=1...n} Y_{k}}
$$

mhuser, 2024-02-17:
Non-efficient code:
```Python
def _Xk(self, cnx_k: list[tuple]) -> np.ndarray:
        """
        Return the scattering matrices [X]_k of the individual intersections k.

        Parameters
        ----------
        cnx_k : list of tuples
            each tuple contains (network, port)

        Returns
        -------
        Xs : :class:`numpy.ndarray`
            shape `f x n x n`
        """
        # Xnn = self._Xnn_k(cnx_k)  # shape: (nb_freq, nb_n)
        # Xmn = self._Xmn_k(cnx_k)  # shape: (nb_freq, nb_n)
        # # for loop version
        # Xs = []
        # for (_Xnn, _Xmn) in zip(Xnn, Xmn):  # for all frequencies
        #       # repeat Xmn along the lines
        #     _X = np.tile(_Xmn, (len(_Xmn), 1))
        #     _X[np.diag_indices(len(_Xmn))] = _Xnn
        #     Xs.append(_X)

        # return np.array(Xs) # shape : nb_freq, nb_n, nb_n

        # vectorized version
        # nb_n = Xnn.shape[1]
        # Xs = np.tile(Xmn, (nb_n, 1, 1)).swapaxes(1, 0)
        # Xs[:, np.arange(nb_n), np.arange(nb_n)] = Xnn

        # return Xs # shape : nb_freq, nb_n, nb_n

        # TEST : Could we use media.splitter() instead ? -> does not work
        # _media = media.DefinedGammaZ0(frequency=self.frequency)
        # Xs = _media.splitter(len(cnx_k), z0=self._cnx_z0(cnx_k))
        # return Xs.s
        n_port = len(cnx_k)
        s = np.zeros((self.frequency.npoints, n_port, n_port), dtype = complex)
        y_k = self._Y_k(cnx_k)
        # sii
        for i in range(n_port):
            (ntw_i, port_i) = cnx_k[i]
            s[:,i,i] = 2. / (ntw_i.z0[:,port_i] * y_k) - 1
            #sji
            for j in range(n_port):
                if j != i:
                    (ntw_j, port_j) = cnx_k[j]
                    y_eq = np.sqrt((1./ntw_i.z0[:,port_i]) \
                                   * (1./ntw_j.z0[:,port_j]))
                    s[:,j,i] = 2. *y_eq / y_k
                    s[:,i,j] = s[:,j,i]
        return s
```

jhillairet, 2024-02-17:
Great! So it confirms that the equation (3) is incorrect in the generalized case of different impedances. 

I was working in parallel on this too ;) I made a new implementation of `renormalize_s` which reduces the numerical noise when renormalizing the S-param of a N-port junction. But it only supports power waves.

mhuser, 2024-02-17:
Most test are passing with the code above (only 2 fails). It still need to be double checked.
Also the 3-port splitter example shows a good agreement with an equivalent circuit with `connect`.
I am a bit lost on how to optimize the code to use vectorized versions and so on.
[image]

```Python
# standard imports
import numpy as np
import matplotlib.pyplot as plt
import skrf as rf
rf.stylely()

# frequency band
freq = rf.Frequency(start=0, stop=2, npoints=501, unit='GHz')

# tee and thru
media50 = rf.media.DefinedGammaZ0(frequency=freq, z0=50)
tee1 = media50.tee(name='tee1');
thru1 = media50.thru(name='thru1');
thru2 = media50.thru(name='thru2');

# branches
z0_branches = np.sqrt(2) * 50
beta = freq.w/rf.c
mtl1 = rf.media.DefinedGammaZ0(frequency=freq, z0=z0_branches, gamma=0+beta*1j)
d1 = mtl1.theta_2_d(90, deg=True)
tl1 = mtl1.line(d1, unit='m', name='tl1')
tl2 = mtl1.line(d1, unit='m', name='tl2')

# ports
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)
port3 = rf.Circuit.Port(freq, name='port3', z0=50)

# Connection setup
'''   
       +--tl1--port2
port1--+
       +--tl2--port3
'''
connections_A = [
           [(port1, 0), (tl1, 0), (tl2, 0)],
           [(port2, 0), (tl1, 1)],
           [(port3, 0), (tl2, 1)],
        ]
ACircuit = rf.Circuit(connections_A)

#reference circuit with tee
AConnect = rf.network.connect(tee1, 1, tl1, 0, 1)
AConnect = rf.network.connect(AConnect, 2, tl2, 0, 1)
AConnect = rf.network.connect(AConnect, 1, thru1, 0, 1)
AConnect = rf.network.connect(AConnect, 2, thru2, 0, 1)
AConnect.name = "ref"

fig, axs = plt.subplots(1, 2)
ACircuit.network.plot_s_db(m=1, n=0, ax=axs[0])
AConnect.plot_s_db(m=1, n=0,ax=axs[0], ls='--')
axs[0].set_title('3-port IL')
ACircuit.network.plot_s_db(m=0, n=0, ax=axs[1])
ACircuit.network.plot_s_db(m=1, n=1, ax=axs[1])
AConnect.plot_s_db(m=0, n=0,ax=axs[1], ls='--')
AConnect.plot_s_db(m=1, n=1,ax=axs[1], ls='--')
axs[1].set_title('3-port RL')
fig.tight_layout()
```

mhuser, 2024-02-17:
> I was working in parallel on this too ;) I made a new implementation of `renormalize_s` which reduces the numerical noise when renormalizing the S-param of a N-port junction. But it only supports power waves

Sounds great !

jhillairet, 2024-02-17:
Here is an optimized version:

```python
    def _Xk(self, cnx_k: List[Tuple]) -> np.ndarray:
        """
        Return the scattering matrices [X]_k of the individual intersections k.

        Parameters
        ----------
        cnx_k : list of tuples
            each tuple contains (network, port)

        Returns
        -------
        Xs : :class:`numpy.ndarray`
            shape `f x n x n`
        """
        y0s = np.array([1/ntw.z0[:,ntw_port] for (ntw, ntw_port) in cnx_k]).T
        y_k = y0s.sum(axis=1)

        Xs = np.zeros((len(self.frequency), len(cnx_k), len(cnx_k)), dtype='complex')
        
        Xs = 2 *np.sqrt(np.einsum('ij,ik->ijk', y0s, y0s)) / y_k[:, None, None]       
        np.einsum('kii->ki', Xs)[:] -= 1  # Sii
```

With this, the methods `_Xmn_k`, `_Xnn_k` and `_Y_k` are no more useful and the associated tests can be suppressed.

There is still one test failing, I did not investigate yet.

closed 2024-02-18 by mhuser