scikit-rf/scikit-rf #1026: Phase mismatch between symmetrical error networks

Ismail-Majed, 2024-02-09:
When extracting the error networks from my calibration using  `TRL.error_ntwk` I find that though the two sides of the calibration are essentially identical, the resultant Smith Charts seem to suggest that the forward network may be acting normally but the reverse network is not.
**Reverse Excitation Network:**
[image]

**Forward Excitation Network:**
[image]

I noticed there is a TODO related to the phase within the code that generates the error coefficients but I'm unsure if this is related. I am trying to only deembed the reverse network from another network, without having to add a delay.

Thank you for any help.

Ttl, 2024-02-11:
Can you give more information on how you are doing the calibration? Is this synthetic or real data? Are you doing TRL as second calibration?

If you are doing TRL as a second calibration and your TRL launches are identical on both ports, then the error networks should also be identical. However, if you don't have first calibrated the VNA before TRL calibration then the error networks also include cables and VNAs own internal errors that in general are not identical on both ports.

Ismail-Majed, 2024-02-11:
The data is all real, the TRL is my secondary calibration I have the VNA calibrated up to some stainless steel cables which enter a cryogenic chamber. Everything from the calibration plane into the cryostat is identical.

I am trying to calibrate a circuit which is just an impedance tuner within the cryostat which will show a known impedance at its output port. 

When I de-embed the reverse excitation error network (error_ntwk[1]) from my raw tuner measurement network, I see my S22 spinning around the Smith chart as though there is a significant positive or negative delay. 

[image]

If I instead de-embed the forward excitation error network (error_ntwk[0]) the S22 now appears to be very close to what is expected.
[image]

I'm not sure if I'm misinterpreting the results but I don't see why the 2 error networks should not act similarly, if not identically.

closed 2024-02-11 by Ismail-Majed

Ttl, 2024-02-11:
```python
#!/usr/bin/env python
import skrf
from skrf.media import Coaxial

freq = skrf.F(1, 1, 1, 'GHz')
coax1mm = Coaxial(freq, z0_port=50, Dint=0.434e-3, Dout=1.0e-3, sigma=1e8)

X = coax1mm.line(1, 'm', z0=58, name='X') ** coax1mm.line(100, 'um', z0=10)
Y = X.copy()
Y.flip()

gamma_f = coax1mm.match(z0=50)
gamma_r = coax1mm.match(z0=50)

dut = coax1mm.random(n_ports=2)
through = coax1mm.line(0, 'um', z0=50)
line = coax1mm.line(300, 'um', z0=50)
reflect = coax1mm.short(nports=2)

def measure(net):
    return X**net**Y

ideals = [through, reflect, line]
measured = [measure(n) for n in ideals]
cal = skrf.TRL(measured = measured, ideals=ideals)

forward, reverse = cal.error_ntwk

# These should be all identical
print('Actual error network:')
print(X.s)
print('Solved forward error network:')
print(forward.s)
print('Solved reverse error network:')
print(reverse.s)

dut_cal = cal.apply_cal(measure(dut))
print()
# These should also be all identical
print('DUT:')
print(dut.s)
print('Calibrated DUT:')
print(dut_cal.s)
print('De-embedded error networks')
flipped_reverse = reverse.copy()
flipped_reverse.flip()
print((forward.inv ** measure(dut) ** flipped_reverse.inv).s)
```

I tested it with synthetic data and it seems to work fine. In this case there is 180 degree phase difference in solved error networks, but that does not matter for calibration. Check what you are doing against that example.

If you have any phase difference in the cables between the ports forward and reverse error networks won't be identical.