scikit-rf/scikit-rf #1047: module 'collections' has no attribute 'Sequence'

williamsmi3, 2024-03-23:
**Describe the bug**
trying to use skrf in JupyterLab 4.0.11 on Anaconda environment with Python 3.11.8. Error happens on import `import skrf as rf`

```
Error:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[3], line 3
      1 import numpy as np
      2 import pandas as pd
----> 3 import skrf as rf
      4 from skrf import Network
      5 from matplotlib import pyplot as plt

File /opt/anaconda3/envs/myenv/lib/python3.11/site-packages/skrf/__init__.py:14
      9 __version__ = '0.18.0'
     10 ## Import all  module names for coherent reference of name-space
     11 #import io
---> 14 from . import frequency
     15 from . import network
     16 from . import networkSet

File /opt/anaconda3/envs/myenv/lib/python3.11/site-packages/skrf/frequency.py:42
     40 from numpy import fft, shape, gradient# used to center attribute `t` at 0
     41 import re
---> 42 from .util import slice_domain,find_nearest_index
     43 #from .constants import ZERO
     44 global ZER0

File /opt/anaconda3/envs/myenv/lib/python3.11/site-packages/skrf/util.py:281
    276                 f.write(s)
    279 # general purpose objects
--> 281 class HomoList(collections.Sequence):
    282     '''
    283 
    284     A Homogeneous Sequence
   (...)
    320     >>> h[h.prop==value].func()
    321     '''
    324     def __init__(self, list_):

AttributeError: module 'collections' has no attribute 'Sequence'
```

**System**
 - OS: MacOS 14
 - scikit-rf version 0.18.1

jhillairet, 2024-03-23:
> * scikit-rf version 0.18.1

this is a rather old version. How did you install it? The [recommended way](https://scikit-rf.readthedocs.io/en/latest/tutorials/Installation.html) using anaconda is: 

`conda install -c conda-forge  scikit-rf`

jhillairet, 2024-03-26:
Does it solve your issue?

closed 2024-03-27 by jhillairet