scikit-rf/scikit-rf #1049: Support request for network.z_time_step

Nimrodxv230, 2024-04-03:
I would like to generate NumPy arrays for z time step. i could then export the array to csv.

is it possible to access the data in this way?

denzchoe, 2024-04-03:
I believe this is supposed to go to the Discussion Thread rather than issue.

You could already do so with Numpy.savetxt or pandas to csv, no?

Vinc0110, 2024-04-03:
You can calculate the step response with [`Network.step_response()`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.step_response.html). See this [example in the docs](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Time%20domain%20reflectometry%2C%20measurement%20vs%20simulation.html#Measurement-vs-simulation-comparison). 
If your data does not include the dc point, you should consider using [`Network.extrapolate_to_dc()`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.extrapolate_to_dc.html).