scikit-rf/scikit-rf #1051: Adding support to the Keysight N5242B VNA

Ustun-R-Sunay, 2024-04-09:
I recently just learned about this package yesterday and have been really enjoying all of the features. I currently have access to a Keysight N5242B PNAX and got a warning when connecting using the package:

"This model (N5242B) has not been tested with scikit-rf. By default, all features are turned on but older instruments might be missing    SCPI support for some commands which will cause errors."

I haven't had any issues yet, but would love to contribute to this project. Let me know if there are any test-scripts that I can run to verify your code. Thanks!

jhillairet, 2024-04-10:
@cafeclimber , do you have any comment/idea ?

cafeclimber, 2024-04-11:
Hey yeah! So if you look at the source for the PNA class, it contains a list of supported models. That is just a list of models that have been tested by maintainers to ensure all commands work as expected. If you're able to run through all the command's available to your PNA, and everything works you can submit a PR adding that model to the list! 

You mentioned you have a PNA-X, I'm not sure if that introduces additional functionality that would need to be added, but if so, we can discuss potentially introducing another class for PNA-X models.