scikit-rf/scikit-rf #1056: error in the def "write_spice_subcircuit_s(self, z0, file: str) -> None: "

yifan0407, 2024-04-29:
After performed the function of write_spice_subcircuit_s(z0, spice_filename), I get the .sp file and import it into ADS, but I have no idea to choose which netlist dialect. 
I chose PSPICE and imported the file,  but there is some warnings that remind me WARNING: Referenced circuit "1" not found. WARNING: Referenced circuit ""-1"" not found. 
when I simulated circuit, there is an error "syntax error" in this line:
X1 nt11 0 rcl_vccs_admittance res=2.061 cap=347.907a ind=369.601a gm=970.280m mult=1
when I check the corresponding line in the impoerted circuit file (.net), I found that:
1:x1 nt11 0 rcl_vccs_admittance res "=" "X2.061" cap "=" "X347.907a" ind "=" "X369.601a" \
      gm "=" "X970.280m" mult "="

do you have some ideas about the error? can you please give me some instructions?

yifan0407, 2024-04-29:
yes I found the file can be recognised by ITSPICE software, have you used Keysight ADS?

FranzForstmayr, 2024-04-29:
No I haven't, sorry

Vinc0110, 2024-05-11:
It seems like ADS does not fully support regular SPICE netlists. See this example of a simple diode model:
https://www.edaboard.com/threads/how-to-use-a-spice-model-lib-file-in-keysight-ads-alternatives-helpful-too.376041/

The forums of Texas Instruments are also full of questions on how to include the SPICE netlists for the TI device models into ADS.

I don't currently have access to ADS anymore, but I remember they run their own vector-fitting routines when you include S-parameter blocks into a simulation. Of course, that only works if you can provide your network as a SnP file.

To summarize, I don't think this is an issue with the SPICE netlists exported with `VectorFitting.write_spice_subcircuit_s()`, but rather an ADS issue.

Vinc0110, 2024-11-20:
Lets' close this issue. It does not seem to be a problem in our netlist.

closed 2024-11-20 by Vinc0110