scikit-rf/scikit-rf #1057: Importing scikit-rf imports many modules and is slow

jhillairet, 2024-04-30:
Issue created following [this post on reddit](https://www.reddit.com/r/rfelectronics/comments/1cf9zu2/comment/l1suprm/).

The number of modules imported in the namespace when importing scikit-rf is quite large and can take quite some times. 

Just importing `skrf.constants` imports PIL, scipy, matplotlib, pandas, and all of the skrf modules, and generally takes about 1 second to complete (where numpy takes around 150 ms). 

Here's an example showing the difference between numpy import and skrf imports. First, here's numpy as a baseline.

# numpy
```python
from time import perf_counter
import sys

tic = perf_counter()
from numpy import array

print ("numpy import: ", perf_counter() - tic)
print (list(sorted(sys.modules.keys())))
```
numpy import: 0.13497460004873574

[List of imported modules](https://pastebin.com/jCeXk5V7).

# scikit-rf

```python
from time import perf_counter
import sys

tic = perf_counter()
from skrf.constants import c

print ("SKRF import: ", perf_counter() - tic)

print (list(sorted(sys.modules.keys())))
```

With the result:

SKRF import: 1.014745100052096

[List of imported modules](https://pastebin.com/hfM2MK0e)

cafeclimber, 2024-05-02:
It seems [we are not the only ones discussing](https://github.com/scikit-learn/scikit-learn/issues/23205) the issue of lazy loading modules while maintaining the ergonomics of e.g. `import skrf as rf` by re-exposing all of the modules in the main library `__init__.py`. They refer to [this]() document discussing lazy-loading. I wonder if an audit of the libraries we use to attempt to limit the dependencies we have, or deferring some imports to the function level as needed would be the way to go

cafeclimber, 2024-05-02:
Here's a breakdown of where the import times are coming from.

(This is the command used to get this):

```
$ python -X importtime perf.py 2> import.log
```

where the contents of `perf.py` are simply `import skrf`

```
  import time: self [us] | cumulative | imported package
  import time:       176 |        176 |   _io
  import time:        42 |         42 |   marshal
  import time:       384 |        384 |   posix
  import time:       646 |       1247 | _frozen_importlib_external
  import time:        97 |         97 |   time
  import time:       203 |        299 | zipimport
  import time:        56 |         56 |     _codecs
  import time:       558 |        613 |   codecs
  import time:       378 |        378 |   encodings.aliases
  import time:      1290 |       2280 | encodings
  import time:       192 |        192 | encodings.utf_8
  import time:        81 |         81 | _signal
  import time:        50 |         50 |     _abc
  import time:       266 |        316 |   abc
  import time:       278 |        593 | io
  import time:        63 |         63 |       _stat
  import time:       264 |        326 |     stat
  import time:       931 |        931 |     _collections_abc
  import time:       250 |        250 |       genericpath
  import time:       307 |        557 |     posixpath
  import time:       937 |       2750 |   os
  import time:       342 |        342 |   _sitebuiltins
  import time:       214 |        214 |     __future__
  import time:       258 |        471 |   _virtualenv
  import time:       261 |        261 |   _distutils_hack
  import time:       275 |        275 |   types
  import time:       378 |        378 |       warnings
  import time:       282 |        660 |     importlib
  import time:       117 |        117 |     importlib._abc
  import time:        91 |         91 |         itertools
  import time:       175 |        175 |         keyword
  import time:        65 |         65 |           _operator
  import time:      1015 |       1080 |         operator
  import time:       192 |        192 |         reprlib
  import time:        53 |         53 |         _collections
  import time:       886 |       2475 |       collections
  import time:       106 |        106 |         _functools
  import time:       790 |        895 |       functools
  import time:       593 |       3962 |     contextlib
  import time:       343 |       5081 |   importlib.util
  import time:       170 |        170 |   importlib.machinery
  import time:       104 |        104 |     apport_python_hook
  import time:       103 |        206 |   sitecustomize
  import time:      3401 |      12954 | site
  import time:       107 |        107 |       collections.abc
  import time:       481 |        481 |         enum
  import time:        49 |         49 |           _sre
  import time:       243 |        243 |             sre_constants
  import time:       549 |        791 |           sre_parse
  import time:       351 |       1190 |         sre_compile
  import time:        75 |         75 |         _locale
  import time:       145 |        145 |         copyreg
  import time:       616 |       2505 |       re
  import time:      1423 |       4033 |     typing
  import time:       355 |        355 |     numbers
  import time:        80 |         80 |           numpy._utils
  import time:       218 |        297 |         numpy._globals
  import time:       150 |        150 |         numpy.exceptions
[... 1210 lines cut ...]
  import time:        21 |     159112 |         skrf.io.general
  import time:       663 |     159775 |       skrf.data
  import time:       227 |     160001 |     skrf.media.freespace
  import time:       172 |        172 |       skrf.tlineFunctions
  import time:       227 |        398 |     skrf.media.cpw
  import time:       340 |        340 |     skrf.media.rectangularWaveguide
  import time:       196 |        196 |     skrf.media.circularWaveguide
  import time:       202 |        202 |     skrf.media.coaxial
  import time:       222 |        222 |     skrf.media.mline
  import time:       172 |        172 |     skrf.media.definedAEpTandZ0
  import time:       238 |     162745 |   skrf.media
  import time:       304 |        304 |   skrf.circuit
  import time:       263 |        263 |   skrf.taper
  import time:       205 |        205 |   skrf.instances
  import time:       329 |        329 |   skrf.vectorFitting
  import time:       765 |        765 |   skrf.qfactor
  import time:        87 |         87 |   vi
  import time:       458 |     877009 | skrf
```

A better way to view might be a flamegraph like this one:

[image]

Will let you know what more I can find

cafeclimber, 2024-05-02:
Ok so going through some stuff, I definitely think lazy loading is called for. Especially for things that aren't required for all operations. I often don't need to plot, so I shouldn't be loading matplotlib, even if it's available, at least until I use it. Especially considering roughly 1/3rd of the startup time is due to just mpl. I was able to shave off 300ms just by moving the pandas imports in the io module to the respective functions that need them.

jasondurbin, 2024-05-04:
+1 for lazy loading, because I was the one who brought the issue up.

cafeclimber, 2024-05-04:
I'll try to take a crack at it today!