scikit-rf/scikit-rf #1062: No good results with IEEEP370_MM_ZC_2xThru

Hasko7, 2024-05-03:
### **Description of the issue**
Hi guys, I am using the mixed mode IEEEP370 de-embedding implementation from scikit-rf, but I am not getting good results with MM IEEE 370 de-embedding with impedance correction ZC. 

I simulated a Beatty structure as a DUT in CST and then with the fixtures as a composite structure Fixture1-DUT-Fixture2. I also simulated fixtures without the DUT as 2x-Thru. After the simulation, I produced it as a PCB and measured it.  For validation, I applied the mixed-mode de-embedding from scikit-rf to it and compared it with the simulated DUT from CST. 

I think that fixture 2 is returned incorrectly with ZC, because after de-embedding there is always a ripple on the right side in the time domain. This is also visible in the frequency domain. 

### **Code**
All the utilized MM S-Parameters are in odd/even format.  Therefore i used at the beginning the renumber-methode: 
 .renumber([0,2,1,3], [0, 1, 2, 3])

**This is how I realized the ZC deembedding:
simulated data:** 
```
z0 = 50
dm_mmzc_bend  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru_bend, dummy_fix_dut_fix = se_fdf_bend,
                         z0 = z0, bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin = 1, NRP_enable = True, name = 'mmzc sharp bends simulated')
se_d_dut_bend_zc = dm_mmzc_bend.deembed(se_fdf_bend)
se_zc_fix1 = dm_mmzc_bend.se_side1.copy()
se_zc_fix1.name = 'zc_FIX-1 sharp bends simulated'
se_zc_fix2 = dm_mmzc_bend.se_side2.copy()
se_zc_fix2.name = 'zc_FIX-2 sharp bends simulated'

mm_d_dut_bend_zc = se_d_dut_bend_zc.copy()
mm_d_dut_bend_zc.se2gmm(p = 2)
mm_zc_fix1_bend = se_zc_fix1.copy()
mm_zc_fix1_bend.se2gmm(p = 2)
mm_zc_fix2_bend = se_zc_fix2.copy()
mm_zc_fix2_bend.se2gmm(p = 2)

mm_d_dut_bend_zc_dc = mm_d_dut_bend_zc.extrapolate_to_dc(kind='cubic')
mm_zc_fix1_bend_dc = mm_zc_fix1_bend.extrapolate_to_dc(kind='cubic')
mm_zc_fix2_bend_dc = mm_zc_fix2_bend.extrapolate_to_dc(kind='cubic')

**residuals for consistency check sharp bend:**
se_res_bend_zc = dm_mmzc_bend.deembed(se_2xthru_bend)
se_res_bend_zc.name = 'zc_residuals sharp bends simulated'
mm_res_bend_zc = se_res_bend_zc.copy()
mm_res_bend_zc.se2gmm(p = 2)
mm_res_bend_zc.s += 1e-15 # avoid numeric singularities
```

**measured data:**
```
dm_mmzc_bend_meas  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru_bend_meas, dummy_fix_dut_fix = se_fdf_bend_meas,
                         z0 = z0, bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin = 0, NRP_enable = False, name = 'mmzc sharp bends measured')
se_d_dut_bend_meas_zc = dm_mmzc_bend_meas.deembed(se_fdf_bend_meas)
se_zc_fix1_meas = dm_mmzc_bend_meas.se_side1.copy()
se_zc_fix1_meas.name = 'zc_FIX-1 sharp bends measured'
se_zc_fix2_meas = dm_mmzc_bend_meas.se_side2.copy()
se_zc_fix2_meas.name = 'zc_FIX-2 sharp bends measured'

mm_d_dut_bend_meas_zc = se_d_dut_bend_meas_zc.copy()
mm_d_dut_bend_meas_zc.se2gmm(p = 2)
mm_zc_fix1_bend_meas = se_zc_fix1_meas.copy()
mm_zc_fix1_bend_meas.se2gmm(p = 2)
mm_zc_fix2_bend_meas = se_zc_fix2_meas.copy()
mm_zc_fix2_bend_meas.se2gmm(p = 2)

mm_d_dut_bend_meas_zc_dc = mm_d_dut_bend_meas_zc.extrapolate_to_dc(kind='cubic')
mm_zc_fix1_bend_meas_dc = mm_zc_fix1_bend_meas.extrapolate_to_dc(kind='cubic')
mm_zc_fix2_bend_meas_dc = mm_zc_fix2_bend_meas.extrapolate_to_dc(kind='cubic')

**residuals for consistency check sharp bend**
se_res_bend_meas_zc = dm_mmzc_bend_meas.deembed(se_2xthru_bend_meas)
se_res_bend_meas_zc.name = 'zc_residuals sharp bends measured'
mm_res_bend_meas_zc = se_res_bend_meas_zc.copy()
mm_res_bend_meas_zc.se2gmm(p = 2)
mm_res_bend_meas_zc.s += 1e-15 # avoid numeric singularities
```

### **De-embedding results**
**Simulated results in TD:**
[image]
[image]

**Simulated results in FD:**
[image]

**Measured results in TD:**
[image]
[image]

**Measured results in FD:**
[image]

### **TDR checking according to IEEE370**
FIX-2 from ZC is called with Z11 just like FIX-2 from NZC , because FIX-2 is flipped in scikit-rf, but there is a difference between FIX-2 from NZC and ZC. Therefore I think that FIX-2 is returned incorrectly with ZC:
[image]

However, FIX-2 with Z22 gives more plausible results, although FIX-2 is flipped and should actually be called with Z11 as with NZC:

[image]

This can also be seen in a direct comparison between the fixtures and the composite structure FDF:

[image]

You can also see that the method with the impedance correction works and that the fixtures are modelled better because of the peeling calculation in De-embedding with ZC, if you look at FIX-2 with Z22dd. Here FIX-2 with Z22dd:

[image]

Therefore, I think that FIX-2 is returned incorrectly with ZC and thus the ripples in the de-embedding occur on the right side in the time domain and occur in the frequency domain, although the fixtures with ZC are modeled correctly by the peeling calculation, as can be seen in the figure.

**The simulation and measurement data of the MM S parameters are attached here so that you can also implement it yourself:**
[ZC_check.zip](https://github.com/scikit-rf/scikit-rf/files/15199235/ZC_check.zip)

mhuser, 2024-05-13:
Hello @Hasko7 
I tested your data and got the same results.

```Python
dm_mmzc  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru, dummy_fix_dut_fix = se_fdf,
                         bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin = 0, NRP_enable = False, name = 'mmzc')
```

[image]
If I call

```Python
se_d_dut_zc = dm_mmzc.se_side1.inv ** se_fdf ** dm_mmzc.se_side2.inv
```
The results looks much more coherents.
[image]

This means that the FIX-2-3 is wrongly flipped. This has not been noticed in the [example](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/IEEEP370%20Deembedding.html#IEEEP370_MM_ZC_2xThru-with-impedance-correction) because the FIX was symmetrical.

mhuser, 2024-05-13:
With #1072 and
```Python
dm_mmzc  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru, dummy_fix_dut_fix = se_fdf,
                         bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin = 1, NRP_enable = True, name = 'mmzc')
```
I now get:
[image]

mhuser, 2024-05-15:
@Hasko7 the master branch implement now the fix thanks to your contribution !
Are you able to test it ?
Best regards

Hasko7, 2024-05-26:
@mhuser thanks for responding. I have tested it and it works. However, I have noticed some other problems. I will explain them soon next week.

Hasko7, 2024-06-21:
Hello @mhuser  and @jhillairet ,I am writing to you again regarding the **MM IEEE 370 de-embedding with impedance correction ZC**. 
The ZC is now working much better after the error with FIX-2 was corrected, but I have noticed some additional issues. To help you understand the issues, I will explain my test setup for the de-embedding and show the results. I have developed three different test structures in CST for the validation of the de-embedding. For these three structures, I have modeled the respective 2x-Thru. As a DUT structure, I modeled a Beatty structure alone in CST to compare the results of the de-embedding with it. 

**These are my three structures:**

[image]
[image]
[image]

I also produced these structures and then applied the de-embedding with practical measurements. 

**These are the results of the de-embedding in Time Domain:**

In all structures, it can be seen that there are irregularities on the left side in the time domain. Additionally, after the DUT structure ends in time, the impedance is not normalized to 100 ohms on the right side. This is present in all structures in both the simulation and measurement data from ZC but not present in NZC.

[image]
[image]
[image]

Because the impedance on the right side with ZC does not fall back to 100 ohms, the NZC is still better. The irregularities on the left side also degrade the ZC, making the NZC partially better than the ZC. Normally, the ZC should be better because it replicates the actual impedance of the FDF structure through the peeling calculation method.

**Results in the Frequency Domain:**

In the frequency domain, the NZC has better agreement in the simulation data than the ZC. In the measurement data, the ZC is slightly worse than the NZC at low frequencies and slightly better than the ZC at higher frequencies.

FDF1:
[image]
FDF2:
[image]
FDF3:
[image]

Here I have shown the absolute error of the de-embedded DUT structures compared to the originally simulated DUT structure in CST. The absolute deviation is calculated with: 
[image]

[image]
[image]
[image]

The following shows that the ZC replicates the impedance of the fixtures better than the NZC:

[image]
[image]
[image]

How can the NZC still be better than the ZC? How can the irregularities on the left side in the time domain be explained, and why does the impedance not fall back to 100 ohms? Could this indicate further possible errors in the de-embedding with impedance correction? Or are these results plausible? Another possible assumption is that my S-parameters have causality issues. What is your opinion on this?

**The simulation and measurement data of the MM S parameters are attached here so that you can also implement it yourself:**
[ZC_check.zip](https://github.com/user-attachments/files/15930125/ZC_check.zip)

Thanks!

mhuser, 2024-06-24:
Thanks for these results @Hasko7
The simulated and measured structures look great!
I did not have the time yet to test your dataset, but if you are interested you can test scikit-rf results against the Amphenol's _ACS De-embedding Utility_ to see if the results are similar. It runs the IEEEP370 Matlab code behind the hood (Matlab license is not required to use it).

mhuser, 2024-06-25:
Can you try to disable `leadin` and `NRP_enable` nyquist rate point enforcement? It often add the curious shape on port side.

```Python
dm_mmzc  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru, dummy_fix_dut_fix = se_fdf,
                         bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin = 0, NRP_enable = False, name = 'mmzc')
```

Hasko7, 2024-06-25:
@mhuser my results are already without leadin and NRP_enable, I've already noticed these weird side effects.

Hasko7, 2024-06-25:
I tried Amphenol's ACS De-embedding Utility. The result is completely different and probably incorrect. I had already tested this software in the past with 4 Port MM S-Parameters (.s4p-files) and the results were also very weird. I also made sure to choose the odd/even format. Do I need to consider anything else? Could you perhaps test this with my S-Parameters as well? Here are my simulated and measured S-Parameters again. The file also contains my result from the ACS De-embedding Utility, that is, the de-embedded DUT. The file also contains my result from the ACS De-embedding Utility (wich is the de-embedded DUT from ACS). 
[ZC_check.zip](https://github.com/user-attachments/files/15968142/ZC_check.zip)

mhuser, 2024-06-25:
Hello @Hasko7
Thanks for the dataset.

In the case of simulated structure1, the TDR of 2xThru and FIX_DUT_FIX are not to the reference impedance before the time zero.
Thus, leadin parameter should be used if the fixtures shall embedd the same behaviour and the deembedded dut be close to the simulated one.
I appears the leadin has a bug on left side. If I do a rough correction and use a leadin of 80, the results agreement is not too bad (but it does not reproduce the non-zero reflexion of simulated DUT before time 0):

[image]

I will investigate this further but cannot in the following hours.

Hasko7, 2024-06-25:
Hi @mhuser, all right! Let me know when you have found something. As I said, I also assume that my S-Parameters from the simulation have causality issues. But the points I mentioned are also visible in the measurement data.

mhuser, 2024-06-26:
> As I said, I also assume that my S-Parameters from the simulation have causality issues.

@Hasko7 can you run your test again with one ofthe the simulated data and the latest master branch with `leadin1 = 150, leadin2 = 150`?

It should roughly shift the time domain to a position where the reflexion is zero at time zero, perform the deembedding and unshift the time domain. I think the issue causing a weird left fixture shape when leadin >> 1 is fixed.

Hasko7, 2024-06-27:
Hello @mhuser I ran my test with the simulated data and the latest master branch with leadin1 = 150, leadin2 = 150 with:

```
z0 = 50
# FDF with sharp bend simulation
dm_mmzc_bend  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru_bend, dummy_fix_dut_fix = se_fdf_bend,
                         z0 = z0,  bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin1 = 150, leadin2 = 150, NRP_enable = False, name = 'mmzc sharp bends simulated')

# FDF with curved bend simulation
dm_mmzc_curved  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru_curved, dummy_fix_dut_fix = se_fdf_curved,
                         z0 = z0,  bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin1 = 150, leadin2 = 150, NRP_enable = False, name = 'mmzc curved bends simulated')

# FDF with multiple curved bends simulation
dm_mmzc_curved_multi  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru_curved_multi, dummy_fix_dut_fix = se_fdf_curved_multi,
                         z0 = z0,  bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin1 = 150, leadin2 = 150, NRP_enable = False, name = 'mmzc multiple curved bends simulated')`

```

I don't see any change in the results:
[image]
[image]
[image]

Did you also find out, why the impedance does  not fall back to nominated 100 ohms on the right side of the deembedded DUT with ZC?

mhuser, 2024-06-27:
Hello @Hasko7 
Thank you for your feedback.
Is it possible for various reasons you are not using latest `master` ? Maybe you need to `pip install -e` in your clone.
[image]

Hasko7, 2024-06-28:
Hello @mhuser, I use the latest master version of the repo and also in developer mode with pip install -e. However, my results are still not like yours.
I can also see your merge with fix shiftNPoints in IEEEP370_SE_ZC_2xThru:
[image]
[image]

mhuser, 2024-06-28:
Hello @Hasko7 
This is strange. While your git seems OK, it really looks like your python does not use the right version. This can eventually happens if you use another virtual environment or if another python is called depending the path. 😕 

Does version gives 1.1.0 ? This would not guarantee last commit is used, but you can use the second command to find if the sources are properly located.

```Python
import skrf as rf
import os
print(rf.__version__)
print(os.path.abspath(rf.deembedding.__file__))
```

Hasko7, 2024-06-28:
Hello @mhuser, I have already checked that and used both commands:

```
sys.path.append(os.path.join('scikit-RF'))
import skrf 
from skrf.calibration import (IEEEP370_SE_NZC_2xThru, IEEEP370_MM_NZC_2xThru,
                              IEEEP370_SE_ZC_2xThru, IEEEP370_MM_ZC_2xThru)
skrf.stylely()
print(os.path.dirname(skrf.__file__))
print(os.path.dirname(np.__file__))
print(skrf.__version__)
print(os.path.abspath(skrf.deembedding.__file__))
```
This is my output for this: 
[image]

=> Here you can see that I am using numpy from my venv and skrf from my repo folder, which is scikit-rf.

And this is my folder structure from the scikit-rf repo:
[image]

And this is my venv: 

 (.venv) PS C:\Users\....\....\.....\python_vscode\.venv\Lib\site-packages> ls

    Directory: C:\Users\...\...\...\python_vscode\.venv\Lib\site-packages

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----        15.02.2024     14:53                adodbapi
d-----        15.02.2024     14:54                asttokens
d-----        15.02.2024     14:54                asttokens-2.4.1.dist-info
d-----        15.02.2024     14:54                colorama
d-----        15.02.2024     14:54                colorama-0.4.6.dist-info
d-----        15.02.2024     14:54                comm
d-----        15.02.2024     14:54                comm-0.2.1.dist-info
d-----        15.02.2024     15:13                contourpy
d-----        15.02.2024     15:13                contourpy-1.2.0.dist-info
d-----        15.02.2024     15:13                cycler
d-----        15.02.2024     15:13                cycler-0.12.1.dist-info
d-----        15.02.2024     14:54                dateutil
d-----        15.02.2024     14:53                debugpy
d-----        15.02.2024     14:54                debugpy-1.8.1.dist-info
d-----        15.02.2024     14:53                decorator-5.1.1.dist-info
d-----        15.02.2024     14:53                executing
d-----        15.02.2024     14:53                executing-2.0.1.dist-info
d-----        15.02.2024     15:13                fontTools
d-----        15.02.2024     15:13                fonttools-4.48.1.dist-info
d-----        15.02.2024     14:54                ipykernel
d-----        15.02.2024     14:54                ipykernel-6.29.2.dist-info
d-----        15.02.2024     14:54                IPython
d-----        15.02.2024     14:54                ipython-8.21.0.dist-info
d-----        15.02.2024     14:53                isapi
d-----        15.02.2024     14:54                jedi
d-----        15.02.2024     14:54                jedi-0.19.1.dist-info
d-----        15.02.2024     14:54                jupyter_client
d-----        15.02.2024     14:54                jupyter_client-8.6.0.dist-info
d-----        15.02.2024     14:54                jupyter_core
d-----        15.02.2024     14:54                jupyter_core-5.7.1.dist-info
d-----        15.02.2024     15:13                kiwisolver
d-----        15.02.2024     15:13                kiwisolver-1.4.5.dist-info
d-----        15.02.2024     15:13                matplotlib
d-----        15.02.2024     15:14                matplotlib-3.8.3.dist-info
d-----        15.02.2024     15:13                matplotlib.libs
d-----        15.02.2024     14:54                matplotlib_inline
d-----        15.02.2024     14:54                matplotlib_inline-0.1.6.dist-info
d-----        15.02.2024     15:14                mpl_toolkits
d-----        15.02.2024     14:53                nest_asyncio-1.6.0.dist-info
d-----        15.02.2024     15:10                numpy
d-----        15.02.2024     15:10                numpy-1.26.4.dist-info
[... 45 lines cut ...]
d-----        15.02.2024     14:53                wcwidth
d-----        15.02.2024     14:53                wcwidth-0.2.13.dist-info
d-----        15.02.2024     14:53                win32
d-----        15.02.2024     14:53                win32com
d-----        15.02.2024     14:53                win32comext
d-----        15.02.2024     14:53                zmq
d-----        27.06.2024     21:42                __pycache__
-a----        15.02.2024     14:53          16752 decorator.py
-a----        15.02.2024     14:54            512 ipykernel_launcher.py
-a----        15.02.2024     14:54            155 jupyter.py
-a----        15.02.2024     14:53           7490 nest_asyncio.py
-a----        15.02.2024     15:10              0 numpy-1.26.4-cp311-cp311-win_amd64.whl
-a----        15.02.2024     15:13             90 pylab.py
-a----        15.02.2024     14:53            139 pythoncom.py
-a----        15.02.2024     14:53        2648456 PyWin32.chm
-a----        15.02.2024     14:53            178 pywin32.pth
-a----        15.02.2024     14:53              5 pywin32.version.txt
-a----        15.02.2024     15:10              0 scipy-1.12.0-cp311-cp311-win_amd64.whl
-a----        15.02.2024     14:53          34549 six.py
**-a----        27.06.2024     21:42             89 __editable__.scikit_rf-1.1.0.pth**
**-a----        27.06.2024     21:42           5404 __editable___scikit_rf_1_1_0_finder.py**

I think this must be right, because i have the editable scikit-rf repo.

Somehow, this is strange :-(

mhuser, 2024-06-28:
Oh I am so sorry leadin = 150 and not leadin1 = , leadin2 =, I did write it wrong 😭

Hasko7, 2024-06-28:
Ah okay  😆, now it worked for me, but it takes significantly longer. I am using de-embedding for 3 FDF structures, both simulation and measurement data. It took 8 minutes for both datasets (so the ZC for 3 FDFs took 8 minutes).

These are my results:

FDF1: 
[image]
FDF2:
[image]
FDF3:
[image]

For the simulation data, it is now significantly better, but for the measurement data on the right side of the DUT, the impedance does not drop to the normalized 100 ohms, which worsens the result. What could be the reason for this?

mhuser, 2024-07-01:
@Hasko7 
Thank you for your results. Your setup is a good test for the deembedding algorithm and it already permitted us to fix two issues : the fixture flip and the incorrect leadin shift. 😄 

Now the deembedding works well with the simulated data. For FDF1, we see `IEEEP370_MM_ZC_2xThru` outperform `IEEEP370_MM_NZC_2xThru `because it correct also the impedance before zero.

For the measured data, the `IEEEP370_MM_ZC_2xThru` make a more complex fixture that removes the residuals shapes left by `IEEEP370_MM_NZC_2xThru `. The DUT zone has a reasonnable agreement inbetween NZC and ZC. However, the impedance do not return to reference at the end wit ZC.

Regarding the processing time, it works, but I do not think `leadin ` usually needs to be such a big number. I am not skilled in simulations, but there might be a way to achieve a zero reflection close to time zero in the time domain so you would not need a big `leadin `any more.

I will have a try on my side with your measured data. The deembedding should work well with transmission-line-likes simulated structures. Measured data, however will have more complex behavior such as coaxial to planar transitions. Even if the VNA is well calibrated, there will be energy reflected by connectors and non-linearities in impedance. This causes less energy to arrive to the DUT and the signal-over-noise decreases.  The deembedding cannot enhance this fact very much. However this is not a satisfactory explanation why the deembedding seems to works well and then the impedance has an offset 😅

mhuser, 2024-07-01:
[image]

I am not sure yet, but the Z corrections seems to be less accurate on the right side. (also the measured FIX-DUT-FIX) has an starting impedance that differ from the left side, which is curious)

These can also be plot artifacts caused by the DC extrapolation used to plot in time-domain. I will check the Z correction when possible.

mhuser, 2024-07-02:
Hello @Hasko7,
I ran a test to see if the impedance difference we see on the right is a plot artifacts and it appears it is.
Under the hood, ieee370 use a sophisticated dc extrapolation method that assume the network is like a thru, while the more generic scikit-rf `extrapolate_to_dc()` make no assumption but can be less precise (the time domain is only a display and does not change the S-Parameters exept it add a first point at frequency 0).

[image]

So unfortunately it seems both fixtures agree well with FIX-DUT-FIX and there is no good explanation, except if the beatty standard leak energy by radiation or the cable behind is not well 50 ohm. I tried the AICC tool, but it seemed to return incoherent results.

Just to be sure of your setup : do you use a full 4-port calibration ?

```Python
import skrf as rf
import matplotlib.pyplot as plt
import numpy as np
from numpy import concatenate, zeros, conj, flip, real
from numpy.fft import fftshift, irfft
from scipy.interpolate import interp1d
from skrf.calibration import IEEEP370_MM_NZC_2xThru, IEEEP370_MM_ZC_2xThru
rf.stylely()

def makeStep(impulse):
    #mhuser : no need to call step function here, cumsum will be enough and efficient
    #step = np.convolve(np.ones((len(impulse))), impulse)
    #return step[0:len(impulse)]
    return np.cumsum(impulse, axis=0)

def COM_receiver_noise_filter(f,fr):
    """
    receiver filter in COM defined by eq 93A-20
    """
    f = f / 1e9
    fdfr = f / fr
    # eq 93A-20
    return 1 / (1 - 3.414214 * fdfr**2 + fdfr**4 + 1j*2.613126*(fdfr - fdfr**3))

def DC2(s, f):
    DCpoint = 0.002 # seed for the algorithm
    err = 1 # error seed
    allowedError = 1e-10 # allowable error
    cnt = 0
    df = f[1] - f[0]
    n = len(f)
    t = np.linspace(-1/df,1/df,n*2+1)
    ts = np.argmin(np.abs(t - (-3e-9)))
    Hr = COM_receiver_noise_filter(f, f[-1]/2)
    while(err > allowedError):
        h1 = makeStep(fftshift(irfft(concatenate(([DCpoint], Hr * s)), axis=0), axes=0))
        h2 = makeStep(fftshift(irfft(concatenate(([DCpoint + 0.001], Hr * s)), axis=0), axes=0))
        m = (h2[ts] - h1[ts]) / 0.001
        b = h1[ts] - m * DCpoint
        DCpoint = (0 - b) / m
        err = np.abs(h1[ts] - 0)
        cnt += 1
    return DCpoint

def dc_interp(s, f):
    """
    enforces symmetric upon the first 10 points and interpolates the DC
    point.
    """
    sp = s[0:9]
    fp = f[0:9]

    snp = concatenate((conj(flip(sp)), sp))
    fnp = concatenate((-1*flip(fp), fp))
    # mhuser : used cubic instead spline (not implemented)
    snew = interp1d(fnp, snp, axis=0, kind = 'cubic')
    return real(snew(0))

def add_dc(sin):
[... 67 lines cut ...]
# plot them all
plt.figure(figsize=(12, 4))

plt.subplot(1, 2, 1)
plt.title(f'{se_fdf.name} (1-2 side)')
#mm_dut_dc.s11.plot_z_time_step(window='hamming', color = 'k', linestyle='dashed', label = 'DUT, sdd11')
#mm_d_dut_nzc_dc.s11.plot_z_time_step(window='hamming', color = '0.5', label = 'nzc_DUT, sdd11')
mm_d_dut_zc_dc.s11.plot_z_time_step(window='hamming', color = 'r', label = 'zc_DUT, sdd11')
mm_fdf_dc.s11.plot_z_time_step(window='hamming', color = 'g', label = 'FIX-DUT-FIX, sdd11')
mm_2xthru_dc.s11.plot_z_time_step(window='hamming', color = 'c', label = '2x-Thru, sdd11')
mm_zc_side1_dc.s11.plot_z_time_step(window='hamming', color = 'b', label = 'FIX-1_2, sdd11')
plt.xlim((-0.6, 1.4))
plt.legend(loc='lower right')

plt.subplot(1, 2, 2)
plt.title(f'{se_fdf.name} (3-4 side)')
#mm_dut_dc.s22.plot_z_time_step(window='hamming', color = 'k', linestyle='dashed', label = 'DUT, sdd22')
#mm_d_dut_nzc_dc.s22.plot_z_time_step(window='hamming', color = '0.5', label = 'nzc_DUT, sdd22')
mm_d_dut_zc_dc.s22.plot_z_time_step(window='hamming', color = 'r', label = 'zc_DUT, sdd22')
mm_fdf_dc.s22.plot_z_time_step(window='hamming', color = 'g', label = 'FIX-DUT-FIX, sdd22')
mm_2xthru_dc.s22.plot_z_time_step(window='hamming', color = 'c', label = '2x-Thru, sdd22')
mm_zc_side2_dc.s11.plot_z_time_step(window='hamming', color = 'b', label = 'FIX-3_4, sdd11')
plt.xlim((-0.6, 1.4))
plt.legend(loc='lower right')
```

Hasko7, 2024-07-02:
Hello @mhuser, I have access to very good measurement equipment from Rohde & Schwarz. Before the measurement, I performed a 4-port TOSM calibration (SOLT). Therefore, the calibration should be correct.

Hasko7, 2024-07-02:
Hello @mhuser   I think the impedance correction for the right fixture is correct. However, there is no perfect match visible in your image because the FDF structure is not exactly normalized to 100 ohms, whereas the created FIX-2 structure for the right side is normalized to 100 ohms. Could this perhaps be the problem?

[image]

mhuser, 2024-07-02:
Hello @Hasko7,
The base impedance is mostly determined by the dc point that depends on the extrapolation technique. However, the deembedding is performed without the dc point, so the base impedance is mostly visual for the plot. What is important is the amplitude of the features from the base impedance. In your case you have a kind of step in the impedance.

I finally ran the AICC Deembedding tool by saving renumbered s4p files for the sequential scheme (did not work with odd/even scheme).

[image]

The deembedded results have an excellent agreement with `scikit-rf` results:
[image]

So this time I am afraid it is not an implementation issue.

I wondered if you use a simulation engine that accounts for radiated energy loss (not like FEM in ADS/Empro), in case the beatty standard would act like a kind of patch antenna if its dimension is bigger than lambda/10 at high frequency.

mhuser, 2024-07-03:
I thought a little about this issue. The TDR shows variation of reflexion with time, represented like impedance. On the start of TDR the reflexion should be the impedance of the start of the structure, at the end of the TDR the reflexion is the made from the dc resistance/conductance of the full structure and the load (thus for long cable, TDR often exhibit a rising slope).

If there was radiation or dielectric loss, the impedance would be smaller than reference. So here it is like if we have serie resistance (conductor loss or connector contact).

It will be interesting to test dembedding of a resistor to check this hypothesis.

mhuser, 2024-07-03:
Let's try this with some code:
As we can see with blue curve, dielectric losses cause a negative slope in the transmission line section of TDR and a small negative offset at the end (like a parallel resistor in RLGC model). On red curve, serie resistance add the positive offset at the end.

[image]

```Python
import skrf as rf
import matplotlib.pyplot as plt
from skrf.media import MLine
rf.stylely()

R   = 0.5           # ohm
W   = 3.00e-3       # m
H   = 1.55e-3       # m
T   = 35e-6         # m
L     = 0.16        # ns
ep_r = 4.5
tand = 0.015
f_epr_tand = 1e9    # Hz

freq = rf.Frequency(10e-3,67,6700, 'GHz')

# 50 ohm segment
MSL1 = MLine(frequency=freq, z0_port=50, w=W, h=H, t=T,
        ep_r=ep_r, mu_r=1, rho=1.712e-8, tand=tand, rough=0.15e-6,
        f_low=1e3, f_high=1e12, f_epr_tand=f_epr_tand,
        diel='djordjevicsvensson', disp='kirschningjansen')

# Capacitive segment
MSL2 = MLine(frequency=freq, z0_port=50, w=3*W, h=H, t=T,
        ep_r=ep_r, mu_r=1, rho=1.712e-8, tand=tand, rough=0.15e-6,
        f_low=1e3, f_high=1e12, f_epr_tand=f_epr_tand,
        diel='djordjevicsvensson', disp='kirschningjansen')

# line
l = MSL1.line(L, 'ns')
l = l.extrapolate_to_dc(kind = 'cubic')

# line and serie resistor
lr = MSL1.resistor(R = R) ** MSL1.line(L, 'ns') ** MSL1.resistor(R = R)
lr = lr.extrapolate_to_dc(kind = 'cubic')

# beatty
b = MSL1.line(1.3*L, 'ns') ** MSL2.line(0.5*L, 'ns') ** MSL1.line(1.3*L, 'ns')

# beatty and resistor
br = MSL1.resistor(R = R) **  MSL1.line(1.3*L, 'ns') \
    ** MSL2.line(0.5*L, 'ns') ** MSL1.line(1.3*L, 'ns') ** MSL1.resistor(R = R)
br = br.extrapolate_to_dc(kind = 'cubic')

# plot them all
plt.figure(figsize=(2*6.4, 4.8))

plt.subplot(1, 2, 1)
plt.title('Transmission line')
l.s11.plot_z_time_step(window='hamming', label = 'Lossy TL')
lr.s11.plot_z_time_step(window='hamming', label = 'Serie resistor and lossy TL')
plt.xlim((-0.6, 1.4))
plt.legend(loc='lower right')

plt.subplot(1, 2, 2)
plt.title('Beatty standard')
b.s11.plot_z_time_step(window='hamming', label = 'Beatty standard')
br.s11.plot_z_time_step(window='hamming', label = 'Serie resistor and Beatty standard')
plt.xlim((-0.6, 1.4))
plt.legend(loc='lower right')
plt.savefig('resistor.png', dpi = 200)
```

mhuser, 2024-07-03:
`IEEEP370_MM_ZC_2xThru` fixture model is made of a serie of transmission line sections of different impedances but the same gamma that mimics the FIX-DUT-FIX left and right with a total length of half 2xThru. It cannot embed the behaviour of a series resistor.

Yet you can remove a series resistor using `scikit-rf` and that makes the DUT curve going to reference at the end of TDR. The process of searching resistor value could be automatized but the first question is: is it a series resistance and what would be the cause? This is probably out of the scope of this issue,

[image]

```Python
from skrf.media import DefinedGammaZ0
# resistor
R = 0.4 # ohm
media = DefinedGammaZ0(frequency = se_2xthru.frequency, z0 = 50)
res = media.resistor(R = R)
# concat two resistors (coupled line)
se_res = rf.concat_ports([res, res], port_order='second')
se_res_inv = se_res.inv
# ZC
dm_mmzc  = IEEEP370_MM_ZC_2xThru(dummy_2xthru = se_2xthru, dummy_fix_dut_fix = se_fdf,
                         bandwidth_limit = 67e9, pullback1 = 0, pullback2 = 0,
                         leadin = 1, NRP_enable = False, name = 'mmzc', verbose = False)

se_d_dut_zc = dm_mmzc.deembed(se_fdf)
# remove serie resistance
se_d_dut_zc = se_res_inv ** se_d_dut_zc ** se_res_inv
```

Hasko7, 2024-07-03:
Hello @mhuser I won't be able to respond to you today. I will reply tomorrow, and I think this will make things clearer. 
BR

mhuser, 2024-07-06:
Hello @Hasko7 
I made a summary plot to show the current state of this issue in the case of FDF1 structure (click on it for better resolution).
[image]

### IEEE370 NZC (left column)
#### Fixture (first row)
The left fixture (red) is superposed to the first half of 2xThru (black dashed). This is how NZC algorithm is intended to work, it splits mathematically the s-parameters of the 2xThru into two halves based on a transmission line assumption.

#### Deembedding (second row)
The deembedded DUT (red curve) is coherent with FIX-DUT-FIX (green dotted) with remaining artifacts due to differences between 2xTru and FIX-DUT-FIX launch sections.

Comparison with AICC tool (black dashed) shows a poor agreement on the right section after the Beatty standard. If we force a wrong right fixture flip (magenta), the agreement with AICC is excellent. In conclusion, the mixed-modes NZC implementation of AICC tool version 1.0.0 has a wrong right fixture flip, and using scikit-rf is an advantage here.

With Scikit-rf, the impedance seems to approach the reference impedance to the right of the plot.

### IEEE370 ZC (right column)

#### Fixture (first row)
The left fixture (red) is superposed to the first half of FIX-DUT-FIX (black dashed). This is how ZC algorithm is intended to work, it computes first the length of the fixture being the half of 2xThru. Then it mimics FIX-DUT-FIX by peeling its time-domain point-by-point and rebuilding it using transmission lines sections. The Leadin parameter enables to capture impedance shape before time zero (at the expense of processing time).

#### Deembedding (second row)
The deembedded DUT (red curve) is coherent with FIX-DUT-FIX (green dotted) with little remaining artifacts but it does not approach the reference impedance on the right.

Comparison with AICC tool (black dashed) shows an excellent agreement.

Adding a resistor to the model (blue) "seems" to help to approach the reference on the right (we will see below it is not the case).

## De-zooming time domain 
[image]

If we observe the time domain on a broader interval, there is no offset but rather slopes in measurement curves. The resistor trick only fix the shape on a narrow time interval. Thus it is not that a good idea.

You can see typical shapes on TDR (resistors, etc.) on [this link](https://www.tek.com/en/documents/fact-sheet/tdr-impedance-measurements-foundation-signal-integrity).

mhuser, 2024-07-06:
It does not help a lot, but other sources about TDR measurement exhibit the same non-immediate returning to reference behaviour.
[image]

Such as p. 7 in Anritsu [Time Domain Measurements Using Vector Network Analyzers](https://dl.cdn-anritsu.com/en-us/test-measurement/files/Application-Notes/Application-Note/11410-00722A.pdf)

I wonder if this is a contribution of connector + cables or something else. It does not appear so strong in simulations of transmission lines only.

mhuser, 2024-07-06:
I might be wrong, but when the VNA measure S11, SOLT calibration only correct for the cable between port 1 and DUT. It does not know and care if there is another port connected. The full thru correction only has effect on the measure is between two different port, like S21. This make the length of 2xThru in S21 correct.

For reflexion Sii measurements of transmission lines, there is then an effect of the loss and mismatch of the cable on the non-excited port, that is not calibrated out. 🤔

Hasko7, 2024-07-07:
Hello @mhuser, sorry for the late response. I have read through your comments. Thanks for that! I don't think the influence of the cables can be seen, as I have done a comprehensive 4 port calibration. This eliminates the instrument errors and the losses caused by the cables. It also eliminates the errors and differences between all the ports because you do a through measurement between all the ports. This means that with a full 4 port TOSM calibration (SOLT) you measure a total of 18 times. OSM on all 4 ports makes 12 measurements and 6 through measurements with all port combinations. But the increasing slope may be due to the solderless PCB connectors I used. This is my test board: 
[image]
[image]

I used the these connectors from Rosenberger: 
https://products.rosenberger.com/radio-frequency/connectors/470621/08k80e-40ml5-right-angle-jack-pcb#

And here is a general portfolio of the solderless pcb connectors of RO: 
https://www.rosenberger.com/fileadmin/content/headquarter/Downloads/RF_Coaxial_Connectors/TM_Solderless_PCB_Flyer.pdf

Maybe the rising slope is because of the solderless connectors, but after measuring the structures you also have the influences of the connectors in the measurement data and therefore you can remove them in the deembedding. What is your opinion on this?

I'll show you the current de-embedding results in the FD again. I have already shown the current results in TD above. 

FDF1:
[image]

FDF2:
[image]

FDF3:
[image]

mhuser, 2024-07-07:
@Hasko7 Thank you for the pictures and the plot.
The circuit looks really great!
The agreement already looks quite decent. There is still a little enhancement possible on the upper frequencies of return loss with `NRP_enable = True`  (could remove the final glitch with FDF1 and FDF2).

These connectors seems excellents. You could check if there is no oxidation at the contact with an ohm-meter between center conductors of the connector on a track. The result should me much smaller than 1 ohm.

I am doing a modification of the `verbose = True` to plot the attenuation determination. With `bandwidth_limit = 0` there is no fit and with  `bandwidth_limit = 67e9` the attenuation is fitted with a 2nd-order polynomial. However whether it is raw or fitted has not much influence on the results. 
[image] 
[image]

>after measuring the structures you also have the influences of the connectors in the measurement data and therefore you can remove them in the deembedding.

In theory, yes. However, the deembedding is limited by the model used and by the signal-to-noise ratio. Regarding ZC, there are some effects that can be difficult to model with short transmission lines sections (e. g. if the sections are not short enough compared to the discontinuities). For coupled lines, the assumption of ZC is that there is only common and differential modes for the fixture. With the coupled bends, this may not be 100% true (if we think of a differential signal, there will be a phase shift on a section of the path). Also, behind the strong reflection of the beatty standard, the SNR will be decreased because most of the energy has already bounced back.

Hasko7, 2024-07-08:
Hello @mhuser, I have taken into account that short transmission lines affect the halving in the time domain. I have made a comprehensive analysis according to the guidelines of the IEEE 370 standard. IEEE 370 states that 2x-Thru must have a minimum length in order to provide sufficient resolution in time domain. IEEE 370 says that 2x-Thru should have a minimum length of 3 times the wavelength at the highest frequency. this can be calculated with this formula:
[image]

For my structures, this has resulted in a minimum length of 7.71 mm. I therefore have a straight length of 15 mm for 2x-thru1 and 2x-thru2 without including the transition from wide to close coupling (due to the connectors). 
[image]

Therefore, the discontinuities should not interfere with the bisection or time gating in the time domain for the halfment of the 2x-thru. I also showed the halving of 2x-Thru in the time domain above (in my previous messages) and there you can see that the discontinuities do not interfere and that the structure is sufficiently long. 

Regarding the mode conversion parameter, the IEEE 370 says the following in the Fixture electrical requirements (FER): The difference between differential to common-mode conversion and
insertion loss is lower than -15 dB at all frequencies of interest.
I have also considered this. This is fulfilled for 2x-Thru1 and 2x-Thru2 but not for 2x-Thru3. 
I also looked at the FER of IEEE 370 in general:
[image]

My results were very good. The structures were classified as either class A or B.

Hasko7, 2024-07-08:
I have now also set the NRP to true. I can see an improvement at the higher frequencies in the RL, but I also get stronger spikes at the higher frequencies.

FDF1:
[image]

FDF2:
[image]

FDF3:
[image]

Hasko7, 2024-07-08:
In general, that suits me. ZC is also better, especially with the measured results. Only with regard to the rising slope on the right side of FDF with ZC I have no plausible explanation so far.

mhuser, 2024-07-08:
> Therefore, the discontinuities should not interfere with the bisection or time gating in the time domain for the halfment of the 2x-thru.

I hope I did not give you the impression of underestimating your printed circuit. In fact, when I first saw the measurements, I thought that this is probably the best demonstration circuit I have observed. Everything is clearly visible in the TDR. That helped to see and fix the remaining issues in the scikit-rf implementation and I am grateful for this!

I was more generally speaking of difficulties in modeling sections such as the connector coaxial-to-planar transition, in which dimensions are inside the connector and cannot be changed by the designer. Also, the coupled lines and curved bend of FDF3 add some skew and imbalance that may explain some more differential-to-common mode, but the results remain excellent.

mhuser, 2024-07-08:
> In general, that suits me. ZC is also better, especially with the measured results. Only with regard to the rising slope on the right side of FDF with ZC I have no plausible explanation so far.

Neither have I explanation yet, but I saw in the Rhode & Schwarz application note [Accurate Test Fixture Characterization and De-Embedding](https://www.rohde-schwarz.com/us/applications/accurate-test-fixture-characterization-and-de-embedding-application-note_56280-1271617.html), that they also have the "non-returning to reference" shape in their differential mode measurement TDR.

[image]

Hasko7, 2024-07-08:
Ah okay, if you find out anything else beyond that, let me know. Thanks for the conversation.

mhuser, 2024-07-10:
Hello @Hasko7, you are welcome !

I still do not have the answer yet, but let me share here an experience:
The code below simulate a VNA measurement with the differential 2xThru used in [scikit-rf deembedding example](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/IEEEP370%20Deembedding.html). The DUT is surrounded by four simulated coaxial cables of 0.5 meter and the port of the VNA are well mached or have a small mismatch. The system is calibrated as a post-processing using 4-Port SOLT.

[image]

With the Mighty VNA that has perfect 50-ohm ports, the agreement after calibration is excellent.

[image]

If the VNA ports have some mismatch, there are offsets appearing in the TDR (but not slopes) and ripple at cable length in return loss that are not removed by calibration. Real VNA are not 100% perfect regarding source and load match. Maybe this can cause some imbalance on differential DUT.

My proposition is to close this issue as of now, with some mystery remaining.

```Python
import skrf as rf
import matplotlib.pyplot as plt
from skrf.media import Coaxial, DefinedGammaZ0
from skrf.calibration import MultiportCal, SOLT, IEEEP370_MM_ZC_2xThru
rf.stylely()

# differential 2xThru
se_2xthru = rf.Network('diff_2xthru.s4p')
freq = se_2xthru.frequency

# Pasternack PE-P047HF-BULK 110 GHz
COAX = Coaxial(freq, Dint = 0.29e-3, Dout = 0.92e-3, epsilon_r = 2.1,
               sigma = 6.2e7, tand_delta = 0.02, z0_port = 50)

# cable
cable = COAX.line(0.5, 'm', name = 'PE-P047HF 0.5m cable')
cables = rf.concat_ports([cable, cable], port_order='second')

# VNA mismatched ports
M50 = DefinedGammaZ0(freq)
n_ports = 4
# with inperfect source/load port match
title = 'Mismatched port VNA'
z_ports = [50.5, 49, 51, 49.5]
# mighty VNA with perfect match
# title = 'Mighty Match VNA'
# z_ports = [50, 50, 50, 50]

ports = [M50.line(0, z0 = z) for z in z_ports]

vna_left = rf.concat_ports([ports[0], ports[1]], port_order='second')
vna_right = rf.concat_ports([ports[2], ports[3]], port_order='second')

# VNA and cables
left = vna_left ** cables
left.renormalize(50)
right = vna_right ** cables
right.renormalize(50)

# simulate measurement
se_raw_2xthru = left ** se_2xthru ** right.flipped()
se_raw_2xthru.name = 'Raw measured 2xThru'
# dirty fix
se_raw_2xthru.port_modes = ['S'] * 4

# ideals
short_ideal = COAX.short(nports = 2, name = 'short')
open_ideal = COAX.open(nports = 2, name = 'open')
load_ideal = COAX.match(nports = 2, name = 'load')
thru_ideal = COAX.thru(name = 'thru')
ideals = [short_ideal, open_ideal, load_ideal, thru_ideal]

# measured
measured_01 = [
    ports[0] ** cable ** i ** cable.flipped() ** ports[1].flipped() for i in ideals]

    
measured_02 = [
    ports[0] ** cable ** i ** cable.flipped() ** ports[2].flipped() for i in ideals]
    
measured_03 = [
    ports[0] ** cable ** i ** cable.flipped() ** ports[3].flipped() for i in ideals]

# calibration
cal_dict = {(0, 1): {'method': SOLT, 'ideals': ideals, 'measured': measured_01},
            (0, 2): {'method': SOLT, 'ideals': ideals, 'measured': measured_02},
            (0, 3): {'method': SOLT, 'ideals': ideals, 'measured': measured_03},
            }

cal = MultiportCal(cal_dict)
cal.run()
se_cal_2xthru = cal.apply_cal(se_raw_2xthru)

# mixed-modes conversion
mm_2xthru = se_2xthru.copy()
mm_2xthru.se2gmm(p = 2)
mm_raw_2xthru = se_raw_2xthru.copy()
mm_raw_2xthru.se2gmm(p = 2)
mm_cal_2xthru = se_cal_2xthru.copy()
mm_cal_2xthru.se2gmm(p = 2)

# add dc
mm_2xthru_dc = IEEEP370_MM_ZC_2xThru.extrapolate_to_dc(mm_2xthru)
mm_2xthru_dc.name = 'ideal 2x Thru'
mm_raw_2xthru_dc = IEEEP370_MM_ZC_2xThru.extrapolate_to_dc(mm_raw_2xthru)
mm_raw_2xthru_dc.name = 'Raw measured 2xThru'
mm_cal_2xthru_dc = IEEEP370_MM_ZC_2xThru.extrapolate_to_dc(mm_cal_2xthru)
mm_cal_2xthru_dc.name = '4-Port SOLT calibrated measured 2xThru'

# plot them all
plt.subplots(1, 2, figsize=(2*6.4, 4.8))

plt.subplot(1, 2, 1)
plt.title('Time Domain')
mm_raw_2xthru_dc.s11.plot_z_time_step(window='hamming', linestyle = 'dotted')
mm_cal_2xthru_dc.s11.plot_z_time_step(window='hamming')
mm_2xthru_dc.s11.plot_z_time_step(window='hamming', color = 'k', linestyle = 'dashed')

plt.xlim((-1, 10))
plt.legend(loc='upper right')

plt.subplot(1, 2, 2)
plt.title('Frequency Domain')
mm_raw_2xthru_dc.plot_s_db(0, 0, linestyle = 'dotted')
mm_cal_2xthru_dc.plot_s_db(0, 0)
mm_2xthru_dc.plot_s_db(0, 0, color = 'k', linestyle = 'dashed')
plt.ylim((-50, 0))
plt.text(1e9, -2.5, 'Port match')
for i, z in enumerate(z_ports):
    plt.text(1e9, -5 - i * 2.5, f'P{i + 1} = {z} ohm')
plt.legend(loc='upper right')

plt.suptitle(title)
plt.tight_layout()
plt.savefig('calibration.png', dpi = 100)
```

Ttl, 2024-07-10:
The calibrated measurement should be identical to the ideal one. VNA port impedances and all other errors are corrected in the calibration. The reason why your example doesn't match is that measurement port impedances are not equal (`print(measured_01[0].z0)`). SOLT calibration uses a mix of S-parameter direct manipulation, which don't consider port z0 mismatch, and network cascading, which does consider it, resulting in incorrect result. Renormalizing measurements or replacing `ports = ...` with `ports = [M50.impedance_mismatch(50, z) for z in z_ports]`  results in equal port impedances in measurements and exact match with calibrated and ideal measurements.

Hasko7, 2024-07-10:
@mhuser, @Ttl thank you for the message. But is there a practical alternative to the 4-Port SOLT calibration for differential line structures (coupled transmission lines) when performing a measurement with a VNA followed by the 2x-Thu de-embedding? Because my measurement data was taken with good measurement equipment and calibration equipment. Both are from R&S. Or is there no alternative to remove the mismatch through calibration and you have to do post-processing to remove the port mismatch? Is that common?

mhuser, 2024-07-10:
Hello @Ttl , Hello @Hasko7 
You are right @Ttl thanks you, I did forgot the renormalization in the measurement simulation (there is a mismatch but the reference impedance of the network must be remormalized to 50 ohm). If I correct this, the SOLT remove perfectly source and load match, as per its error model.
@Hasko7 this example with synthetic measurements shows that the 4-Port SOLT and port mismatch works well. The intention was to verify if the calibration properly removed the VNA mismatch + cable contribution, and it does. Therefore if there is some imbalance, it should be on the measured network.

[image]

```Python
import skrf as rf
import matplotlib.pyplot as plt
from skrf.media import Coaxial, DefinedGammaZ0
from skrf.calibration import MultiportCal, SOLT, IEEEP370_MM_ZC_2xThru
rf.stylely()

# differential 2xThru
se_2xthru = rf.Network('diff_2xthru.s4p')
freq = se_2xthru.frequency

# Pasternack PE-P047HF-BULK 110 GHz
COAX = Coaxial(freq, Dint = 0.29e-3, Dout = 0.92e-3, epsilon_r = 2.1,
               sigma = 6.2e7, tand_delta = 0.02, z0_port = 50)

# cable
cable = COAX.line(0.5, 'm', name = 'PE-P047HF 0.5m cable')

# VNA mismatched ports + cables
M50 = DefinedGammaZ0(freq)
n_ports = 4
# with inperfect source/load port match
title = 'Mismatched port VNA'
z_ports = [50.5, 49, 51, 49.5]
# mighty VNA with perfect match
# title = 'Mighty Match VNA'
# z_ports = [50, 50, 50, 50]

ports = []
for z in z_ports:
    p = M50.line(0, z0 = z) ** cable
    p.renormalize(50)
    ports.append(p)

left  = rf.concat_ports([ports[0], ports[1]], port_order='second')
right = rf.concat_ports([ports[2], ports[3]], port_order='second')

# simulate measurement
se_raw_2xthru = left ** se_2xthru ** right.flipped()
se_raw_2xthru.name = 'Raw measured 2xThru'
# dirty fix
se_raw_2xthru.port_modes = ['S'] * 4

# ideals
short_ideal = COAX.short(nports = 2, name = 'short')
open_ideal = COAX.open(nports = 2, name = 'open')
load_ideal = COAX.match(nports = 2, name = 'load')
thru_ideal = COAX.thru(name = 'thru')
ideals = [short_ideal, open_ideal, load_ideal, thru_ideal]

# measured
measured_01 = [
    ports[0] ** i ** ports[1].flipped() for i in ideals]

    
measured_02 = [
    ports[0] ** i ** ports[2].flipped() for i in ideals]
    
measured_03 = [
    ports[0] ** i ** ports[3].flipped() for i in ideals]

# calibration
cal_dict = {(0, 1): {'method': SOLT, 'ideals': ideals, 'measured': measured_01},
            (0, 2): {'method': SOLT, 'ideals': ideals, 'measured': measured_02},
            (0, 3): {'method': SOLT, 'ideals': ideals, 'measured': measured_03},
            }

cal = MultiportCal(cal_dict)
cal.run()
se_cal_2xthru = cal.apply_cal(se_raw_2xthru)

# mixed-modes conversion
mm_2xthru = se_2xthru.copy()
mm_2xthru.se2gmm(p = 2)
mm_raw_2xthru = se_raw_2xthru.copy()
mm_raw_2xthru.se2gmm(p = 2)
mm_cal_2xthru = se_cal_2xthru.copy()
mm_cal_2xthru.se2gmm(p = 2)

# add dc
mm_2xthru_dc = IEEEP370_MM_ZC_2xThru.extrapolate_to_dc(mm_2xthru)
mm_2xthru_dc.name = 'ideal 2x Thru'
mm_raw_2xthru_dc = IEEEP370_MM_ZC_2xThru.extrapolate_to_dc(mm_raw_2xthru)
mm_raw_2xthru_dc.name = 'Raw measured 2xThru'
mm_cal_2xthru_dc = IEEEP370_MM_ZC_2xThru.extrapolate_to_dc(mm_cal_2xthru)
mm_cal_2xthru_dc.name = '4-Port SOLT calibrated measured 2xThru'

# plot them all
plt.subplots(1, 2, figsize=(2*6.4, 4.8))

plt.subplot(1, 2, 1)
plt.title('Time Domain')
mm_raw_2xthru_dc.s11.plot_z_time_step(window='hamming', linestyle = 'dotted')
mm_cal_2xthru_dc.s11.plot_z_time_step(window='hamming')
mm_2xthru_dc.s11.plot_z_time_step(window='hamming', color = 'k', linestyle = 'dashed')

plt.xlim((-1, 10))
plt.legend(loc='upper right')

plt.subplot(1, 2, 2)
plt.title('Frequency Domain')
mm_raw_2xthru_dc.plot_s_db(0, 0, linestyle = 'dotted')
mm_cal_2xthru_dc.plot_s_db(0, 0)
mm_2xthru_dc.plot_s_db(0, 0, color = 'k', linestyle = 'dashed')
plt.ylim((-50, 0))
plt.text(1e9, -2.5, 'Port match')
for i, z in enumerate(z_ports):
    plt.text(1e9, -5 - i * 2.5, f'P{i + 1} = {z} ohm')
plt.legend(loc='upper right')

plt.suptitle(title)
plt.tight_layout()
plt.savefig('calibration.png', dpi = 100)
```

mhuser, 2024-07-10:
Unfortunately, I do not have a simulation element  "coupled line with small imbalance" to test against. So I am afraid we shall end this endeavour there 😄

jhillairet, 2024-07-10:
> [...]. The reason why your example doesn't match is that measurement port impedances are not equal (`print(measured_01[0].z0)`). SOLT calibration uses a mix of S-parameter direct manipulation, which [...]

For reference, fixed in #1121 by @Ttl

mhuser, 2024-07-14:
@Hasko7 I am closing this as of now, but feel free to write again in case of new discoveries.
Thank you for conversation and sharing results that enabled to fix existing issues in scikit-rf IEEE 370 implementations !
Your printed circuit and plots look great.

closed 2024-07-14 by mhuser