scikit-rf/scikit-rf #1090: Error with NanoVNA if frequency npoints = 1

mxxmxn, 2024-06-07:
**Describe the bug**
When I try to set the frequency to one specific point, I get an error.

**To Reproduce**
Run the following example with a NanoVNA:
```
instr.frequency = Frequency(start=2.4, stop=2.4, npoints=1, unit='GHz')
```

**Expected behavior**
I should be able to set the frequency.

**System**
 - OS: Linux
 - scikit-rf version: 1.0.0

FranzForstmayr, 2024-06-18:
Can you post the exact error message? I don't have access to a NanoVNA.

mxxmxn, 2024-06-20:
The issue can be reproduced without a NanoVNA, the device string only needs to be a valid serial interface:
```
import skrf
from skrf.vi.vna.nanovna import NanoVNA

instr = NanoVNA("ASRL/dev/tty1::INSTR")
freq = skrf.Frequency(start=2.4, stop=2.4, unit='GHz', npoints=1)
instr.frequency = freq
```

I get the following Error:
```
skrf/frequency.py:412: RuntimeWarning: invalid value encountered in scalar divide
  return self.span/(self.npoints-1.)
Traceback (most recent call last):
  File "scratches/scratch.py", line 6, in <module>
    instr.frequency = freq
    ^^^^^^^^^^^^^^^
  File "skrf/vi/vna/nanovna/nanovna.py", line 155, in frequency
    self.write(OP.WRITE8, REG_ADDR.SWEEP_STEP, 8, f.step)
  File "skrf/vi/vna/nanovna/nanovna.py", line 74, in write
    arg = int(arg).to_bytes(nbytes, byteorder="little", signed=False)
          ^^^^^^^^
ValueError: cannot convert float NaN to integer
```

closed 2024-06-28 by mxxmxn