scikit-rf/scikit-rf #1106: About Circuit voltages

Asachoo, 2024-06-27:
### Discussed in https://github.com/scikit-rf/scikit-rf/discussions/1105

<div type='discussions-op-text'>

<sup>Originally posted by **Asachoo** June 26, 2024</sup>
Hello,

I have a question about voltage in a Circuit. There's a Network with an unusual reference impedance (`z0` not equal to 50 Ohms). Should a different voltage be calculated in this case?

I noticed that the Network has the same S-parameters for the export ports, and when all `z0` values are the same, the voltages at the serial ports are equal. However, when the `z0` of one of the Networks changes, the calculated voltage results also change. The following example and output better express my question.

Is this an issue, or should the voltages actually change with Network.z0?

Thanks in advance!

```python
import numpy as np

import skrf as rf

def gen_random_s(npoints: int, nports: int = 2) -> np.ndarray:
    s = 2 * np.random.rand(npoints, nports, nports) * 1.0j - 1
    s += 2 * np.random.rand(npoints, nports, nports) - 1
    return s

np.random.seed(123)
npoints = 1
freq = rf.Frequency(1, 100, npoints, "ghz")

porta = rf.Circuit.Port(frequency=freq, name="porta")
portb = rf.Circuit.Port(frequency=freq, name="portb")

ntwka = rf.Network(frequency=freq, s=gen_random_s(npoints), name="ntwka")
ntwkb = rf.Network(frequency=freq, s=gen_random_s(npoints), name="ntwkb")

ntwkc = ntwkb.copy()
ntwkc.renormalize(40)

for ntw in (ntwkb, ntwkc):
    circuit = [
        [(porta, 0), (ntwka, 0)],
        [(ntwka, 1), (ntw, 0)],
        [(ntw, 1), (portb, 0)],
    ]

    circuit = rf.Circuit(circuit)
    print(f"Result for z0: {ntw.z0}:")
    print(circuit.s_external, "S-parameters in external port")

    print(circuit.voltages([1, 0], [0, 0]), "Voltage across ports")
    print()
```
And the result are shown in blow:
```bash
Result for z0: [[50.+0.j 50.+0.j]]:
[[[-0.21995821+1.22760113j -0.25338403-1.11897886j]
  [-0.27041083-0.0546119j   0.99939488+1.82918502j]]] S-parameters in external port
[[ 7.80041793+12.27601126j  7.80041793+12.27601126j
  -1.44312993 +1.28197898j -1.44312993 +1.28197898j
  -2.70410829 -0.546119j   -2.70410829 -0.546119j  ]] Voltage across ports

Result for z0: [[40.+0.j 40.+0.j]]:
[[[-0.21995821+1.22760113j -0.25338403-1.11897886j]
  [-0.27041083-0.0546119j   0.99939488+1.82918502j]]] S-parameters in external port
[[ 7.80041793+12.27601126j  7.80041793+12.27601126j
  -1.76283288 +1.28713896j -1.57672566 +1.15125208j
  -2.68903881 -0.54307558j -3.00643678 -0.60717696j]] Voltage across ports
```
</div>

Ttl, 2024-06-27:
Outputting the circuit network shows that the network port impedances and S-parameters for the both cases are identical, but calculated voltages are different.

```
import numpy as np

import skrf as rf

def gen_random_s(npoints: int, nports: int = 2) -> np.ndarray:
    s = 2 * np.random.rand(npoints, nports, nports) * 1.0j - 1
    s += 2 * np.random.rand(npoints, nports, nports) - 1
    return s

np.random.seed(123)
npoints = 1
freq = rf.Frequency(1, 100, npoints, "ghz")

porta = rf.Circuit.Port(frequency=freq, name="porta")
portb = rf.Circuit.Port(frequency=freq, name="portb")

ntwka = rf.Network(frequency=freq, s=gen_random_s(npoints), name="ntwka")
ntwkb = rf.Network(frequency=freq, s=gen_random_s(npoints), name="ntwkb")

ntwkc = ntwkb.copy()
ntwkc.renormalize(40)

for ntw in (ntwkb, ntwkc):
    circuit = [
        [(porta, 0), (ntwka, 0)],
        [(ntwka, 1), (ntw, 0)],
        [(ntw, 1), (portb, 0)],
    ]

    circuit = rf.Circuit(circuit)
    print(f"Result for z0: {ntw.z0}:")
    n = circuit.network
    print(n.z0, 'Z0')
    print(n.s, "S-parameters in external port")

    print(circuit.voltages([1, 0], [0, 0]), "Voltage across ports")
    print()
```

```
Result for z0: [[50.+0.j 50.+0.j]]:
[[50.+0.j 50.+0.j]] Z0
[[[-0.21995821+1.22760113j -0.25338403-1.11897886j]
  [-0.27041083-0.0546119j   0.99939488+1.82918502j]]] S-parameters in external port
[[ 7.80041793+12.27601126j  7.80041793+12.27601126j
  -1.44312993 +1.28197898j -1.44312993 +1.28197898j
  -2.70410829 -0.546119j   -2.70410829 -0.546119j  ]] Voltage across ports

Result for z0: [[40.+0.j 40.+0.j]]:
[[50.+0.j 50.+0.j]] Z0
[[[-0.21995821+1.22760113j -0.25338403-1.11897886j]
  [-0.27041083-0.0546119j   0.99939488+1.82918502j]]] S-parameters in external port
[[ 7.80041793+12.27601126j  7.80041793+12.27601126j
  -1.76283288 +1.28713896j -1.57672566 +1.15125208j
  -2.68903881 -0.54307558j -3.00643678 -0.60717696j]] Voltage across ports
```

Clearly both can't be true at the same time. Circuit Port object does have z0 property so that defaults to 50 ohms, so it's calculated voltages that aren't correct. It seems that calculating voltages (and currents) doesn't consider the port impedances correctly.

Asachoo, 2024-06-28:
It appears that different reference impedance of Ports result in varying S-parameters for circuits. Similarly, different reference impedance for Networks can lead to incorrect Voltage/Current distribution in connections.

To accurately calculate the voltage/current in a circuit, the S-parameters need to be "de-normalized (I don't know if this statement is reasonable.)" to address the issue of port reference impedance. Additionally, the impedance between each pair of connected nodes (networks) within a Circuit object must be considered to resolve the impedance mismatch between internal ports.

To illustrate this, I created a more complex example where the reference impedance of the three networks differ, thereby highlighting the impedance issues between internal ports.

```python
import numpy as np

import skrf as rf

def gen_random_s(npoints: int, nports: int = 2) -> np.ndarray:
    s = 2 * np.random.rand(npoints, nports, nports) * 1.0j - 1
    s += 2 * np.random.rand(npoints, nports, nports) - 1
    return s

np.random.seed(123)
npoints = 1
freq = rf.Frequency(1, 100, npoints, "ghz")

porta = rf.Circuit.Port(frequency=freq, name="porta")
portb = rf.Circuit.Port(frequency=freq, name="portb")

ntwka = rf.Network(frequency=freq, s=gen_random_s(npoints), name="ntwka")
ntwkb = rf.Network(frequency=freq, s=gen_random_s(npoints), name="ntwkb")
ntwkc = rf.Network(frequency=freq, s=gen_random_s(npoints), name="ntwkc")

ntwkx = ntwka.copy()
ntwky = ntwkb.copy()
ntwkz = ntwkc.copy()

ntwkx.renormalize(40)
ntwky.renormalize(30)
ntwkz.renormalize(20)

for ntw1, ntw2, ntw3 in (
    (ntwka, ntwkb, ntwkc),
    (ntwkx, ntwky, ntwkz),
):
    circuit = [
        [(porta, 0), (ntw1, 0)],
        [(ntw1, 1), (ntw2, 0)],
        [(ntw2, 1), (ntw3, 0)],
        [(ntw3, 1), (portb, 0)],
    ]

    circuit = rf.Circuit(circuit)
    print(f"Result for z0: {ntw1.z0[0]}, {ntw2.z0[0]}, {ntw3.z0[0]}:")
    print(circuit.s_external, "S-parameters in external port")

    print(circuit.currents([1, 0], [0, 0]), "Current across ports")
    print(circuit.voltages([1, 0], [0, 0]), "Voltage across ports")
    print()
```
I seem to have fixed the problem of different matching between internal ports in https://github.com/scikit-rf/scikit-rf/pull/1107, and the output of the above case in that PR is:
```bash
Result for z0: [50.+0.j 50.+0.j], [50.+0.j 50.+0.j], [50.+0.j 50.+0.j]:
[[[-0.29645462+1.1632858j   0.20393204+0.02071413j]
  [ 0.03676862-0.12222617j -0.64527139+0.63791598j]]] S-parameters in external port
[[-0.25929092+0.23265716j  0.25929092-0.23265716j -0.06306092-0.0103588j
   0.06306092+0.0103588j   0.03483   -0.01447254j -0.03483   +0.01447254j
  -0.00735372+0.02444523j  0.00735372-0.02444523j]] Current across ports
[[ 7.03545381+11.63285797j  7.03545381+11.63285797j
  -2.11274857 +1.88695305j -2.11274857 +1.88695305j
  -0.35352955 -0.32933193j -0.35352955 -0.32933193j
   0.36768618 -1.22226169j  0.36768618 -1.22226169j]] Voltage across ports

Result for z0: [40.+0.j 40.+0.j], [30.+0.j 30.+0.j], [20.+0.j 20.+0.j]:
[[[-0.29645462+1.1632858j   0.20393204+0.02071413j]
  [ 0.03676862-0.12222617j -0.64527139+0.63791598j]]] S-parameters in external port
[[-0.25929092+0.23265716j  0.25929092-0.23265716j -0.06306092-0.0103588j
   0.06306092+0.0103588j   0.03483   -0.01447254j -0.03483   +0.01447254j
  -0.00735372+0.02444523j  0.00735372-0.02444523j]] Current across ports
[[ 7.03545381+11.63285797j  7.03545381+11.63285797j
  -2.11274857 +1.88695305j -2.11274857 +1.88695305j
  -0.35352955 -0.32933193j -0.35352955 -0.32933193j
   0.36768618 -1.22226169j  0.36768618 -1.22226169j]] Voltage across ports
```
However, when the ports are inconsistent, the output current/voltage is still problematic.

Asachoo, 2024-06-28:
A circuit with different reference impedances for Port/Network is built to check the current/voltage calculation results.
```python
import skrf as rf

ntwk = rf.Network('ntwkb40.s2p')
line = rf.media.DefinedGammaZ0(frequency=ntwk.frequency)

port1 = rf.Circuit.Port(frequency=ntwk.frequency, z0=50, name='port1')
port2 = rf.Circuit.Port(frequency=ntwk.frequency, z0=30, name='port2')

connections = [
    [(port1, 0), (ntwk, 0)],
    [(ntwk, 1), (port2, 0)]
]

circuit = rf.Circuit(connections)
print("Voltages:",circuit.voltages([1,0], [0,0]))
```
[Corresponding SNP file](https://github.com/user-attachments/files/16023879/ntwkb40.zip). The output and verification are as follows:
```bash
Voltages: [[-3.15719706+16.85731147j -3.15719706+16.85731147j
  -6.72272615 +8.74371818j -6.72272615 +8.74371818j]]
```
[image]

Therefore, it can be considered that the correct voltage/current results can be obtained when the reference impedance of the `Port` is different.

closed 2024-07-09 by Asachoo

Asachoo, 2024-07-09:
https://github.com/scikit-rf/scikit-rf/pull/1107 is merged now, and this issue has been resolved if reference impedances are pure real numbers.