scikit-rf/scikit-rf #1110: Test failing to capture warning

NGC2023, 2024-06-30:
Hi,

There is a test error while testing in Debian autopkgtest testsuite.

```
=================================== FAILURES ===================================
________________________ NetworkTestCase.test_nf_circle ________________________

self = <test_network.NetworkTestCase testMethod=test_nf_circle>

    def test_nf_circle(self):
        # Check whether the noise figure circle agrees with that calculated with Microwave Office
        nf_circle_mwo = np.loadtxt(os.path.join(self.test_dir, 'nf_circle_mwo.csv'), encoding='utf-8',
                                           delimiter=',')
    
        assert np.allclose(
            self.ntwk_noise["1GHz"].nf_circle(nf=1.0, npoints=6).flatten().real,
            nf_circle_mwo[:6,0],
        )
        assert np.allclose(
            self.ntwk_noise["1GHz"].nf_circle(nf=1.0, npoints=6).flatten().imag,
            nf_circle_mwo[:6,1],
        )
    
        assert np.allclose(
            self.ntwk_noise["2GHz"].nf_circle(nf=2.0, npoints=6).flatten().real,
            nf_circle_mwo[6:12,0],
        )
        assert np.allclose(
            self.ntwk_noise["2GHz"].nf_circle(nf=2.0, npoints=6).flatten().imag,
            nf_circle_mwo[6:12,1],
        )
    
        # Check whether an error is raised when the network is not 2 port.
        net = rf.Network(f=[1], s=np.eye(3), z0=50)
        with pytest.raises(ValueError):
            net.nf_circle(nf=1.0)
    
        # Check whether an error is raised when the number of points is not positive.
        with pytest.raises(ValueError):
            net.nf_circle(nf=1.0, npoints=0)
    
        # Check whether an error is raised when the network is missing noise data.
        with pytest.raises(ValueError):
            self.ntwk1.nf_circle(nf=1.0, npoints=0)
    
        # Check whether the specified noise figure is too small.
>       with pytest.raises(RuntimeWarning):
E       Failed: DID NOT RAISE <class 'RuntimeWarning'>

tests/test_network.py:1944: Failed
```

I have created a patch to address this failure,

```patch
--- a/skrf/tests/test_network.py
+++ b/skrf/tests/test_network.py
@@ -711,14 +711,14 @@ class NetworkTestCase(unittest.TestCase)
 
     @pytest.mark.skip(reason="not supporting this function currently ")
     def test_connect_fast(self):
-        self.assertEqual(rf.connect_fast(self.ntwk1, 1, self.ntwk2, 0) , \
+        self.assertEqual(rf.connect(self.ntwk1, 1, self.ntwk2, 0) , \
             self.ntwk3)
 
         xformer = rf.Network()
         xformer.frequency = rf.Frequency(1, 1, 1, unit='GHz')
         xformer.s = ((0,1),(1,0))  # connects thru
         xformer.z0 = (50,25)  # transforms 50 ohm to 25 ohm
-        c = rf.connect_fast(xformer,0,xformer,1)  # connect 50 ohm port to 25 ohm port
+        c = rf.connect(xformer,0,xformer,1)  # connect 50 ohm port to 25 ohm port
         self.assertTrue(np.all(np.abs(c.s-rf.impedance_mismatch(50, 25)) < 1e-6))
 
     def test_flip(self):
@@ -1941,9 +1941,16 @@ class NetworkTestCase(unittest.TestCase)
             self.ntwk1.nf_circle(nf=1.0, npoints=0)
 
         # Check whether the specified noise figure is too small.
-        with pytest.raises(RuntimeWarning):
+        with warnings.catch_warnings(record=True) as w:
+            warnings.simplefilter("always")
             self.ntwk_noise['1GHz'].nf_circle(nf=0.1)
 
+            # Check that a warning was raised
+            assert len(w) > 0, "Expected a warning to be raised"
+
+            # Check that the warning is a RuntimeWarning
+            assert any(item.category == RuntimeWarning for item in w), "Expected RuntimeWarning was not raised"
+
     def test_de_embed_by_floordiv(self):
         ntwk_result_1 = self.ntwk1 // self.ntwk2
         ntwk_result_2 = self.ntwk1 // (self.ntwk2)

```

This patch includes fix for 'DeprecationWarning: connect_fast is deprecated.'

FranzForstmayr, 2024-07-18:
Is there a way to reproduce this locally?

NGC2023, 2024-07-19:
I should have provided more information on reproducibility. You can reproduce this issue locally in Debian unstable in the chroot environment within the autopkgtest suite. Additionally, it can be reproduced outside the chroot environment in Debian unstable while running the autopkgtest suite.

FranzForstmayr, 2024-08-21:
I tried to fix all the applied debian patches in #1158.

I think the changes at `def test_connect_fast` are not necessary, as this test should get skipped.

closed 2024-08-27 by FranzForstmayr