scikit-rf/scikit-rf #1111: Improve Time Domain Gating technique ?

jhillairet, 2024-07-01:
According to the following paper recently published:

A. Zohner, F. Irnstorfer, R. Weigel and C. Pfannenmüller, "Performance Evaluation of a New Universal Time Domain Gating Algorithm," 2024 IEEE International Instrumentation and Measurement Technology Conference (I2MTC), Glasgow, United Kingdom, 2024, pp. 1-6, doi: 10.1109/I2MTC60896.2024.10560830.
https://ieeexplore.ieee.org/abstract/document/10560830

In the paper, the authors benchmark their TDG technique for the following test case with scikit-rf and obtained worse results than with their proposed method. 
[image]

[image]

Vinc0110, 2024-07-02:
We've had several issues or questions regarding those edge effects of the time gating method at the band limits. 

In our implementation, there are large difference between the selected gating method (parameter `method='convolution/fft/rfft'`) and, in case of convolution mode, also the frequency extension (parameter `conv_mode='reflect/constant/nearest/mirror/wrap'`. I did not read the paper (no access), but I have a feeling they only tried the default parameters...

https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.time.time_gate.html

Vinc0110, 2024-07-02:
In [this old issue](https://github.com/scikit-rf/scikit-rf/issues/757#issuecomment-1264458216_) we did many tests with the different paramters.

mhuser, 2024-07-02:
I had the same first feeling. But then I also considered that it is positive that scikit-rf is the reference !

It does not suppose our algorithms are bad, but rather that scikit-rf is widely available, known and all our implementations are open-source, so ideal to check against.

We can then absorb the ehancements and become even better 😄