scikit-rf/scikit-rf #1115: IEEEP370 example:  import problem from skrf.calibration.

yuming-gmx, 2024-07-08:
Hello together,

i am quite new in scikit-rf and just following the application examples "IEEE370" deembedding.

i have one small problem:  as i tried to  import "IEEEP370_SE_NZC_2xThru" from skrf.calibration, there comes a error as below:
 
maybe someone could give some suggestions, how i could import the module. 

thanks in advance!
08.07.2024

from skrf.calibration import IEEEP370_SE_NZC_2xThru
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[5], line 2
      1 import skrf as rf
----> 2 from skrf.calibration import IEEEP370_SE_NZC_2xThru
      4 thru_result = meas['P1-MSL_Thru_100-P2']
      5 thru_result

ImportError: cannot import name 'IEEEP370_SE_NZC_2xThru' from 'skrf.calibration' (C:\Users\d****\.conda\envs\rf_EMC\lib\site-packages\skrf\calibration\__init__.py)

mhuser, 2024-07-08:
Hello @yuming-gmx 

Can you print
```Python
import skrf 
print(skrf.__version__)
```
And verify the output is 1.1.0 ?

yuming-gmx, 2024-07-08:
> Hello @yuming-gmx
> 
> Can you print
> 
> ```python
> import skrf 
> print(skrf.__version__)
> ```
> 
> And verify the output is 1.1.0 ?

hallo mhuser,
output is "0.18.0".
this is the version, which i get using "conda update.."

mhuser, 2024-07-08:
The package is now on conda-forge. The command shall be like: `conda update scikit-rf -c conda-forge`

yuming-gmx, 2024-07-08:
> The package is now on conda-forge. The command shall be like: `conda update scikit-rf -c conda-forge`

hello mhuser,
many thanks to you. 

i  have tried to update skrf to version 1.1.0, and now i have a new problem:
already by "import skrf as rf", man has already an error:

yuming-gmx, 2024-07-08:
hallo everyone, 
problem solved.

Error #1:   ImportError: cannot import name 'IEEEP370_SE_NZC_2xThru' from 'skrf.calibration'
solution:   update the skrf to update version using  "conda update scikit-rf -c conda-forge".

after the update, one new error comes:
Error #2:  "AttributeError: module 'numpy.typing' has no attribute 'NDArray' "
solution:   update numpy;   if it does works, uninstall numpy and install it again.

thanks "mhuser" again..

here is the

closed 2024-07-08 by yuming-gmx

mhuser, 2024-07-08:
You are welcome !