scikit-rf/scikit-rf #1131: Weird result when de-embedding bad transmission line using IEEEP370_SE_ZC_2XThru

shengfang2017, 2024-07-23:
Hi guys,

I got weird result when de-embedding a bad transmission line. If changes to good transmission line, the result is reasonable and close to DUT. 

This is the fixture TDR result, it has big capacitor at the very front.  
[image]

After de-embedding, the DUT TDR of ZC results: 

[image]

S-parameter results are:

[image]

NZC can get a roughly close result to dut, but ZC is nowhere close to it,  gets positive result in S11.

Here is the s2p files for 2XThru, DUT, and FDF. 
[s2p.zip](https://github.com/user-attachments/files/16352884/s2p.zip)

shengfang2017, 2024-07-23:
forgot to pin @mhuser, I believe you are the expert of IEEEP370

mhuser, 2024-07-24:
Hello @shengfang2017 

Thank you for your report and dataset.

I do not know if I am an expert, but looking to your data I am not surprised ZC does not works. There is a key difference between `NZC` and `ZC`, it does not just correct for the impedance.

- `NZC` S-parameters bisection is done by time gating S11 and S22, taking the
    proper square root of the S21 corrected by return loss, and remixing the
    parameters according to the fixture signal flow graph. This method gives
    crude results but is robust.
- `ZC` computes the length of the fixtures by halving the length of
    2x-Thru and also extract the propagation constant gamma from it. It then peels iteratively FIX-DUT-FIX by determining start
    impedance in time-domain, deembedding a one time sample long transmission
    line of this impedance and loop again until the fixtures mimic FIX-DUT-FIX
    sides impedance profile.

Not all fixtures are properly modeled by a cascade of short transmission lines with different impedance and the same propagation constant gamma, especially when Z is far from reference and gamma vary along the path. This may led `ZC` to gives terrible results.

In other words, `ZC` works only with a **good transmission line**. Your 2xThru fail most of the IEEE370 fixture electrical requirement, thus deembedding it may get you in troubles.

Here we can test it by passing also 2xThru both as 2xThru and FIX-DUT-FIX and test the residuals of deembedding the fixtures from it which should be 0 dB and 0 deg.

[image]
 
Variation of results depending of determined gamma:
```Python
dm = IEEEP370_SE_ZC_2xThru(dummy_2xthru = s2xthru, dummy_fix_dut_fix = s2xthru,
                        bandwidth_limit = 0, verbose = True, NRP_enable = False, leadin = 0)
```
[image]
[image]
[image]

```Python
dm = IEEEP370_SE_ZC_2xThru(dummy_2xthru = s2xthru, dummy_fix_dut_fix = s2xthru,
                        bandwidth_limit = 21e9, verbose = True, NRP_enable = False, leadin = 0)
```
[image]
[image]
[image]

mhuser, 2024-07-24:
@shengfang2017 
Can you share more details about how is the bad transmission line made?

mhuser, 2024-07-24:
Removing some of the fixture in the middle seems to enhance the results. Is the line made of multiple boards connected or it is just aliasing in the time domain?
```Python
dm = IEEEP370_SE_ZC_2xThru(dummy_2xthru = s2xthru, dummy_fix_dut_fix = s2xthru, pullback1 = 18, pullback2 = 18,
                        bandwidth_limit = 40e9, verbose = False, NRP_enable = False, leadin = 0)
```

[image]

shengfang2017, 2024-07-24:
Hi @mhuser, thanks for the detail explanation. I see the problem now. The big cap is caused by an unskilled PCB layout. 

Here is the top layer metal of brd file in Allegro, everything looks good: 
[image]

Turning on Inner 2 etch shows where the mistake is.  You can clearly see the underfill on inner 2 GND under where the connector resides on the 2 experiment lines
[image]

mhuser, 2024-07-24:
Oops. Thank you for these insights!

I am adding the final touch to #1114 With this PR you will be able to plot the 2x-Thru against IEEE 370 limit lines and see the gamma determination in ZC through `verbose = True`.

shengfang2017, 2024-07-24:
Thanks @mhuser, it is really helpful!

mhuser, 2024-07-25:
You are welcome @shengfang2017 
An example of how to use the new methods to plot fixture electrical requirements can be found in the notebook:
https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/IEEEP370%20Deembedding.html

It is available on the master branch until the next scikit-rf release.
I close this issue as of now, but feel free to write again in case of need.

closed 2024-07-25 by mhuser