scikit-rf/scikit-rf #1154: ynp file type support

ChaseRSN2, 2024-08-12:
**Is your feature request related to a problem? Please describe.**
I wanted to use this package for my RF work, but currently it seems like only snp file type is supported. The package would also need to have some support for ynp file types to be relevant for what I was trying to use it for.

**Describe the solution you'd like**
Support for ynp file type. First by allowing for Network class to take ynp file as input to extract the frequency points and y-parameters and store them in the network object. Later on support for plotting y-parameter data as well.

FranzForstmayr, 2024-08-13:
Do you have an example File?

The Touchstone spec is aware of Y Parameters, so this should work. If your files are names ynp the parses will probably fail, however snp and ts files with correct admitance Header Definition should work.

ChaseRSN2, 2024-09-11:
So would just changing the extension from yNp to sNp resolve the issue? I do not have a y1p I can share at the moment (can create one if needed) but an example header would be **# GHz Y RI R 1**

ChaseRSN2, 2024-09-11:
This seems to be working for me, but it would still be more convenient to accept files with extension yNp to avoid writing code to copy the file and change its extension.

Since the header is what is used to parse the data, is it safe to say that in all cases the file extension does not actually matter for parsing the data (besides the needing to be sNp or ts for the parsing to actually work) ?

FranzForstmayr, 2024-09-11:
I think we could add yNp to the collection of allowed formats, although I haven't seen any other tool which uses this extension. Can you name me one?

 > Since the header is what is used to parse the data, is it safe to say that in all cases the file extension does not actually matter for parsing the data (besides the needing to be sNp or ts for the parsing to actually work) ?

This is actually true for all file formats except sNp, as the number of ports is encoded in the filename and unfortunately not in the header.

ChaseRSN2, 2024-09-11:
MATLAB yparameters() method allows for yNp file types. I am currently trying to rewrite some MATLAB code in Python.

closed 2024-09-18 by FranzForstmayr