scikit-rf/scikit-rf #1155: error in the def load_file(self, fid: typing.TextIO)

ShilinBai, 2024-08-16:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**
When I use skrf. network. Network (s_path) to process the ts2.0 file obtained from SIwave (with the txt file suffix changed to. ts, which is the file I am using), an error occurs at skrf \ io \ touchstone. py ", line 524, with the error message ValueError: cannot reshape array of size 27800 into shape (21, new axis)

**To Reproduce**
Steps to reproduce the behavior:
import skrf
path = "C:/temp/siwave.ts"
network = skrf.network.Network(path)

**Expected behavior**
A clear and concise description of what you expected to happen.
I hope to be able to read it successfully so that I can proceed with further processing

**Data**
If applicable, add the data leading to the issue (a minimum example is appreciated)
[siwave.txt](https://github.com/user-attachments/files/16632829/siwave.txt)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**System**
 - OS: Windows-10-10.0.19045-SP0
 - scikit-rf version：Version: 1.2.0

**Additional context**
Add any other context about the problem here if relevant.

FranzForstmayr, 2024-08-16:
This is an duplicate of issue #1146, which is already fixed through #1149.

This fix is currently in master branch but not released on pypi yet. For a quick work around you could try to save your data in full matrix format (don't know how this is done in SIWave) or touchstone 1.0 format, aka snp.

ShilinBai, 2024-08-16:
Okay, thank you very much for your reply

closed 2024-08-16 by ShilinBai