scikit-rf/scikit-rf #1159: Unexpected behavior in Circuit with network ports not explicitly connected

DanielSank, 2024-08-21:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**

When constructing a `Circuit`, there is unexpected behavior with terminals that have nothing connected.

* If a terminal never appears in the connections list, then skrf adds a resistor (i.e. adds a port).
* On the other hand, if we list the terminal but with no connections:
    ```
    connections = [
        [(my_network, port_index),],
        etc.
    ]
    ```
    this somehow signals to skrf to really leave that port open.

**To Reproduce**

Compare this
```
frequency = skrf.Frequency(2, 5, 1000, unit="GHz")  # whatever
port = skrf.Circuit.Port(frequency=frequency, name="port", z0=50)
media = DefinedAEpTandZ0(frequency=frequency, z0=50, tanD=0)
tline = media.line(1E-12, unit="s", name="line")
connections = [
        [(port, 0), (tline, 0)],
]
network = skrf.Circuit(connections).network
print(network.s[:, 0, 0])
```
with this
```
frequency = skrf.Frequency(2, 5, 1000, unit="GHz")  # whatever
port = skrf.Circuit.Port(frequency=frequency, name="port", z0=50)
media = DefinedAEpTandZ0(frequency=frequency, z0=50, tanD=0)
tline = media.line(1E-12, unit="s", name="line")
connections = [
        [(port, 0), (tline, 0)],
        [(tline, 1),],  # <--- this line is new
]
network = skrf.Circuit(connections).network
print(network.s[:, 0, 0])
```

They give different results.

**Expected behavior**

I expect either
1. the behavior illustrated above to be described in `Circuit.__init__`, or
2. Those two cases should behave the same.

**Data**
N/A

**Screenshots**
N/A

**System**
 - OS: Linux
 - scikit-rf version: 0.32.0

**Additional context**

N/A

Asachoo, 2024-08-23:
**Thank you so much for bringing this up!**

Previously, I manually added an `media.open()` object to handle terminal port opens. Your issue made me realize there's actually a more efficient way to implement port opens in circuits.

After some consideration, I believe the unexpected behavior could be due to the following reasons:

1. `S-parameters` are defined under the assumption of **port-matched loads**. Without additional settings, each port defaults to a matched state (typically 50 Ohms), which corresponds to your observation that `"adds a resistor" (i.e., adds a port)`. 
    > They differ from these, in the sense that S-parameters do not use open or short circuit conditions to characterize a linear electrical network; instead, [**matched loads**](https://en.wikipedia.org/wiki/Impedance_matching) are used.

3. Circuit calculations in `scikit-rf` are based on the formula from [P. Hallbjörner, Microw. Opt. Technol. Lett. 38, 99 (2003)](https://onlinelibrary.wiley.com/doi/abs/10.1002/mop.10983). Formula (1) $\left[X_{nn}\right]^k=\frac{2}{Z_{n}Y_{k}}-1$ describes the reflection coefficient for intersections. When a port is listed without any other terminals connected, it can be considered $$n=1$$. According to the formula, the **reflection coefficient becomes 1**, which is equivalent to having an `open()` object on the listed port.

Therefore, I believe it’s reasonable that those two cases don’t behave the same, and this should be preserved to adhere to the underlying the algorithm. The best approach, as you suggested in your first point, is to document this behavior in the `Circuit.__init__` method.

jhillairet, 2024-08-25:
Thank you @DanielSank for reporting. As explained by @Asachoo, the behavior is "normal", while not documented explicitly. The documentation has been clarified by @Asachoo in #1161 (Than you!)

closed 2024-08-25 by jhillairet