scikit-rf/scikit-rf #1174: scikit-rf fails on aarch64

hpretl, 2024-09-09:
Using `scitkit-rf` version 1.3.0 I get the following error on import:
```
/foss/designs > python
Python 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import skrf

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.1.1 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "<stdin>", line 1, in <module>
  File "/headless/.local/lib/python3.10/site-packages/skrf/__init__.py", line 11, in <module>
    from . import (
  File "/headless/.local/lib/python3.10/site-packages/skrf/calibration/__init__.py", line 18, in <module>
    from . import calibration, calibrationSet, deembedding
  File "/headless/.local/lib/python3.10/site-packages/skrf/calibration/calibration.py", line 109, in <module>
    from ..io.touchstone import read_zipped_touchstones
  File "/headless/.local/lib/python3.10/site-packages/skrf/io/__init__.py", line 31, in <module>
    from .general import *
  File "/headless/.local/lib/python3.10/site-packages/skrf/io/general.py", line 80, in <module>
    from pandas import DataFrame, ExcelWriter, Series
  File "/usr/local/lib/python3.10/dist-packages/pandas/__init__.py", line 49, in <module>
    from pandas.core.api import (
  File "/usr/local/lib/python3.10/dist-packages/pandas/core/api.py", line 28, in <module>
    from pandas.core.arrays import Categorical
  File "/usr/local/lib/python3.10/dist-packages/pandas/core/arrays/__init__.py", line 1, in <module>
    from pandas.core.arrays.arrow import ArrowExtensionArray
  File "/usr/local/lib/python3.10/dist-packages/pandas/core/arrays/arrow/__init__.py", line 5, in <module>
    from pandas.core.arrays.arrow.array import ArrowExtensionArray
  File "/usr/local/lib/python3.10/dist-packages/pandas/core/arrays/arrow/array.py", line 64, in <module>
    from pandas.core.arrays.masked import BaseMaskedArray
  File "/usr/local/lib/python3.10/dist-packages/pandas/core/arrays/masked.py", line 60, in <module>
    from pandas.core import (
  File "/usr/local/lib/python3.10/dist-packages/pandas/core/nanops.py", line 52, in <module>
    bn = import_optional_dependency("bottleneck", errors="warn")
  File "/usr/local/lib/python3.10/dist-packages/pandas/compat/_optional.py", line 135, in import_optional_dependency
    module = importlib.import_module(name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.10/dist-packages/bottleneck/__init__.py", line 7, in <module>
    from .move import (move_argmax, move_argmin, move_max, move_mean, move_median,
AttributeError: _ARRAY_API not found
```

The issue seems to be due to Bottleneck failing with numpy>2.*:
```
>>> import bottleneck

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.1.1 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/bottleneck/__init__.py", line 7, in <module>
    from .move import (move_argmax, move_argmin, move_max, move_mean, move_median,
AttributeError: _ARRAY_API not found
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/bottleneck/__init__.py", line 7, in <module>
    from .move import (move_argmax, move_argmin, move_max, move_mean, move_median,
ImportError: numpy.core.multiarray failed to import
```

Questions:
- Are you using bottleneck?
- If yes, can you throw it out to reduce dependencies?

In principle, bottleneck report this (via pipdeptree):
```
Bottleneck==1.4.0
└── numpy [required: Any, installed: 2.1.1]
```

hpretl, 2024-09-09:
Opened an issue with Bottleneck as well: https://github.com/pydata/bottleneck/issues/457

Asachoo, 2024-09-10:
It looks like `bottleneck` is listed as an optional dependency for `pandas`, which is typically used to boost performance on certain operations. Under the `[project.optional-dependencies]` section, we can see that `bottleneck` is included in the performance-related optional dependencies.
https://github.com/pandas-dev/pandas/blob/main/pyproject.toml
However, upon inspection of `scikit-rf`, no direct usage or invocation of the `bottleneck` package was identified.

FranzForstmayr, 2024-09-10:
I think we can close this one.
@hpretl have you tried to install older version of numpy manually? `pip install numpy<2`?

closed 2024-09-10 by FranzForstmayr