scikit-rf/scikit-rf #1184: Difference in S-parameters of lossy line compared to AWR

gvcallen, 2024-10-10:
I am having issues coming to terms with the difference in S-parameters I am getting between scikit's Media.line method and a transmission line formulated using the TLINP element in AWR.

The line's have the same S-parameters when there is no loss, however as soon as loss is introduced (using non-zero tan-delta), the results are different. I believe the same happens when loss is introduced through conductive losses.

Although the differences are small for an individual line, they prove to ultimately have a catastrophically large effect in the overall S-parameters for a project I am working on.

I have attached a minimal code sample below, along with the AWR project, data and the graphs I generated using the code sample. I can generate more AWR data, if needed. 

`
import skrf as rf
from skrf.media.definedAEpTandZ0 import DefinedAEpTandZ0
import matplotlib.pyplot as plt

f = rf.Frequency(10, 500, 1001, 'MHz')
skrf_network = DefinedAEpTandZ0(frequency=f, z0=75, tanD=0.5, z0_port=50).line(1000, 'mm', name='skrf')
awr_network = rf.Network('awr_1000mm_75ohm_tanD0p50.s2p', name='awr')

skrf_network.plot_s_db(m=0, n=0)
awr_network.plot_s_db(m=0, n=0)
plt.savefig('S11.png')
plt.show()

skrf_network.plot_s_db(m=1, n=0)
awr_network.plot_s_db(m=1, n=0)
plt.savefig('S21.png')
plt.show()
`

[image]

[image]

[skrf_vs_awr.zip](https://github.com/user-attachments/files/17325604/skrf_vs_awr.zip)

gvcallen, 2024-10-10:
Update: I believe AWR uses the 'djordjevicsvensson' model when tan-delta is specified at all i.e. there is no option to have tan-delta without this model. I'm not 100% sure of this yet, though, and am still investigating, as I am still unable to match the graphs

mhuser, 2024-10-10:
Hello @gvcallen 
This is a bit strange indeed. However, to my knowledge, this is the first time a comparison has been made with AWR.

What are the values of the parameters marked with yellow in AWR ?
[image]

It is indeed possible to use the `'djordjevicsvensson'` model in scikit-rf too, but it does not seems to help with the agreement.

```Python
m = DefinedAEpTandZ0(frequency=f, z0=75, ep_r=1.0, tanD=0.5, z0_port=50,
                     f_low=1e3, f_high=1e12, f_Ep=1e9,
                     model='djordjevicsvensson')
skrf_network = m.line(1000, 'mm', name='skrf')
```

gvcallen, 2024-10-11:
I have attempted to fit the measured data of a PCB circuit using scikit-rf, however cannot get as good results as in AWR. I am attaching the Python file, as well as measured data, below, as this might be an easier problem to tackle than trying to match AWR directly.

The measured data is of a 75mm open-circuited microstrip stub on FR-4, exposed using an SMA connector. The measurement is from 10 MHz to 1 GHz. The AWR circuit (shown in awr_circuit.png in the .zip) matches the data very well. This was done using AWR's built-in "Robust Simplex" optimizer, with the cost function being the sum of the absolute value of the differences of the db magnitudes of the circuit and measured data.

I have implemented a python script to attempt to fit this data in scikit-rf. The script fits the data using scipy's "Nelder-Mead" optimizer, and the plots the data before and after optimization, alongside the AWR data. As you can see, although scikit's model is close, it is not nearly as exact.

I have tried the fit in scikit-rf both with and without djordjevicsvensson, and in AWR, however with no success. Further, in AWR, `IsEpsFreqDep` was `No`.

Let me know if anyone has any ideas as to what AWR could be doing better.

[sma_microstrip_open.zip](https://github.com/user-attachments/files/17344873/sma_microstrip_open.zip)

mhuser, 2024-10-12:
Hello @gvcallen 

> I am working on the REACH project, where it is very important to accurately model the loss in our antenna's balun

Sounds exciting!

I did some comparison yesterday and got some results:
[image]

In skrf, without frequency dispersion, if the line has the same impedance as the ports, it shows the ideal behavior with constant $\epsilon_r$ and `tanD` (blue traces). If it is renormalized from 75 0hm to 50 ohm, the values change at low frequencies (red traces).

Your AWR data shows a similar behavior but that does not agree well. Especially the $\epsilon_r$ does not reach the 1.0 value within the frequency range (green curves).

Could you provide s-parameters data for the same line in AWR but surrounded by 75 ohm ports for comparison's sake? This would helps to determine if the raw line parameters are different or if it is the mismatch with the ports that differs.

Comparison script:
[awr_tlinp_vs_skrf.zip](https://github.com/user-attachments/files/17350405/awr_tlinp_vs_skrf.zip)

mhuser, 2024-10-14:
Hello @gvcallen,
This is interesting! Maybe this presentation could also help to grasp the idea of the wideband-deby model:
https://www.simberian.com/Presentations/Tutorial_A_Material_World_Final.pdf

[image]

However, the AWR data, especially the $\epsilon_r$ does not look like that dispersion. So there might be something else here, especially because the $\epsilon_r$ is tagged as frequency independent (which it does not really looks like).

Behaviour in Python (log frequency axis):

[image]

```Python
import skrf as rf
from skrf.media.definedAEpTandZ0 import DefinedAEpTandZ0
import numpy as np
import matplotlib.pyplot as plt

f_djsv = rf.Frequency(1e2, 1e15, 10001, 'Hz', sweep_type='log')
m_djsv = DefinedAEpTandZ0(frequency=f_djsv, z0=50, ep_r=4.25, tanD=0.02, z0_port=50,
                     f_low=1e4, f_high=1e13, f_Ep=1e9,
                     model='djordjevicsvensson')

fig = plt.figure()
ax_ep_r = fig.add_subplot(2, 1, 1)
ax_ep_r.set_xscale('log')
ax_tand = fig.add_subplot(2, 1, 2)
ax_tand.set_xscale('log')
ax_ep_r.plot(f_djsv.f_scaled, np.real(m_djsv.ep_r_f),
             label = "f_min=1e4, f_max=1e13, f_ep=1e9, ep_r=4.25, tand=0.02")
ax_tand.plot(f_djsv.f_scaled, m_djsv.tand_f)
ax_ep_r.set_ylabel(r'$\epsilon_{r,eff}$')
ax_ep_r.legend(loc="upper right")
ax_ep_r.grid(True, which='both', ls="-")
ax_ep_r.set_ylim((3.5, 6))
ax_tand.set_xlabel(f'Frequency ({f_djsv.unit})')
ax_tand.set_ylabel('tanD')
ax_tand.grid(True, which='both', ls="-")
fig.suptitle('Djordjevic and Svensson Dispersion')
fig.tight_layout()
fig.savefig('djordjevicsvensson.png')
```

mhuser, 2024-10-14:
> Perhaps we are dealing with two problems here... however the fact that AWR implements a roll-off "by default" is rather interesting as that doesn't exactly fit the definition of an ideal line (though this option might still be useful in skrf)

Indeed. I usually worked above 100 MHz, but some deviations to the standard transmission line model appear when the frequency is close to DC: The conductor loss is no more represented by the skin effect and becomes the conductor bulk resistance, and so on and so forth. Probably there is no tanD at DC too and only a neglectable (at least for printed circuit materials) effect of the substrate insulation resistance.

gvcallen, 2024-10-14:
Yes that makes sense for me too.

At the end of the day, I am modelling a microstrip-like line in my circuit, so will maybe investigate just using the MLine function in my model. However, our operating frequency is between 50 and 150 MHz (where this skin effect "transition" seems take place), so picking the right model will be critical.

I will read up more on the Debye model, as well as the microstrip models in skrf, and get back to you.

Thanks for the help thus far!

mhuser, 2024-10-14:
Hello @gvcallen 
I did achieve promising results by changing your cost function to:
```Python

def cost(x, network_measured) -> float:
    params = array_to_params(x)
    network_sk = sma_microstrip_open(params)

    s11_fit_db = network_sk.s_db[:, 0, 0]
    s11_measured_db = network_measured.s_db[:, 0, 0]
    s11_db_diff = s11_fit_db - s11_measured_db

    cost = np.sum(s11_db_diff**2)
    print(f'cost = {cost}')
    return cost
```
In your code `s11_fit_db = 10*np.log10(s11_fit)` war returning a complex number. It should have been  `s11_fit_db = 10*np.log10(np.abs(s11_fit))`.
[image]

I also implemented a rough bump function to mimic AWR low-frequency behaviour so I now need to clean the code to see if it is still useful.

mhuser, 2024-10-14:
OK, I removed the thing with a smooth transition to tand = 0 towards DC and skrf frequency-invariant model still seems to give a reasonable agreement with the measured data. However the sma optimized `z0` is a bit high, so I had to increase the bound. Also changed the inductance to nH, but it does not seem to have much influence.

[image]
[sma_microstrip_open_modified_cost.zip](https://github.com/user-attachments/files/17369154/sma_microstrip_open_modified_cost.zip)

gvcallen, 2024-10-15:
@mhuser thanks for all the help! I will investigate further (e.g. I need to make sure the phase fits as well) but this looks promising!

mhuser, 2024-10-15:
@gvcallen you are welcome!
Attached you shall find a script that plots magnitude and phase residuals. It is possible to weight a phase contribution in the cost function but the results are not so good with the magnitude when both aspects are considered.

[image]

[sma_microstrip_open_with_phase.zip](https://github.com/user-attachments/files/17378822/sma_microstrip_open_with_phase.zip)

```Python
def cost(x, network_measured) -> float:
    params = array_to_params(x)
    network_sk = sma_microstrip_open(params)
    # magnitude
    s11_fit_db = network_sk.s_db[:, 0, 0]
    s11_measured_db = network_measured.s_db[:, 0, 0]
    s11_db_diff = s11_fit_db - s11_measured_db
    #phase
    s11_fit_deg = network_sk.s_deg[:, 0, 0]
    s11_measured_deg = network_measured.s_deg[:, 0, 0]
    s11_deg_diff = s11_fit_deg - s11_measured_deg
    # weights may need some tuning
    cost = np.sum(np.abs(s11_db_diff)) + 2.5e-3 * np.sum(np.abs(s11_deg_diff))
    print(f'cost = {cost}')
    return cost
```

mhuser, 2024-10-16:
Just to be sure : is the AWR open a perfect open and not a small capacitance? I am pretty sure it should be, but they also stated the $\epsilon_r$ was frequency invariant while it was not at low frequency.

Tought it is not documented how they did it, it make sense the line model could change from a lumped RLGC model to the distributed model when frequency increase from DC. Because at DC $R=0$ (unless a parameter is added stating otherwise), the tanD of $R$ and $X_C$ will be zero too.

gvcallen, 2024-10-17:
@mhuser yes I've tried the cost weighting method in the past but never achieved as good results. The AWR results were optimized simply on minimizing the real and imaginary differences and achieve good magnitude and phase response in my eyes.

The AWR model was just an open - I have added some edge-capacitance in the past but it doesn't really help much at these frequencies.

I have uploade a zip with a photo of the circuit, the circuit parameters, and an .s1p and .s2p for the full model (S11) and only the transmission line in that model (two-port). Note that the optimization is slightly different (better) from what I previously sent, but ultimately was done on only the real and imaginary parts of S11.

[sma_microstrip_open_awr.zip](https://github.com/user-attachments/files/17407636/sma_microstrip_open_awr.zip)

mhuser, 2024-10-21:
After comparisons of a suitable `DefinedAEpTandZ0` line with the `awr_tlinp.s2p` you sent, it seems that S11 and S22 magnitudes are close within ± 0.006 dB and S22 phase is close within 0.05° which is good. Only S11 phase is out by some degrees. 
[image]

This has however a significant impact on the reflection response with an open.
If a dirty fix is used to fix the S11 phase, then the network with the open looks better.
[image]

Even if the S11 magnitude with the open looks terrible without the fix, the absolute deviation is small (± 0.04 dB). It is only because the S11 of the artifact is within 0.5 dB that it seems big considering the relative deviation.

Yet, it is not satisfactory that AWR gives better results.

Considering the optimization, it would be possible to reduce the number of parameters, e.g. by fixing the physical length of the microstrip to the value of the layout and let the dielectric constant only be optimized. Another possibility would be to measure two lines of different lengths with SMA connectors on both ends to [characterize the microstrip](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Correlating%20microstripline%20model%20to%20measurement.html#Dielectric-effective-relative-permittivity-extraction-by-multiline-method) effective dielectric constant and the losses without the connector's effects. Of course, it is beautiful to pack everything in a single optimization, especially if AWR can successfully do it and the circuit is quite simple.
[image]

With the dirty fix (script attached below) the results of skrf are closer to AWR. I have to think more about that. Especially it would not work if the line properties are slightly changed. It is as if the return loss of the line would be non-perfectly matched when the port impedances are equal to the characteristic impedance of the line in AWR while the insertion loss is coherent. But this gives surprisingly a better agreement with the measurement.
[sma_microstrip_open.zip](https://github.com/user-attachments/files/17452516/sma_microstrip_open.zip)

gvcallen, 2024-10-21:
@mhuser thanks for all the work on this so far. 

I am unfortunately away from my office until the 30th, so won't be able to correspond until then, but will get back on this as soon as I'm at my PC again. 

Talk soon

jhillairet, 2024-10-21:
Just see this comment. 
@mhuser shall we wait before releasing a new version?

mhuser, 2024-10-22:
> @mhuser shall we wait before releasing a new version?

Hello @jhillairet 
I think we should not stall on this.
I have done more trials on this, and if a small imaginary part is added to the line's characteristic impedance, the results look very close to AWR. I'd like to know if they implement a kind of impedance dispersion behind the hood to mimic the behavior of microstrips.
So far I have compared the results of `DefinedAEpTandZ0` results + `renormalize()` to the hyperbolic function model mismatched lossy line in `IEEEP370.makeTL`and they give the same results if the same `gamma` and impedances are used. There is no evidence that `DefinedAEpTandZ0` is not giving the expected results, even if it is not the same as AWR.

```Python
DefinedAEpTandZ0(
[...]
z0=microstrip_z0 * (1. + 0.1j * microstrip_epr * microstrip_tanD)
[...]
```
[image]

jhillairet, 2024-10-28:
> I think we should not stall on this.

Sure, but in practice, how can we know what is implemented in AWR ? 

Do you want to make a PR for the next release (this week?) ?

mhuser, 2024-10-28:
> Sure, but in practice, how can we know what is implemented in AWR ?
> 
That is the problem. The doc I have seen did not explain this. The fix with the complex impedance (while it was promising because for the microstrip we already know ADS do this by using complex $\epsilon_r$) does not works for the 75 ohm line example. It probably works with this specific case only by luck.

> Do you want to make a PR for the next release (this week?) ?
>
No because I did unfortunately not find the root cause yet. I should maybe see if I can get an AWR microwave office demo and test various cases (and also the microstrip vs `media.Mline`), but I am somehow running short of time.

gvcallen, 2024-11-06:
@mhuser thanks for all the work on this. I am back at the office now and will be fully on this. Will have a look at what you have done and let you know if I have any further updates

gvcallen, 2024-11-08:
A number of updates before going into the weekend @mhuser , after lots of investigation:
- It seems our previous observations that AWR applies some sort of "roll-of" to ep_r and tanD are incorrect. The error seems to lie in the method we are using to calculate the phase constant beta, which effectively contradicts that used by AWR (as explained below). Using Matlab's s2rlgc function, I get near-constant values for the dielectric constant, loss tangent etc. across frequency for the AWR line (besides some numerical noise) .
- Your observation that S11 phase led me to finding what I believe is the underlying issue, though I am still trying to make sense of it. I will post results/data next week but need to think on it for a bit. If I create a matched (50ohm), 100mm line in AWR with any tanD and with no conductor loss, I get a non-zero S11 value, whereas, in scikit-rf, a matched line has no reflection at its ends (S11 = S22 = 0 for all frequencies). I am not sure which is more correct - at first I would assume having S11 be 0 makes the most sense, since a lossy 50 ohm line is perfectly matched but simply absorbs power as a voltage wave moves from one port to the other, however AWR doesn't seem to think so, and has some quite interesting shapes for S11 for both dielectric and conductor losses.
- Some further investigation shows that, although beta and alpha for this 100mm line are EXACTLY the same from skrf and from AWR, the equivalent RLGC parameters are not. I am trying to research if this is a different formulation for lossy lines (e.g. based on ABCD or y-parameters), but ultimately it seems like the approach taken by AWR is some other, probably equally valid, mathematical formulation.

I am digging a bit into Pozar right now, but am running low on energy so will have to continue with this next week :)

gvcallen, 2024-11-11:
Update: The problem is indeed that the characteristic impedance needs to be modeled as complex. If I set conductor losses to zero, and set the characteristic impedance in scikit to Z0 * (1.0 / sqrt(1 - 1j*tanD)), the results match perfectly to AWR! Typical that the solution is relatively simple!

The only issue now is to derive a general formula to also include the conductor losses, which I am currently working on. I also stumbled on this [write-up](https://bpb-us-w2.wpmucdn.com/sites.gatech.edu/dist/4/463/files/2015/06/TLloss08.pdf?bid=463) from Djordevic which describes this exact problem.

mhuser, 2024-11-11:
> Update: The problem is indeed that the characteristic impedance needs to be modeled as complex. If I set conductor losses to zero, and set the characteristic impedance in scikit to Z0 * (1.0 / sqrt(1 - 1j*tanD)), the results match perfectly to AWR. Typical that the end solution ends up being relatively simple!

Oh great! This sounds good! Have you checked this equation agree with the first 75-ohm line too (if no conductor loss)?

Indeed in the `MLine` (for sure) and `Coaxial` (I believe) and for `DistributedCircuit` (equivalent to RLGC) `Media`, the physical parameters can lead to a characteristic impedance with an imaginary part. However, I am quite surprised it is also the case for `TLINP`, but it is maybe a way to mimic a "physical" line better and not a "simple" transmission line.

There is still the question of the electrical loss influence. If we are not sure of the behavior (perfect match or some S11 for 50 ohm line/50 ohm port) we could use some `compatibility_mode = "awr"` switch in a first time.

gvcallen, 2024-11-11:
@mhuser I also don't mind doing the ADS simulation if necessary as I do have access to that too, however some help would be appreciated. Just to note that a tell-tail sign that Zc is complex is if S11 is not 0.

Also, after further investigation, I have figured out that the constrant AWR is imposing is that sqrt(L/C) = Z0. In other words, it is not the magnitude which they keep equal to Z0. This probably makes the most sense, as it is what the user really means when specifying e.g. a 50 ohm line.

gvcallen, 2024-11-11:
A quick test in ADS on my side reveals that Zc is indeed also complex. Here are the RLGC results from Matlab for AWR vs ADS, one with tanD only and the other with conductor losses (A) only. Clearly it is not a simple problem if neither AWR nor ADS can agree!

The parameters are Z0=50, 100mm, epr=1.0, tanD=0.02, A=0.05, f_A=10 MHz. 

Will continue tomorrow.

[image]

[image]

mhuser, 2024-11-12:
Hello @gvcallen 
Thanks to your RLGC hint and [this source](https://owenduffy.net/transmissionline/concept/mptl.htm), I believe I reached the mighty agreement with your first 75 ohm line. It should be double-checked by setting also conductor loss to something.
[image]

$$Z_n = 75$$
$$R = 2 Z_n \alpha_C$$
$$L = \frac{Z_n \sqrt{\epsilon_r}}{C_0}$$
$$G = \frac{2}{Z_n} \alpha_D$$
$$C = \frac{\sqrt{\epsilon_r}}{C_0 Z_n}$$
$$Z_0 = \sqrt{\frac{R + j\omega L}{G + j\omega C}}$$

```Python
import skrf as rf
from skrf.media.definedAEpTandZ0 import DefinedAEpTandZ0
from scipy.constants import c, pi
import numpy as np
from numpy import abs, angle, log, log10, sqrt, unwrap
import matplotlib.pyplot as plt
rf.stylely()

def compute_ep_r(ntwk, L):
    beta_phase = -unwrap(angle(ntwk.s[:, 1, 0]))
    return np.power(beta_phase * c / (2 * pi * ntwk.frequency.f * L), 2)

def compute_alpha(ntwk, L):
    # because lossless would be abs(S11)**2 + abs(S21)**2 = 1
    alpha = np.abs(ntwk.s[:,1,0]) / (1. - abs(ntwk.s[:,0,0]))
    alpha = (20.0 * log10(alpha)) / -8.686
    return alpha * c / (pi * 1.0 * ntwk.frequency.f) / L
    
freq = rf.Frequency(10, 500, 1001, 'MHz')
f = freq.f
L = 1 # m
Zn = 75 # nominal impedance
ep_r = 1.0
A = 0
f_A = 1e9
tand = 0.5
alpha_c = A * log(10)/20 * sqrt(f/f_A)
alpha_d = pi * sqrt(ep_r) * f / c * tand
R = 2 * Zn * alpha_c
L = Zn * sqrt(ep_r) / c
G = 2 / Zn * alpha_d
C = sqrt(ep_r) / c / Zn
Z0 = sqrt((R + 1j * 2 * pi * f * L) / (G + 1j * 2 * pi * f * C))

m = DefinedAEpTandZ0(frequency=freq, z0=Z0, ep_r=ep_r, tanD=0.5, z0_port=50,
                     f_Ep=1e9, model='frequencyinvariant')

skrf_network = m.line(1000, 'mm', name='skrf')
awr_network = rf.Network('awr_1000mm_75ohm_tanD0p50.s2p')

skrf_ep_r = compute_ep_r(skrf_network, 1.0)
awr_ep_r = compute_ep_r(awr_network, 1.0)

# extract ep_r
fig = plt.figure()
ax = fig.add_subplot(2, 1, 1)
ax.plot(freq.f_scaled, skrf_ep_r, label = skrf_network.name)
ax.plot(freq.f_scaled, awr_ep_r, label = awr_network.name, ls = '--')
ax.set_ylabel(r'$\epsilon_{r,eff}$')
ax.legend()

# extract alpha
skrf_alpha = compute_alpha(skrf_network, L)
awr_alpha = compute_alpha(awr_network, L)

ax = fig.add_subplot(2, 1, 2)
ax.plot(freq.f_scaled, skrf_alpha, label = skrf_network.name)
ax.plot(freq.f_scaled, awr_alpha, label = awr_network.name, ls = '--')
ax.set_xlabel(f'Frequency ({freq.unit})')
ax.set_ylabel(r'$\alpha$ (db/m)')

plt.savefig('tlinp.png')
```

gvcallen, 2024-11-12:
Hooray! After using your maths to compare with my 50 ohm 100mm lines, both with tanD only AND with tanD and conductor losses, I have an essentially perfect agreement with AWR, after comparing using Matlab's s2rlgc. Find attached the graphs.

I do believe this is finally enough proof that the maths you sent should be added into skrf, at least in the DefinedAEpTandZ0 class's __init__ function (perhaps with some "z0_nominal" flag OR maybe with some compatibility="AWR" as you suggested). Would you like to make the pull request or should I?

On a slight tangent, I still believe this formulation uses some approximation (based on what I've learnt in the maths so far). From my understanding, you cannot truly uniquely define a line's RLGC parameters based on these parameters alone without imposing some additional constraints (take a look at Pozar's table "Transmission Line Parameters for Some Common Lines"). I am therefore currently still investigating further, and will potentially make a write-up that summarises what I've learnt (and perhaps I'll make a second pull request if I find anything useful), but for now, to at least have this implemented in skrf would be great.

Thanks for all your hard work and I'm glad we were able to nail it in the end!

[image]
[image]

mhuser, 2024-11-12:
> I do believe this is finally enough proof that the maths you sent is correct and should somehow be added into skrf. Maybe this should just be done in the DefinedAEpTandZ0 class's **init** function (perhaps with some "z0_nominal" flag OR maybe with some compatibility="AWR" as you suggested). It could also maybe just be the default option, since it is mathematically correct. Let me know what you think, and then I am happy to make a pull request once we have decided.

This sounds good 😄 

In my opinion, the purpose of the `DefinedAEpTandZ0` is to be used like `TLINP`. Having the distorted model with nominal impedance as a default behavior makes sense. S21 should not change slightly from precedent implementation and S11 should agree better with most measurements. Should the user want to use a raw characteristic impedance, he can fall back to `DefinedGammaZ0` to realize the wildest transmission lines.

I just made a PR by basically copying and pasting the equations in the docstring and init section of the media.

mhuser, 2024-11-12:
> Thanks for all your hard work and I'm glad we were able to nail it in the end!

You are welcome! I would not have found this solution without your RLGC analysis.
The optimized model looks quite good with the PR.
[image]

The residuals with the `awr_tline.s2p` are also mostly in the 1e-6 range.
[image]

gvcallen, 2024-11-13:
@mhuser awesome thanks so much! Those residuals might even still be just due to my export precision settings. It is clear that the maths is the same!

Is there any estimate for when the next release is?

jhillairet, 2024-11-13:
> Is there any estimate for when the next release is?

there is no planning yet, but we can make a new release quicker if required/needed.

mhuser, 2024-11-13:
The changes were merged into master branch if you already want to give it a try.

mhuser, 2024-11-13:
@gvcallen I noticed something strange: the network before optimization (I had to change the inductor value with the one in the .txt file) still exhibit a slight difference against AWR. After optimization, the difference is reduced but values of `z0` and `ep_r` for the SMA line are changed. This line has neither conductor nor dielectric loss set.

Is it possible to have an .s2p parameter for this line too? Or maybe the loss model is not set to full for this component?

gvcallen, 2024-11-21:
@mhuser unfortunately I am very busy for a while now trying to meet a deadline, and the current fix seems to be sufficient in my code. Although I think further investigation might be helpful, I only think I'll be able to help again in the new year. Sorry about that

mhuser, 2024-11-21:
@gvcallen
No problem. I am quite busy too currently and I think `DefinedAEpTandZ0` is already nicely improved in current state and also fit my own needs.

My proposition is to close this issue as of now. It will be possible to reopen it or make another more specific if required in the future.

PR #1202 (2024-11-13): Use RLGC distortion model and nominal impedance in DefinedAEpTandZ0 media
Make  `DefinedAEpTandZ0` aware of the distortion caused by the losses.
Because of the distortion introduced by the conductor and dielectric losses, the characteristic impedance can be frequency-dependent with an imaginary part. The characteristic impedance is computed with an RLGC model as follow.

$$R = 2 Z_n \alpha_C$$
$$L = \frac{Z_n \sqrt{\epsilon_r}}{C_0}$$
$$G = \frac{2}{Z_n} \alpha_D$$
$$C = \frac{\sqrt{\epsilon_r}}{C_0 Z_n}$$
$$Z_0 = \sqrt{\frac{R + j\omega L}{G + j\omega C}}$$

Where $Z_n$ is the nominal impedan [... cut]
closed 2025-01-02 by mhuser