scikit-rf/scikit-rf #1185: Djordjevic-Svensson dispersion lead to negative tanD in DefinedAEpTandZ0 Media

mhuser, 2024-10-14:
**Describe the bug**
The tanD should be positive and tend to zero before and after the poles, like illustrated in [this presentation](https://www.simberian.com/Presentations/Tutorial_A_Material_World_Final.pdf).
[image]

Currently the tanD become negative before and after the poles with `DefinedAEpTandZ0` `Media`:
[image]

**To Reproduce**
```Python
import skrf as rf
from skrf.media.definedAEpTandZ0 import DefinedAEpTandZ0
import numpy as np
import matplotlib.pyplot as plt

f_djsv = rf.Frequency(1e2, 1e15, 10001, 'Hz', sweep_type='log')
m_djsv = DefinedAEpTandZ0(frequency=f_djsv, z0=50, ep_r=4.25, tanD=0.02, z0_port=50,
                     f_low=1e4, f_high=1e13, f_Ep=1e9,
                     model='djordjevicsvensson')

fig = plt.figure()
ax_ep_r = fig.add_subplot(2, 1, 1)
ax_ep_r.set_xscale('log')
ax_tand = fig.add_subplot(2, 1, 2)
ax_tand.set_xscale('log')
ax_ep_r.plot(f_djsv.f_scaled, np.real(m_djsv.ep_r_f),
             label = "f_min=1e4, f_max=1e13, f_ep=1e9, ep_r=4.25, tand=0.02")
ax_tand.plot(f_djsv.f_scaled, m_djsv.tand_f)
ax_ep_r.set_ylabel(r'$\epsilon_{r,eff}$')
ax_ep_r.legend(loc="upper right")
ax_ep_r.grid(True, which='both', ls="-")
ax_ep_r.set_ylim((3.5, 6))
ax_tand.set_xlabel(f'Frequency ({f_djsv.unit})')
ax_tand.set_ylabel('tanD')
ax_tand.grid(True, which='both', ls="-")
fig.suptitle('Djordjevic and Svensson Dispersion')
fig.tight_layout()
fig.savefig('djordjevicsvensson.png')
```

**System**
 - scikit-rf version master e668596ba1cc102500e4a6959f3ce4c26599b45f

mhuser, 2024-10-14:
Discovered in #1184

gvcallen, 2024-10-14:
Without understanding the maths yet myself, it seems like the implementation of `djordjevicsvensson` in `MLine` does not have this error. That is, if I replace the following in `media/definedAEpTandZo.py` in method `ep_r_f`
``
# compute the slope for a log frequency scale, tanD dependent.
m = (ep_r*tand) * (pi/(2*log(10)))
# value for frequency above f_high
ep_inf = (ep_r - 1j*ep_r*tand - m*log((f_high + 1j*f_ep)/(f_low + 1j*f_ep)))
return ep_inf + m*log((f_high + 1j*f)/(f_low + 1j*f))
``
with the maths from `media/mline.py`
``
k = log((f_high + 1j * f_ep) / (f_low + 1j * f_ep))
fd = log((f_high + 1j * f) / (f_low + 1j * f))
ep_d = -tand * ep_r  / imag(k)
ep_inf = ep_r * (1. + tand * real(k) / imag(k))
return ep_inf + ep_d * fd
``
I get:

[image]

Maybe this is the solution?

mhuser, 2024-10-14:
@gvcallen thank you for this proposition, you are perfectly right!
I already proposed this fix in #1186 so we did read the same lines of code in parallel 😉

PR #1186 (2024-10-15): Fix Djordjevic-Svensson dispersion lead to negative tanD in DefinedAEpTandZ0
Fix #1185 
Should looks like this:
[image]

Before the fix, tanD is going below zero at low and high frequencies outside the poles:
[image]

After the fix, tanD tanD tends toward zero and keep positive:
[image]
closed 2024-10-15 by mhuser