scikit-rf/scikit-rf #1197: `plot_s_db_time()` and keyword arguments `m` & `n` not working as expected

thenorthcore, 2024-11-06:
**Describe the bug**
Plotting S-parameters specified by keywords `m` and `n` for `ntwk.plot_s_db_time()` is throwing a `TypeError`, but is working for `ntwk.plot_s_db()`.

**To Reproduce**
Minimal example to reproduce the issue:
```
import skrf
import numpy as np
import matplotlib.pyplot as plt

if __name__ == "__main__":
    f = np.linspace(10, 1000, 101)
    s=np.random.standard_normal((101, 2, 2)) + 1.j*np.random.standard_normal((101, 2, 2))
    ntwk = skrf.Network(name="Random Network", s=s, f=f, f_unit='MHz')

    plt.subplot(2,1,1)
    ntwk.plot_s_db(m=1, n=0)

    plt.subplot(2,1,2)
    ntwk.plot_s_db_time(m=1, n=0)

    plt.show()
```

This results in the following `TypeError` response:
```
Traceback (most recent call last):
  File "[...]\2024-11-06-skrf_s_plot_db_time_mn_bug.py", line 15, in <module>
    ntwk.plot_s_db_time(m=1, n=0)
  File "[...]\site-packages\skrf\network.py", line 4927, in plot_s_db_time
    return rfplt.plot_s_db_time(self, center_to_dc, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "[...]\site-packages\skrf\plotting.py", line 994, in plot_s_db_time
    return netw.windowed(window, normalize, center_to_dc).plot_s_time_db(*args,**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "[...]\site-packages\skrf\util.py", line 88, in method
    return func(self, *args1, *args2, **kwargs1, **kwargs2)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "[...]\site-packages\skrf\util.py", line 110, in wrapper
    func(*args, ax=ax, **kwargs)
TypeError: Network.plot_attribute() got multiple values for argument 'm'
```

**Expected behavior**
The time domain response of the network is plotted the same way as the call to `ntwk.s21.plot_s_db_time()`, where we first access a subset of the network.

**System**
 - OS: Win11 \w miniconda 
 - scikit-rf version 1.0.0 & 1.4.0

Asachoo, 2024-11-12:
> As far as I understand the documentation so is `plot_s_db_time()` not doing the same signal processing as `plot_s_time_db()`. Inspection of the corresponding code shows the following signal processing [ref plotting.py](https://github.com/scikit-rf/scikit-rf/blob/71f2270dbc51d1b27c15dc05c459d6ca3199e226/skrf/plotting.py#L992C1-L994C89) :
> 
> ```
> def plot_s_db_time(netw: Network, *args, window: str | float | tuple[str, float]=('kaiser', 6),
>         normalize: bool = True, center_to_dc: bool = None, **kwargs):
>     return netw.windowed(window, normalize, center_to_dc).plot_s_time_db(*args,**kwargs)
> ```
> 
> That means `plot_s_db_time()` applies windowing to the network before calling `plot_s_time_db()`.
> 
> My current workaround is the following:
> 
> ```
> ntwk.windowed(('kaiser', 16), True, None).plot_s_time_db(m=i, n=j)
> ```
> 
> where I basically copy the intended behaviour of `plot_s_db_time()` passing `m` and `n` to the underlying function call. Hence, I suspect the bug to be related to passing `*args` and `**kwargs` from `plot_s_db_time()` to `plot_s_time_db()`, but I don't know how this works internally.

My apologies for the confusion earlier. I'll investigate the internal implementation to understand the issue.

thenorthcore, 2024-11-12:
Thank you for looking into it!
I understood the distinction between both functions only after reading about it in the [Time Domain and Gating Example](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Time%20Domain.html#Interpreting-Time-Domain) in detail. Especially the [What the hell is Windowing?](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Time%20Domain.html#What-the-hell-is-Windowing?) part is quite nicely written. Not sure if the function names can be made clearer than they are.

FranzForstmayr, 2024-11-12:
The function name is indeed confusing...

closed 2024-11-15 by Asachoo

Asachoo, 2024-11-15:
The fix for this issue has been merged into the master branch. To resolve the issue, you can update your local repository to the latest master version. @thenorthcore. 
Regarding the `ntwk.plot_s_db_time()` method, does anyone have any suggestions for a more intuitive name that would better reflect its functionality?

thenorthcore, 2024-12-20:
Could be that `plot_s_windowed_time_db()` and `plot_s_time_db()` are more expressive. But on the other hand, the first name would be even longer. :/