scikit-rf/scikit-rf #1199: Incompatibility with numpy 1

neutrinonerd3333, 2024-11-08:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**
Vector fitting not compatible with numpy 1. This is due to [this diff](https://github.com/scikit-rf/scikit-rf/pull/1108/commits/aed181c2b751db26762bbe42a2520d0747a18dd1) upgrading a call to `np.trapz` to numpy 2 syntax `np.trapezoid`. We should either
* explicitly require `numpy>=2.0.0` in `pyproject.toml`
* revert this diff and change the `numpy` dependency in `pyproject.toml` to `numpy~=1.21`
* keep the numpy dependency the same and call the appropriate trapezoid function depending on detected numpy version.

**To Reproduce**
1. Install `numpy<2.0.0` (say, 1.26.4)
2. Run code involving `VectorFitting.auto_fit()`
3. Get `AttributeError: module 'numpy' has no attribute 'trapezoid'`

**Expected behavior**
No errors should be thrown. 

**System**
 - OS: Mac OS 14.1.1
 - scikit-rf version 1.4.0 (but this should also be reproducible on versions down to 1.2.0)

Vinc0110, 2024-11-10:
Wow, why did they change the name of the function? Seems totally unnecessary...

@FranzForstmayr and others: Any opinions on our options?

Judging by the commit message, this change has been done explicitly in an attempt to upgrade to numpy 2. I think we should change the project requirements to numpy>=2.0.0.

jhillairet, 2024-11-10:
I suggest requiring numpy >=2.0 is now ok. Numpy 2.0 is compatible with relatively old python versions, so it should not break too much retro compatibility..

FranzForstmayr, 2024-11-10:
I would keep numpy<2 as Long we keep Python 3.8, because no pre built wheels are available for 3.8. In this case numpy would need to get build for 3.8 which could make troubles.

jhillairet, 2024-11-10:
Python 3.8 security support ended a month ago. So, should we keep it supported?

FranzForstmayr, 2024-11-10:
I would like to keep it a little bit longer as we have VNAs running windows 7 (only python3.8) somewhere in the company but having scikit rf directly installed. These VNAs are not connected to the internet obviously :).
But I know at some point we need to drop it, however maybe a few months more :D

Asachoo, 2024-11-11:
I had initially presumed that the scikit-rf version on older Windows systems or legacy projects (based on Python 3.8) might still be at the state when the project was initiated.

Upon reviewing  [PyPi Download Stats](https://pypistats.org/packages/scikit-rf), it's evident that while Python 3.11 has become the predominant version, there is still a **significant 10% to 20%** of users download scikit-rf on versions <= 3.8.

Thus, abruptly ending support for versions below 3.8 could leave those preparing to install in a challenging position. It is indeed prudent to allocate some time to encourage users to upgrade their Python environments.

closed 2024-11-12 by FranzForstmayr