scikit-rf/scikit-rf #1211: Pylint alerts corrections as part of an intervention experiment

evidencebp, 2024-11-19:
Pylint alerts are correlated with tendency to bugs and harder maintenance.
I'd like to conduct a software engineering experiment regarding the benefit of [Pylint](https://pypi.org/project/pylint/) alerts removal.
The experiment is described [here](https://github.com/evidencebp/pylint-intervention/).

In the experiments, Pylint is used with some specific alerts, files are selected for intervention and control.
After the interventions are done, one can wait and examine the results.

Your repository is expected to benefit from the interventions.
I'm asking for your approval for conducting an intervention in your repository.

See examples of interventions in [stanford-oval/storm](https://github.com/stanford-oval/storm/pull/181), [gabfl/vault](https://github.com/gabfl/vault/pull/82), and [coreruleset/coreruleset](https://github.com/coreruleset/coreruleset/pull/3837).

You can see the [planed interventions](https://github.com/evidencebp/pylint-intervention/tree/main/interventions/candidates/scikit-rf_scikit-rf_interventions_September_28_2024.csv)

May I do the interventions?

FranzForstmayr, 2024-11-19:
This project does not use pylint anymore, a subset of your rules are checked with the ruff linter. 
We won't switch back to pylint, as the ruff experience is great so far. 

The currently enforced ruleset is not complete and could be expanded by some more rules, however a big refactoring is critical as long as the test coverage isn't at a better level. The risk of introducing new bugs is bigger than the gain of having clean code.

So we definitely won't accept a huge PR with all the planned interventions, but personally I'm open for selected improvements in the code style as long as backward compability is preserved and the code is well covered with unit tests.

evidencebp, 2024-11-19:
Part of the design of the experimnet is to choose interventions by random, to avoid selection bais.
Therefore, I can choose alert type before intervention and still avoid the bias.
Are you ok with me doing interventions of
{'broad-exception-caught': 1,
 'line-too-long': 17,
 'simplifiable-if-statement': 1,
 'superfluous-parens': 7,
 'unnecessary-pass': 2,
 'using-constant-test': 1}

Another design need is to do a small needed change, to avoid unrelated influence.
Adding new tests, should be such a source of benefit hence might make the intervention not suitable for the experiment.
The alerts types that I choose above are local and usuallly a change of few lines.
Is it OK to do them without adding new tests?

Regardless of your decision, please note the alert using-constant-test in the file skrf\media\circularWaveguide.py
In line 356 in the method gamma of the class CircularWaveguide there is the if:
       if False: #self.m==1 and self.n==0:

The True section is never reached, which seems to be unintended.
Maybe it is a leftover from debug and the condition should be the one in the comment.
Git blame hints 3e78224a2 Max Kellermeier 2020-05-18

evidencebp, 2024-11-19:
By the way, if the correction to the if False is indeed the commented condition, I'll be happy to fix it and have at least tat as part of the experiment.

jhillairet, 2024-11-19:
not interested, thank you.