scikit-rf/scikit-rf #1217: The function n_twoports_2_nport cannot accurately distinguish port numbers.

lyz1111111121, 2024-12-03:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**
When I was reading a large-dimensional S-parameter port matrix (e.g., 18 ports), I found that the data did not correspond to the generated S-parameter matrix ports. For example, S{14,18} was read as S{14,1}.

**To Reproduce**
If a S-parameter matrix with more than 10 ports is constructed, theoretically S{2,1} may be confused with S{2,10}.
Specifically, the number of ports I am using is 18, and in the results, S{14,18} is confused with S{14,1}.
> note that , the s2p subnetwork is named with 'S{i,j}', where i, j are port indexes.

**Expected behavior**
The output n-port network should correspond to each S2p file.

**System**
 - OS: Windows 10
 - scikit-rf version:1.3.0

**Here is one approach to fix this problem**
Using the re-expr, searching the port index in subntwk.name, like this: 
[image]

ZiadHatab, 2025-02-02:
@lyz1111111121 

Not sure if I understood your issue well. I use this function quite often and it works as intended for me. One thing to be aware of is that port numbering starts from left and continues to the other side. So, if you have a two port network and consider S21, in the 2N-port your S21 is located at S[i+N, i] where i is up to you to pick when naming the ports.

Here is an image illustrating this:

[image]

lyz1111111121, 2025-02-05:
@ZiadHatab 
Thanks for your reply, 

I suspect the issue might be in the following lines of code:  
```python
for subntwk in ntwk_list:
    for m, n in nport.port_tuples:   
        if m != n and m > n:
            if f"{m + offby}{port_sep}{n + offby}" in subntwk.name: 
                pass
            elif f"{n + offby}{port_sep}{m + offby}" in subntwk.name: 
                subntwk = subntwk.flipped()
            else:
                continue

            for mn, jk in zip(product((m, n), repeat=2), product((0, 1), repeat=2)):
                m, n, j, k = mn[0], mn[1], jk[0], jk[1]
                nport.s[:, m, n] = subntwk.s[:, j, k]
                nport.z0[:, m] = subntwk.z0[:, j]
```

The intended functionality of this code is to iterate over each `subntwk` in `ntwk_list` and populate `nport` with the corresponding data. Specifically, it checks whether a port pair `(m + offby, n + offby)` matches `subntwk.name`. However, there is a flaw in the condition:  

```python
f"{m + offby}{port_sep}{n + offby}" in subntwk.name
```

This can lead to incorrect matches. Consider the following scenario:  

- `subntwk.name = 'S[14,10]'`
- `m = 13, n = 0, offby = 1`
- This results in `"{m + offby}{port_sep}{n + offby}" = '14,1'`
- The condition `'14,1' in 'S[14,10]'` evaluates to `True`

As a result, the data from `'S[14,10]'` is written to `nport.s[:, 13, 0]`, potentially causing port mismatches. This might be the reason behind the observed issue.