scikit-rf/scikit-rf #1221: innerconnect() produces numerical errors in special cases

Asachoo, 2024-12-20:
**Describe the bug**
The `Network.innerconnect()` method may encounter numerical inaccuracies under certain specific conditions.

**To Reproduce**
```python3
import numpy as np

import skrf as rf

def gen_s_para(nports: int):
    s_para = np.zeros((1, nports, nports), dtype=complex)

    rng = np.random.default_rng(123)
    s_para += rng.random((1, nports, nports)) * 2.0 - 1.0
    s_para += 1.0j * (rng.random((1, nports, nports)) * 2.0 - 1.0)
    s_para += np.swapaxes(s_para, 1, 2)
    s_para /= 2.0
    return s_para

if __name__ == "__main__":
    """
    Generate a random S-parameter matrix and it's inverse Network object.
    """
    s_para = gen_s_para(3)
    ntwk = rf.Network(s=s_para, f=1, name="ntwk", z0=(3, 4, 5))
    ntwk_mir = rf.Network(s=np.linalg.inv(s_para), f=1, name="ntwk_mir", z0=(3, 4, 5))

    print(
        '"""\n\
First step: Connect ntwk and ntwk_mirs ports 1.\n\
"""'
    )
    # Connect ntwk and ntwk3_mirs' ports 1 in connect() method
    ntwk_cnx1 = rf.connect(ntwk, 1, ntwk.mirror, 1)

    # Connect ntwk and ntwk3_mirs' ports 2 in Circuit() method
    port1 = rf.Circuit.Port(frequency=ntwk.frequency, name="port1", z0=3)
    port2 = rf.Circuit.Port(frequency=ntwk.frequency, name="port2", z0=5)
    port3 = rf.Circuit.Port(frequency=ntwk.frequency, name="port3", z0=3)
    port4 = rf.Circuit.Port(frequency=ntwk.frequency, name="port4", z0=5)

    ckt1 = [
        [(port1, 0), (ntwk, 0)],
        [(port2, 0), (ntwk, 2)],
        [(port3, 0), (ntwk_mir, 0)],
        [(port4, 0), (ntwk_mir, 2)],
        [(ntwk, 1), (ntwk_mir, 1)],
    ]
    ntwk_ckt1 = rf.Circuit(ckt1, name="ntwk_ckt1").network

    # Check if the result is equal
    print(f"Dose the result of ntwk_ckt1 equal to ntwk_cnx1? {np.allclose(ntwk_ckt1.s, ntwk_cnx1.s)}")
    print("ntwk_cnx1:\n", ntwk_cnx1.s)
    print("ntwk_ckt1:\n", ntwk_ckt1.s)
    print()

    print(
        '"""\n\
Second step: Connect ntwk_cnx1 and ntwk_ckt1 itself.\n\
"""'
    )
    # Connect ntwk_cnx in innerconnect() method
    ntwk_cnx2 = rf.innerconnect(ntwk_cnx1, 1, 3)

    # Connect ntwk_ckt in Circuit() method
    ckt_step2 = [
        [(port1, 0), (ntwk_ckt1, 0)],
        [(port3, 0), (ntwk_ckt1, 2)],
        [(ntwk_ckt1, 1), (ntwk_ckt1, 3)],
    ]
    ntwk_ckt2 = rf.Circuit(ckt_step2, name="ntwk_ckt2").network

    # Check if the result is equal
    print(f"Dose the result of ntwk_cnx2 equal to ntwk_ckt2? {np.allclose(ntwk_cnx2.s, ntwk_ckt2.s)}")
    print("ntwk_cnx2:\n", ntwk_cnx2.s)
    print("ntwk_ckt2:\n", ntwk_ckt2.s)
    print()

    print(
        '"""\n\
Third step: Cross validate the result.\n\
"""'
    )

    # Connect ntwk_cnx in innerconnect() method
    ntwk_cnx3 = rf.innerconnect(ntwk_ckt1, 1, 3)

    # Connect ntwk_ckt in Circuit() method
    ckt_step3 = [
        [(port1, 0), (ntwk_cnx1, 0)],
        [(port3, 0), (ntwk_cnx1, 2)],
        [(ntwk_cnx1, 1), (ntwk_cnx1, 3)],
    ]
    ntwk_ckt3 = rf.Circuit(ckt_step2, name="ntwk_ckt2").network

    # Check if the result is equal
    print(f"Dose the result of ntwk_cnx3 equal to ntwk_ckt3? {np.allclose(ntwk_cnx3.s, ntwk_ckt3.s)}")
    print("ntwk_cnx2:\n", ntwk_cnx3.s)
    print("ntwk_ckt2:\n", ntwk_ckt3.s)

```
And the results,
```bash
"""
First step: Connect ntwk and ntwk_mirs ports 1.
"""
Dose the result of ntwk_ckt1 equal to ntwk_cnx1? True
ntwk_cnx1:
 [[[ 0.16965925+0.18890222j -0.35995574-0.10801374j
    0.66611608+0.17465429j  0.32283623+0.33501835j]
  [-0.35995574-0.10801374j  0.53812674-0.1408294j
    0.28739038-0.47797245j  0.33388392-0.17465429j]
  [ 0.66611608+0.17465429j  0.28739038-0.47797245j
   -0.1286421 -0.54419515j -0.60019549+0.34130622j]
  [ 0.32283623+0.33501835j  0.33388392-0.17465429j
   -0.60019549+0.34130622j  0.61273796+0.59274434j]]]
ntwk_ckt1:
 [[[ 0.16965925+0.18890222j -0.35995574-0.10801374j
    0.66611608+0.17465429j  0.32283623+0.33501835j]
  [-0.35995574-0.10801374j  0.53812674-0.1408294j
    0.28739038-0.47797245j  0.33388392-0.17465429j]
  [ 0.66611608+0.17465429j  0.28739038-0.47797245j
   -0.1286421 -0.54419515j -0.60019549+0.34130622j]
  [ 0.32283623+0.33501835j  0.33388392-0.17465429j
   -0.60019549+0.34130622j  0.61273796+0.59274434j]]]

"""
Second step: Connect ntwk_cnx1 and ntwk_ckt1 itself.
"""
Dose the result of ntwk_cnx2 equal to ntwk_ckt2? False
ntwk_cnx2:
 [[[ 0.02373405+0.07948226j  0.87699915+0.00496608j]
  [ 0.84452264+0.09092597j -0.07370249-0.25693232j]]]
ntwk_ckt2:
 [[[-6.10622664e-16+0.00000000e+00j  1.00000000e+00+0.00000000e+00j]
  [ 1.00000000e+00-1.22124533e-15j -8.88178420e-16-4.44089210e-16j]]]

"""
Third step: Cross validate the result.
"""
Dose the result of ntwk_cnx3 equal to ntwk_ckt3? False
ntwk_cnx2:
 [[[-0.23290456-0.07062239j  1.18700692-0.30899656j]
  [ 1.57996566+0.11850763j -0.38320353+0.79048859j]]]
ntwk_ckt2:
 [[[-6.10622664e-16+0.00000000e+00j  1.00000000e+00+0.00000000e+00j]
  [ 1.00000000e+00-1.22124533e-15j -8.88178420e-16-4.44089210e-16j]]]
```

**Expected behavior**
The `innerconnect()` function should return a `Network` that is approximately equal to `Thru`, similar to what the `Circuit()` method achieves, given that `ntwk_mir` is a is a meticulously designed `Network` whose s-parameters are derived from the `inverse` of `ntwk`'s s-parameters. This extreme case providing a reliable and precise counterpart for simulations and analysis.

Asachoo, 2024-12-20:
The `parallelconnect()` method performs well in handling this corner case.
```python3
Python 3.13.0 (main, Oct 16 2024, 08:05:40) [Clang 18.1.8 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy as np
>>> import skrf as rf
matplotlib not found while setting up plotting
>>> def gen_s_para(nports: int):
...     s_para = np.zeros((1, nports, nports), dtype=complex)
...     rng = np.random.default_rng(123)
...     s_para += rng.random((1, nports, nports)) * 2.0 - 1.0
...     s_para += 1.0j * (rng.random((1, nports, nports)) * 2.0 - 1.0)
...     s_para += np.swapaxes(s_para, 1, 2)
...     s_para /= 2.0
...     return s_para
...     
>>> s_para = gen_s_para(3)
>>> ntwk = rf.Network(s=s_para, f=1, name="ntwk", z0=(3, 4, 5))
>>> ntwk_mir = rf.Network(s=np.linalg.inv(s_para), f=1, name="ntwk_mir", z0=(3, 4, 5))
>>> ntwk_pcnx1 = rf.parallelconnect((ntwk, ntwk_mir), (1, 1), name="ntwk_pcnx1")
>>> ntwk_pcnx2 = rf.parallelconnect((ntwk_pcnx1,), ((1, 3),), name="ntwk_pcnx2")
>>> ntwk_pcnx2
2-Port Network: 'ntwk_pcnx2',  1.0-1.0 Hz, 1 pts, z0=[3.+0.j 3.+0.j]
>>> ntwk_pcnx2.s
array([[[-2.77555756e-17-1.17961196e-16j,
          1.00000000e+00-4.44089210e-16j],
        [ 1.00000000e+00+8.32667268e-17j,
          0.00000000e+00-1.11022302e-16j]]])
>>> 
```

Asachoo, 2024-12-23:
## Update
I conducted verifications in `ADS` and `Qucs-S`, and it can be observed that the model using the `Circuit` method (`ADS`, `skrf.Circuit()`) yielded one set of results, while the model using the `connect()` method (`Qucs-S`, `skrf.connect/skrf.innerconnect()`) produced another set of results.

## Notes
1. It is confirmed through the source code that `Qucs-S` employs the `connect()` method. And my assumption that `ADS` uses the `Circuit()` method is based on the fact that `ADS` can provide s-parameters for ports that are not directly connected (such as S61), which does not seem to be the behavior of subnetwork growth;
2. The current `innerconnect()` method exhibits numerical errors due to `det=[[0, 0], [0, 0]]`, which were not present before vectorization.

## `ADS`'s results
[image]

## `Quos-S`'s results
[image]

## `scikit-rf`'s results
```python3
import skrf as rf

media = rf.media.DefinedGammaZ0(z0=50)

# Create 2 tee objects
t1 = media.tee(name="t1")
t2 = media.tee(name="t2")

t1.write_touchstone("t1.s3p")

# Schematic of the test circuit
#    +------+    +------+
#    |     1|----|0     |
# p1-|0     |    |     2|-p2
#    |     2|----|1     |
#    +------+    +------+
#       t1          t2

# `Connect` method to connect them to be an thru
tmp = rf.connect(t1, 1, t2, 0)
tmp.write_touchstone("tmp.s4p")
rst = rf.innerconnect(tmp, 1, 2)

# Check the result from `innerconnect` method
print(f"Connect result from `innerconnect` method: {rst.s[0, :, :]}")

# `Circuit` method to connect them to be an thru
p1 = rf.Circuit.Port(frequency=media.frequency, name="p1")
p2 = rf.Circuit.Port(frequency=media.frequency, name="p2")

cnx = [
    [(p1, 0), (t1, 0)],
    [(t1, 1), (t2, 0)],
    [(t1, 2), (t2, 1)],
    [(p2, 0), (t2, 2)],
]

# Check the result from `Circuit` method
circuit = rf.Circuit(cnx)
ntwk = circuit.network
print(f"Connect result from `Circuit` method: {ntwk.s[0, :, :]}")
```
```bash
(scikit-rf) ➜  scikit-rf git:(FixIssue) ✗ python inner_connect_issue.py 
/home/lansus/WorkFile/Git/scikit-rf/skrf/network.py:6426: RuntimeWarning: divide by zero encountered in divide
  tmp_a = Ael * (Alk / det) + Aek * (All / det)
/home/lansus/WorkFile/Git/scikit-rf/skrf/network.py:6426: RuntimeWarning: invalid value encountered in divide
  tmp_a = Ael * (Alk / det) + Aek * (All / det)
/home/lansus/WorkFile/Git/scikit-rf/skrf/network.py:6426: RuntimeWarning: invalid value encountered in multiply
  tmp_a = Ael * (Alk / det) + Aek * (All / det)
/home/lansus/WorkFile/Git/scikit-rf/skrf/network.py:6427: RuntimeWarning: divide by zero encountered in divide
  tmp_b = Ael * (Akk / det) + Aek * (Akl / det)
/home/lansus/WorkFile/Git/scikit-rf/skrf/network.py:6427: RuntimeWarning: invalid value encountered in divide
  tmp_b = Ael * (Akk / det) + Aek * (Akl / det)
/home/lansus/WorkFile/Git/scikit-rf/skrf/network.py:6427: RuntimeWarning: invalid value encountered in multiply
  tmp_b = Ael * (Akk / det) + Aek * (Akl / det)
Connect result from `innerconnect` method: [[nan+nanj nan+nanj]
 [nan+nanj nan+nanj]]
Connect result from `Circuit` method: [[4.44089210e-16+0.j 1.00000000e+00+0.j]
 [1.00000000e+00+0.j 1.11022302e-16+0.j]]
```

kpobrien, 2024-12-23:
Nice work looking into these corner cases @Asachoo! The `Circuit` method also has this issue if you make the first connection with `interconnect` and the second with `Circuit`:
```python
import skrf as rf
import numpy as np

media = rf.media.DefinedGammaZ0(z0=50)

# Create 2 tee objects
t1 = media.tee(name="t1")
t2 = media.tee(name="t2")

# t1.write_touchstone("t1.s3p")

# Schematic of the test circuit
#    +------+    +------+
#    |     1|----|0     |
# p1-|0     |    |     2|-p2
#    |     2|----|1     |
#    +------+    +------+
#       t1          t2

# `Connect` method to connect them to be an thru
tmp = rf.connect(t1, 1, t2, 0)
# tmp.write_touchstone("tmp.s4p")
# rst = rf.innerconnect(tmp, 1, 2)

# `Circuit` method to connect them to be an thru
p1 = rf.Circuit.Port(frequency=media.frequency, name="p1")
p2 = rf.Circuit.Port(frequency=media.frequency, name="p2")

cnx = [
    [(p1, 0), (tmp, 0)],
    [(tmp, 1), (tmp, 2)],
    [(p2, 0), (tmp, 3)],
]

# Check the result from `Circuit` method
circuit = rf.Circuit(cnx)
# x,t,A_idx,B_idx,C_idx,D_idx = s_external_matrices(circuit)
ntwk = circuit.network
print(f"Connect result from `Circuit` method: {ntwk.s[0, :, :]}")
```
```python
LinAlgError                               Traceback (most recent call last)
Cell In[1], line 39
     37 circuit = rf.Circuit(cnx)
     38 # x,t,A_idx,B_idx,C_idx,D_idx = s_external_matrices(circuit)
---> 39 ntwk = circuit.network
     40 print(f"Connect result from `Circuit` method: {ntwk.s[0, :, :]}")

File ~/.local/lib/python3.13/site-packages/skrf/circuit.py:1262, in Circuit.network(self)
   1251 @property
   1252 def network(self) -> Network:
   1253     """
   1254     Return the Network associated to external ports.
   1255 
   (...)
   1259         Network associated to external ports
   1260     """
   1261     return Network(frequency = self.frequency, z0 = self.port_z0,
-> 1262                   s = self.s_external, name = self.name)

File ~/.local/lib/python3.13/site-packages/skrf/circuit.py:1240, in Circuit.s_external(self)
   1236 np.einsum('...ii->...i', t)[:] += 1
   1238 # Get the sub-matrices of inverse of intermediate temporary matrix t
   1239 # The method np.linalg.solve(A, B) is equivalent to np.inv(A) @ B, but more efficient
-> 1240 tmp_mat = np.linalg.solve(t[D_idx], t[C_idx])
   1242 # Get the external S-parameters for the external ports
   1243 # Calculated by multiplying the sub-matrices of x and t
   1244 S_ext = (x[A_idx] - x[B_idx] @ tmp_mat) @ np.linalg.inv(
[... 152 lines cut ...]
rank A: 3
det A: 0j
S_ext1 with np.linalg.lstsq:
 [[0.00000000e+00+0.j 1.00000000e+00+0.j]
 [1.00000000e+00+0.j 2.03960993e-16+0.j]]
S_ext2 with np.linalg.solve:
LinAlgError: Singular matrix

`Circuit` method: Both connections:
shape A: (6, 6)
rank A: 5
det A: (-2.4671622769447928e-17+0j)
S_ext3 with np.linalg.lstsq:
[[1.08958455e-15+0.j 1.00000000e+00+0.j]
 [1.00000000e+00+0.j 1.33248556e-15+0.j]]
S_ext4 with np.linalg.solve:
[[2.22044605e-16+0.j 1.00000000e+00+0.j]
 [1.00000000e+00+0.j 4.44089210e-16+0.j]]

Compare the results:
np.allclose(S_ext1,S_ext3): True
np.allclose(S_ext3,S_ext4): True
```

For `interconnect`, the algorithm uses the analytic inverse of a 2x2 matrix, which could be replaced with the Moore-Penrose pseudoinverse to get the least squares solution.

Asachoo, 2024-12-25:
The solution to the singular value issue in `Circuit.s_external()` has been submitted as suggested @kpobrien, thank you; the resolution for `innerconnect()` will be addressed subsequently.

Asachoo, 2024-12-26:
Least-squares solution using numpy.linalg.lstsq() to address singularity issues within the Network.innerconnect() method has been integrated.

closed 2025-01-05 by jhillairet