scikit-rf/scikit-rf #1222: Sdd21 result somehow got distortion compared to the plot on VNA

morphyhsieh, 2024-12-21:
**Describe the bug**
I got this special s4p from our test engineer on ZNA67. Not sure why, the Sdd21 plot will have different trend compared to it shown on the VNA and other S4P tool(I used S-View to capture one example photo below as I can't access VNA right now) I don't see such big difference with other S4P files.
**To Reproduce**
Below is my code to load the S4P, should be quite simple as straight forward.
```
import skrf as rf
import matplotlib.pyplot as plt

s4p_file = '1T10.s4p'
network = rf.Network(s4p_file)

network.renumber([0, 1, 2, 3], [0, 2, 1, 3])
network.se2gmm(p=2, z0_mm = 85)

print(network)

fig= plt.subplots(1,1, sharex=True, figsize=(10,7))

network.plot_s_db(m=0, n=1)
plt.grid(True)
```

**Expected behavior**
I expect the plot should be aligned as there is not much calculation get involved in se2gmm.

**Data**
[S4P file that has such issue](https://github.com/user-attachments/files/18219882/s4p.zip)

**Screenshots**
The plot on S-View (VNA has similar result)
[image]
The plot generated by SKRF, green circle is one distinct example of distortion, but you actually could see the trend is different after 20GHz
[image]

**System**
 - OS: MacOS 15.1.1
 - scikit-rf version : 1.5.0

Ttl, 2024-12-22:
`network.se2gmm(p=2, z0_mm = 85)` will set all port impedances to 85 ohms, both differential and common mode. This is probably not what you want. If you don't want network to be renormalized then you can leave `z0_mm` out and if you want differential port impedance to be 85 ohms, then you can pass `network.se2gmm(p=2, z0_mm=[85, 85, 21.25, 21.25])`. The second one should be identical to renormalizing the measured S-parameters to 42.5 ohm impedance before mixed mode conversion.

jhillairet, 2024-12-26:
@morphyhsieh please check and tell us if the comment from @Ttl solves your issue.

morphyhsieh, 2024-12-26:
The suggestion of setting the se2gmm solves my problem, thx.

closed 2024-12-27 by Asachoo