scikit-rf/scikit-rf #1237: Compute Error between two Networks

jrmagers, 2025-01-14:
**Is your feature request related to a problem? Please describe.**

To compute the error ($\delta$) between two sets of s-parameters, similar to the `SMODEL` measurement in Microwave Office. For example, the _Average L1 Norm_ between network A ($S^A$) and network B ($S^B$) can be computed as:

$\delta =  \frac{1}{N^2} \sum_{i=1}^{N} \sum_{j=1}^{N} |S_{ij}^A - S_{ij}^B|$

Other error computations are _Average L2 Norm_, _Maximum L1 Norm_, and _Average Normalized L1 Norm_.

**Describe the solution you'd like**

There are two ways this could be implemented:

If `networka` and `networkb` are `Network()` objects:

## Method 1
```Python
error = skrf.smodel(networka, networkb, error_function='average_l1_norm')
```
## Method 2
```Python
error = networka.smodel(networkb, error_function='average_l1_norm')
```

Where `error` is a numpy array with length equal to `networka.frequency`.

I have the code written to implement `smodel()` but I'm looking for guidance on which method above seems more appropriate.

jhillairet, 2025-01-15:
Dear @jrmagers ,

Generally, both approaches are implemented. For example `rf.connect(ntwk1, ntwk2)` and `ntwkA.connect(ntwkB)`, so I would suggest doing both.

Considering the name of the method, `smodel`, I don't think it's super self-describing (I'm not a user of Microwave Office). I let other write their feelings...

jrmagers, 2025-01-15:
> Considering the name of the method, `smodel`, I don't think it's super self-describing (I'm not a user of Microwave Office). I let other write their feelings...

Agreed that `smodel` isn't too descriptive.  Is there an equivalent measurement in ADS?

Other ideas: `error()`, `s_error()`

jhillairet, 2025-01-16:
Maybe we could instead implement an error operator instead, so that we could have `s_error`, but also for all network parameters: `z_error`, `y_error`, etc...  ?