scikit-rf/scikit-rf #1243: Phase difference mTRL / SOLT

Zwelckovich, 2025-01-27:
Hi,
i did some measurements of a calibration substrate and did a comparison of different mTRL-calibrations. When i looked at the phase of an open structure the phase for all mTRL calibrations is different to the SOLT and the ideal model. Is there a reason why especially the mTRL have this phase difference?

[image]

ZiadHatab, 2025-02-02:
@Zwelckovich 

in SOLT you forced the response to your ideal model. It doesn't necessary mean it is correct. In TRL, the reflect standard (i.e., open in your case) is undefined during calculation. What you measuring is the response of the open in reference to your line standard's impedance.

If you doing ISS measurements, I would advice not to do SOLT/SOLR, unless you really sure of the model being correct. Open and shorts on an ISS are tricky to define (especially if they are located at probe tip).

Zwelckovich, 2025-02-05:
@ZiadHatab 
Ok thank you. I measured a AC-2 V2 substrate from MPI. So the the calibration models should be correct because they are described by the datasheet and also by their calibration software (calibria). Thats the reason why iam so curios about the results and the difference of the mTRL in comparison to the SOLT.

ZiadHatab, 2025-02-05:
No, no, no. Those are nominal values. Especially for the open, the exact value depends on many factors, such as whether you are using an absorber chuck, the type of probes, the pitch of the probes... You are better off using TRL or LRM methods. If you are unsure, consult with the technical experts from the vendor you bought the ISS from.

@jhillairet 
this is not a bug! I would move this to discussion.