scikit-rf/scikit-rf #1245: CPW mismatched line not working ?

ZiadHatab, 2025-02-02:
Hey,

I'm noticing that when I specify the impedance in the line function for the CPW media, it doesn't actually implement the mismatched line behavior I'm looking for. The argument `embed` seems not to do anything. I also see a warning that it is deprecated.

Of course, I can implement the impedance transformer and an ideal line and cascade everything manually, but I thought when I specify the `z0` in the line function, the line gets wrapped with this impedance?

```Python
import skrf as rf
from skrf.media import CPW

freq = rf.F(1, 100, 201, 'GHz')
cpw = CPW(freq, w=40e-6, s=51e-6, ep_r=12.9, t=5e-6, rho=2e-8)

mismatch_line = cpw.line(1.5e-3, 'm', z0=60)
mismatch_line_w_embed = cpw.line(1.5e-3, 'm', z0=60, embed=True)

# implementing the mismatch manually. This what the answer should look like
impedance_transformer = cpw.impedance_mismatch(z1=60, z2=cpw.z0)
line = cpw.line(1.5e-3, 'm')
my_mismatch_line = impedance_transformer**line**impedance_transformer.flipped()

# plot
mismatch_line.s21.plot_s_db(label='S21')
mismatch_line_w_embed.s21.plot_s_db(linestyle='--', label='S21 with embed=True')
my_mismatch_line.s21.plot_s_db(linestyle='-.', label='S21 manually implemented')
```

[image]

Ttl, 2025-02-02:
Specifying line and port impedances has for long time been confusing. There's `z0`, `z0_port`, `Z0`, and `z0_override` in few different places. Warning when using `embed=True` says what the problem is: `Lines are renormalized to the media port impedance 'z0_port' if it is not None, otherwise they use characterisic impedance 'z0'.`.

If `z0_port` is not set, the port impedance is the line impedance. `line` `z0` argument changes the line impedance, but without `z0_port` it also changes the port impedance and there isn't any visible change. The only way to change port impedance is to set it when making `media`.

ZiadHatab, 2025-02-02:
OK, well that's not good 😅

So, I tested setting `z0_port` in the media, and it works in creating mismatched lines. But now, I cannot create a matched line. 

Am I crazy when I think port impedance should only be defined when creating the device and not in the media? 😅  The Media only describes the cross-section of the transmission line. There are no ports at that stage. Only after a device is created based on the media should you define the port impedance.

Ttl, 2025-02-02:
That's reasonable. You can work around it by using `renormalize`, that's what the code does internally. Should be easier to use than making mismatches manually:
```
mismatch_line = cpw.line(1.5e-3, 'm')
mismatch_line.renormalize(60)
```

ZiadHatab, 2025-02-02:
I forgot about renormalize. That's good for me. Thx!

closed 2025-02-02 by ZiadHatab